FreeCAD/FreeCAD #18679: Some DraftCreation unit tests fail randomly

kpemartin, 2024-12-22:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When running the full suite of unit tests in the Gui app, some of the `DraftCreation` tests fail non-deterministically.

Several of these tests, such as the one calling itself "Draft Dimension Angular" can sometimes produce a Windows exception 0xC0000005 (memory access violation). Furthermore, when handling this exception another error occurs in the form of a `std::exception` with the message "invalid format specifier" which causes a popup message window with the same text and ends the test run.

The actual original exception occurs in `View3DInventorPy::getSceneGraph` on the line `scene->ref()` because `scene` is null.

This only seems to occur if the entire test suite (TestApp.All) is run. It does not matter if this is done from the command line using `-r 0` or by using the test workbench.

This particular test even admits `Occasionally crashes` with no further explanation. Looking at the code history provides no details as to whether this is the "occasional crash" the comment refers to.

It would appear that the crash occurs because code delayed using `ToDo.delay` is delayed long enough that the document window is already closed (I'm guessing here) so there is no `scene` any more. The call to `ToDo.delay` is in `Tracker.__init__` in `gui_tracker.py`.

The full Console Log is attached, but here is a summarized extract:
```
....log of previous tests...
10:09:26    Temporary document 'DraftCreation'
10:09:26    Test 'Draft Dimension Angular'
10:09:26    Occasionally crashes
10:09:26    center=Vector (0.0, 0.0, 0.0)
10:09:26    angle1=20, angle2=70
10:09:26    dim_line=Vector (3.0, 1.0, 0.0)
10:09:32  Illegal storage access...
10:09:32  Traceback (most recent call last):
  File "D:\Users\KPMartin\Documents\Git\FreeCAD\build\Mod\Draft\draftutils\todo.py", line 138, in doTasks
    f(arg)
  File "D:\Users\KPMartin\Documents\Git\FreeCAD\build\Mod\Draft\draftguitools\gui_trackers.py", line 107, in _insertSwitch
    sg = self.get_scene_graph()
         ^^^^^^^^^^^^^^^^^^^^^^
  File "D:\Users\KPMartin\Documents\Git\FreeCAD\build\Mod\Draft\draftguitools\gui_trackers.py", line 97, in get_scene_graph
    return v.getSceneGraph()
           ^^^^^^^^^^^^^^^^^
RuntimeError: Illegal storage access! Please save your work under a new file name and restart the application!

10:09:32  Traceback (most recent call last):
  File "D:\Users\KPMartin\Documents\Git\FreeCAD\build\Mod\Draft\draftutils\todo.py", line 138, in doTasks
    f(arg)
  File "D:\Users\KPMartin\Documents\Git\FreeCAD\build\Mod\Draft\draftguitools\gui_trackers.py", line 107, in _insertSwitch
    sg = self.get_scene_graph()
         ^^^^^^^^^^^^^^^^^^^^^^
  File "D:\Users\KPMartin\Documents\Git\FreeCAD\build\Mod\Draft\draftguitools\gui_trackers.py", line 97, in get_scene_graph
    return v.getSceneGraph()
           ^^^^^^^^^^^^^^^^^
RuntimeError: Illegal storage access! Please save your work under a new file name and restart the application!

10:09:32  ToDo.doTasks, Unexpected error:
<class 'RuntimeError'>
in <bound method Tracker._insertSwitch of <draftguitools.gui_trackers.gridTracker object at 0x000001C45E806270>>(<pivy.coin.SoSwitch; proxy of <Swig Object of type 'SoSwitch *' at 0x000001C45FB34FC0> >)
...module version information...
10:09:32  Id: 21080: ERROR: SymGetLineFromAddr64, GetLastError: 487 (Address: 00007FFF8D7FEBE5)
10:09:32  Id: 21080: 00007FFF8D7FEBE5 (Coin4): (filename not available): SoBase::ref
...several traceback entries...
10:09:32  Id: 21080: D:\Users\KPMartin\Documents\Git\FreeCAD\src\Gui\Application.cpp (2312): Gui::Application::runApplication
...several more traceback entries...
10:09:32  Id: 21080: 00007FF82125CC91 (ntdll): (filename not available): RtlUserThreadStart
10:09:32  *** Unhandled Exception!
10:09:32     ExpCode: 0xC0000005
10:09:32     ExpFlags: 0
10:09:32  Unhandled std::exception caught in GUIApplication::notify.
The error message is: invalid format specifier
10:09:32  
10:09:32  ------------------------------------------------------------------------------
10:09:32    Temporary document 'DraftCreation'
10:09:32    Test 'Draft Dimension'
10:09:32    Occasionally crashes
10:09:32    a=Vector (0.0, 0.0, 0.0), b=Vector (9.0, 0.0, 0.0)
...module info again...
10:09:32  Id: 21080: G:\FreeCAD\FreeCAD-LibPack-chennes\working\python\Objects\frameobject.c (1498): PyFrame_GetBack
10:09:32  Id: 21080: G:\FreeCAD\FreeCAD-LibPack-chennes\working\python\Objects\frameobject.c (106): frame_getback
10:09:32  Id: 21080: G:\FreeCAD\FreeCAD-LibPack-chennes\working\python\Objects\object.c (1430): _PyObject_GenericGetAttrWithDict
...more traceback lines...
10:09:32  Id: 21080: 00007FF82125CC91 (ntdll): (filename not available): RtlUserThreadStart
10:09:32  *** Unhandled Exception!
10:09:32     ExpCode: 0xC0000005
10:09:32     ExpFlags: 0
```
Full log: [Null scene exception.zip](https://github.com/user-attachments/files/18222703/Null.scene.exception.zip)

This log seems to imply that there are two access violations, which would imply that there are two `ToDo` entries, each of which causes an exception once the message loop runs the delayed code. This double exception might ultimately be the cause of the "invalid format specifier"

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.39675 (Git)
Build type: Release
Branch: Issue16849B
Hash: b20509baff0609145476612fd432ef3e7ccb286b
Python 3.12.4, Qt 6.7.2, Coin 4.0.2, Vtk 9.3.0, OCC 7.8.0
Locale: English/Canada (en_CA)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Classic/
```

### Subproject(s) affected?

None

### Anything else?

The immediate fix is to change `View3DInventorPy::getSceneGraph` to return a python `None` is scene is null.

This does not explain the second "invalid format specifier" error, so perhaps that should be analyzed before fixing `View3DInventorPy::getSceneGraph`.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-12-23:
cc @Roy-043

Roy-043, 2025-01-02:
The issue is indeed that documents can get closed before delayed code that should act on them has had a chance to execute. I suppose this is an example of a 'race condition'. What is interesting is that the referenced code in gui_trackers.py actually checks if there is a valid 3D view. So it appears documents get closed at the exact moment that code runs.
https://github.com/FreeCAD/FreeCAD/blob/e977eb49e37ef836f695809263fa1f56a17cc657/src/Mod/Draft/draftguitools/gui_trackers.py#L93-L100

WorkingPlane.py is responsible for (some of) the delayed code. That code triggers the grid (among other things) based on view changes.
https://github.com/FreeCAD/FreeCAD/blob/e977eb49e37ef836f695809263fa1f56a17cc657/src/Mod/Draft/WorkingPlane.py#L1791-L1804

The tests that close documents too soon are actually dummy tests (`aux.fake_function`). There are 5 test files with such tests. 2 have already been commented out in TestDraft.py. If I comment out the other 3 (related to SVG, DXF and DWG) the issue seems to no longer occur.
[TestDraft.py.txt](https://github.com/user-attachments/files/18293381/TestDraft.py.txt) (Remove the .txt extension).

In a way that solution avoids the problem rather than solving it. To really solve it I suppose we should add `try`-`except` structures whenever we reference a 3D view.

@kpemartin Can please you test with the attached file? Thanks.

```
OS: Windows 8 build 9600
Architecture: x86_64
Version: 1.1.0dev.39708 (Git) Conda
Build type: Release
Branch: main
Hash: 173ff4da1ef4f075d56c0dc617d1bf00d5f49d4b
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: C/Default (C) [ OS: Dutch/Netherlands (nl_NL) ]
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods:
```

kpemartin, 2025-01-02:
I ran the file and it succeeded but the only way I could ever get the failure was to run the complete test suite (-r 0) so I'm not surprised this subset did not fail.

`Draft.get3DView` verifies that the view it returns has a `getSceneGraph` method but not whether that method will succeed so it is possible for it to return a value that already has no `scene`, causing the fault in `getSceneGraph`.

There is perhaps a larger time window between the `ToDo.delay()` call and execution of the delayed code where the main window can have an active window with a null scene. I would expect that eventually in the process of closing the window, it (the window) stops being the active window so `Draft.get3DView` finds nothing at all, but until that happens you're in this time window.

Come to think of it, it is possible that the window being closed by the test that did the delay call has actually closed, popping back to some unrelated previous active window, which may also be closing asynchronously. Asking for the Active window in (arbitrarily) delayed code and expecting to get any particular window just sounds like a Bad Idea because who knows what window will be active at that time.

One way to avoid this is to have the code that does the delay call query the view at that time, and pass it as an argument to the delayed code. But then, perhaps the whole reason for the delay is that this is too *early* to find the view.

kpemartin, 2025-01-19:
I'm trying to understand why `tracker.__init__` is using `ToDo.delay(self._insertSwitch, self.switch)` rather than just directly calling (or even in-lining) the `_insertSwitch` call. Obviously on a vague level I understand that the code is intended to protect against some code being executed when the environment is in some transient state where it is wrong to execute said code.

What I don't understand is:
1. What is this state, and what, specifically, in `_insertSwitch` cannot be done because of this state?
2. Why would we ever be in such a state?

Regarding point number 1:
The documentation string states merely
> Must not be called from an event handler (or other scene graph traversal).

The actions the code performs are calling `self.get_scene_graph()` followed by altering the scene graph to include the switch.
My thought is that this is actually an instance of the common situation of altering the contents of an collection while these contents are being traversed/iterated. Under some circumstances this can be a well-defined operation, but it often causes undefined behaviour (causing the traversal to repeat or skip entries, loop forever, generate an exception, or generate a diagnosed error).
If this is the case, the reference to an event handler in the documentation is a red herring; the situation to avoid is, simply, calling this code during a graph traversal. There should be the tacit understanding that code run in an event handler could occur during such a traversal, just as for any call that should not be made during some particular state.

Regarding point umber 2:
Why would any code be creating a tracker object while a graph traversal is in progress, either synchronously or in an event handler? I have not done a complete survey, but trackers seem to be created as part of main-level code execution of the create and modify actions on various Draft objects, This would mean they should never occur during a graph traversal.
Furthermore, there are other methods on tracker objects (`lowerTracker` and `raiseTracker`) which also obtain the scene graph and modify it but do so synchronously. I see nothing that would explain why these methods are any different from the constructor as to the state in which they are called.

I tried searching the commit history to see if there was some issue addressed by the delay call which might explain the problem this solves, but ran into a brick wall on commit 120ca87 "unify DLL export defines to namespace names" dated dated October 10, 2011, which adds almost 3 million lines of code apparently coming from nowhere.

I have experimentally replaced the `ToDo.delay` call with inlining of the body of `_insertSwitch` and the code appears to work, but of course I may not have run the app through some particular use case where the dely call is necessary. This change does not produce any diagnostic failures.

Similar observations apply to `tracker.finalize`

Roy-043, 2025-01-23:
I do not fully understand the reason for some of these delays either. Getting rid of them where possible would be great. I'll do some testing myself. There are many scenarios that need to be considered. Note that in my testing the delays used in WorkingPlane.py were required. But that may no longer be the case if other delays are removed.

Roy-043, 2025-01-23:
> Similar observations apply to `tracker.finalize`

If I remove the delay from that function the following scenario results in an error:

1. Startup WB is Part.
2. Start FreeCAD.
3. Create a new document.
4. Switch to the Draft WB.
5. Create a circle.

Error message:
```
Unhandled Base::Exception caught in GUIApplication::notify.
The error message is: Access violation
Unhandled Base::Exception caught in GUIApplication::notify.
The error message is: Access violation
```

kpemartin, 2025-01-24:
I got the same exception; it appears the first steps are unnecessary, or at least it is not specific to starting in the Part WB.

My first thought looking at the stack at the first exception is that the scene is damaged, since the exception is about 15 levels deep in Coin4.dll, with the topmost frame with source code being `View3DInventorViewer::renderScene()`

I did have suspicions that the `finalize` might be more troublesome since it might be called as a window (the Task window for the Circle command, maybe) is closing and cleaning up.

Rectangle, BSpline also die, but Line and Polyline work fine (the second needs a double-click to terminate it).

I don't know if this is because they have different tracker types, or they dispose of them differently, or they react to tracking movements differently, or Line and Polyline are just lucky they don't also get the exception.

The Line command *does* fail if "Show working plane tracker" is set in Preferences.

My first idea is that the commands that fail are creating (and deleting) multiple trackers but that might just be a coincidence

Roy-043, 2025-01-25:
If the Draft WB is the startup WB the Draft Tray will not be displayed for some reason. This can have an effect on the display of the grid. That is the reason for the first steps in my scenario.

You can try commenting out the calls to `finalize`. But even with one call the problem occurs.

kpemartin, 2025-01-26:
Yes, I tried completely eliminating the arc tracker in the circle command (so there would only be the radius tracker) and it still dies so perhaps the number of trackers is irrelevant after all

kpemartin, 2025-01-26:
The tracker(s) used by Line are part of Snapper and so are essentially permanent—they are not finalized at the end of each command. Instead they are only made not visible by setting `switch.whichChild` to -1.
Commands like Arc that fail make their own trackers and finalize them in their own `finish` methods.
The working-plane tracker is finalized in `GuiCommandBase.finish` which is why Line failed when the WP tracker was turned on.

For commands like Arc, the Escape key cancels the operation and calls `finalize` on the tracker without any error. The traceback at this point (`drawArc.finish`) is based at `DraftTaskPanel.reject`. This is called from C++ code `TaskDialogPython::reject()` due to a programmed click event on the `Close` button, more or less directly from the Qt event notification. Note that this does not use the Escape key handling within `drawArc.action`.

On the other hand, completing the arc with a mouse click calls `drawArc` which ends calling `self.finish` which queues up the command(s) to actually create the arc into the commit list, then calls `finalize` on the tracker. So removing the delay call in the tracker finalize means the scene is modified within the handling of a mouse event (specifically, button-down on button 1), whereas the original code modifies the scene in a QTimer singleShot callback. The python code executed to actually make the arc is also delayed, but ends up in a different delay queue and so runs after the tracker scene modification in either case.
The call stack when `tracker.finalize` is called starts at the Qt even notification, then through parts of `Coin3D`, `NavigationStyle`, and `TinkerCADNavigationStyle`, to `SofCUnifiedSelection::handleEvent`, `View3DInventorPy::eventCallback`, then into Python code starting at `drawArc.action`. Perhaps the presence of Coin in the call stack at this point is the cause of the error.

As an aside, I may be reading the code incorrectly, but it looks to me like `DraftTool.commit` appends the commands to commit to the commit list, but then `DraftTool.finalize` only actually executes the commands from the *last* entry in the commit list, then clears the list. This sounds error-prone... It seems to me that multiple calls to `commit` should cause an error, or the entire commit list should be executed, or if there in the understanding that only the last `commit` call is actually used, the comments should boldly state this and the commands should replace, rather than append to, the commit list. Or maybe I'm reading the code wrong.

Roy-043, 2025-02-10:
Sorry for the delay. `DraftTool.finish` changes the commit list. If the last item contains a recompute, that recompute is removed and `todo.ToDo.delayAfter` is used to execute it separately. This was done to solve another timing issue (it may be worth revisiting that issue as well). But the rest of the commit list is also executed.

kpemartin, 2025-02-11:
I think my last point was that if `commit` should happen to be called more than once in processing the command (without any intervening `DraftTool.finish`) only the last of these `commit` calls will do anything, since the code that schedules the delayed execution of the committed commands, including scheduling the last command in a later-executing queue if it looks like a commit command, only ever looks at the commands passed to the most recent `commit` call and immediately discards the rest.

I'm assuming that doing all the changes by queueing up Python statements is due to the need for undo/redo because it would otherwise be much simpler to just alter the object properties directly rather than building statements and passing them to the interpreter. Having just typed this, I realized that this does not provide any explanation as to how Undo works (Redo could simply replay the commands).

Roy-043, 2025-02-11:
>  only the last of these commit calls will do anything
This is not the case. You can verify this by adding a 2nd commit to any of the gui_* file. Each commit (that changes the document) will be a separate item in the undo stack.

AFAIK the main reason for the Python statements is macro recording.

kpemartin, 2025-02-12:
Well, macro recording certainly makes more sense than my explanation!

I now see what I misunderstood about the commit code. Where `DraftTool.finish` calls `todo.ToTo.delayCommit` I somehow got it into my head that this was just passing the last entry in `self.commitList`.

Roy-043, 2025-03-07:
There is a peculiar difference between `todo.ToTo.delayCommit` and `todo.ToTo.delayAfter`. The first expects a single commit list, the second appends to a list. I am not sure why that is. Since `doTasks` handles both lists it seems a bit strange.

PR #20552 (2025-04-05): Check for a null scene object
Under certain unusual circumstances getSceneGraph can be called when the scene is null, causing a native exception when the scene's reference count is incremented. This changes the code to return python None in this condition, allowing the calling code to handle the no-scene case itself.

Fixes #18679
At least it fixes the immediate cause of the issue. It does not address the whole issue of the use of the delay mechanism.
closed 2025-04-05 by kadet1090