FreeCAD/FreeCAD #18681: FreeCAD has become dysfunctional and now will not produce gcode.

Lin1Jay1, 2024-12-22:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I was doing fine with CAM in RC02 initially.  When I upgraded to 1.0.0.0 I couldn't produce any Tool Paths at all.  It just wouldn't define the base geometry.  So I uninstalled 1.0.0.0 and reinstalled RC02 and installed RC04.  In that process the Tool Library I had developed was lost.  But in the re-install the Tool Library didn't really work properly.  e.g. when you double clicked on a tool to edit the dimensions nothing happened.
I was able do a workaround and add the tool to the job - not from the TC library.  
I had a model for a sign based on a STEP file.  It wasn't quite right and I changed the design in SolidWorks and imported a new STEP.  I tried to modify the the FCStd file - deleting the old STEP file and related models.  I created new Tool Paths and was able to simulate the Tool Path.  But when I tried to Post Processs to save the gcode I get the message below.
[image]
Now even when I open a file that worked fine yesterday and try to generate gcode I get the same message using RC02 and RC04.
I don't know how to fix this.  I don't understand how doing something in a file can infect the underlying software.

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
__________________________________________________________________
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.0.0RC4.39104 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0rc4)
Hash: 83646753e051eaf438c0d5e9351cb57bd27377ab
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
```

### Subproject(s) affected?

None

### Anything else?

[image]
This is what I got yesterday when the Tool library got messed up.  I somehow was able to fix some of my tools and this went away.
[Ignorance-V Carve-3.5x14.5.zip](https://github.com/user-attachments/files/18223137/Ignorance-V.Carve-3.5x14.5.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

tarman3, 2024-12-23:
At first you select for stock existing solid. I think this is wrong.
 
https://github.com/user-attachments/assets/321ea3e9-a5a0-44f8-90e6-b0acc3f42100

I created folder with tools. I can copy this folder for backup in any time if needed.

https://github.com/user-attachments/assets/1efacfeb-3aeb-4bd4-a4de-e0dcc9a16e81

maxwxyz, 2024-12-23:
Please test on stable 1.0 and the weekly if this is still an issue

Lin1Jay1, 2024-12-23:
Thanks so much for taking the time to make these videos.  I can’t tell you how much I appreciate your help.

I will slow playback speed to 25 and carefully review.

I originally created a Jays Bits directory and it took me awhile to figure out where that was.

I added a      2mm Rad  x 1.45 x 0.25 x 3.0 Tapered Ball Mill cutter to the list  and I was disappointed it did not show up in my Tool List

 

There is a ReadMe file related to this and I’ll see if I can figure out how to create a  .fctb file.  I am surprised that a .fctb extension can yield more than one file type.  

I did manage to reinstall RC02 and re-create this sign and produce the gcode required.  

It only took about 2.5 minutes to do the carving.  The sign is for our son-in-law.  My wife will urethane and paint the letters – white I think – black she thinks.

I still need to add T-slots in the back to hang it up.

If you ever want anything carved don’t hesitate to ask.

Jay H

 

https://github.com/user-attachments/assets/321ea3e9-a5a0-44f8-90e6-b0acc3f42100

https://github.com/user-attachments/assets/1efacfeb-3aeb-4bd4-a4de-e0dcc9a16e81

tarman3, 2024-12-23:
If the desire to learn FreeCAD has not faded, I strongly recommend watching the examples on the channel [Open Source CNC](https://www.youtube.com/@opensourcecnc/videos)

Lin1Jay1, 2024-12-27:
I'm way behind commenting due to Xmas
I am now using RC02 because it has never caused a problem for me.  I tried RC04 and it seemed fine as well but I'm going to stick with RC02 for now.
I have figured out that if I close FreeCAD and reopen it I can get functionality of RC02.  I don't understand why it starts telling me about errors related to coding and becomes non-functional.
I did manage to carve 2 signs in time for Xmas which was a major objective.
[image]
[image]
On these I always use the bottom left corner for zero and I typically use 'Existing Solid Stock'.  Where I need extra stock I design it into the part and still use Existing Solid.  I am working on a fairly complicated tray right now.
[Tray-BirdsEye 12.15.24.pdf](https://github.com/user-attachments/files/18263128/Tray-BirdsEye.12.15.24.pdf)
I start with a frame made in roasted ash which has to be machined from the bottom to accept an insert in a lighter wood.  I did some test cuts and had to make the inert 0.014" smaller radially than the theoretical to get a perfect fit.  FreeCAD is using compensation and I haven't been able to find a definition for that.
[image]
The above shows where my tools are and I getting that back seems to have eliminated all the error messages I was getting.
So right now FreeCAD is working for me and it's really helped me get to where I want to be - Free of Carveco Maker.
Thanks for all the help.  I still have to go slowly thru videos provided by tarman3.  There was valuable stuff there that was totally new to me.

luzpaz, 2024-12-27:
can you zip up your toolbits directory and drag/drop them in to this github ticket ?

Lin1Jay1, 2024-12-28:
[JaysBits.zip](https://github.com/user-attachments/files/18266909/JaysBits.zip)
I have one bit that is a FreeCAD document

luzpaz, 2024-12-28:
Can any CAM folk inspect these toolbits to see if there is an anomaly with them ?

Lin1Jay1, 2024-12-28:
These tool bits are working fine now.  Problem seemed to be FC couldn't find tools I think.

closed 2024-12-28 by Lin1Jay1

Lin1Jay1, 2024-12-29:
I am subscribed to Open Source CNC but I was not aware of all these videos.

FreeCAD is working for me now and has been really worth the journey.  

Thanks for your input.

Jay H