FreeCAD/FreeCAD #18687: Command completion in Expression editor cannot automatically enumerate properties of other objects.

JDMils, 2024-12-23:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When creating an expression for a sketch or pad placement, I would like to reference the Attachment Offset value of another object however when I created the expression, the Expression Editor does not autocomplete for the property.

For example:
I have a Datum Plane called "DatumPlane_Box_Side" which is offset from the body of my model by 1mm. The property "DatumPlane_Box_Side\Attachment Offset\Position\z" is = 1 mm. I'm using the datum plane to start a new sketch which will protrude from both sides of the main body by the amount specified for "DatumPlane_Box_Side\Attachment Offset\Position\z".
I then create a new sketch called "Sketch_Tube" which is a circlular shape sitting on the Datum Plane.
I then pad the sketch "Sketch_Tube" and the length should be the width of my model plus 2 times the Datum Plane Attachment Offset value (model = 40 mm wide + 2x 1 mm=) 42mm. In order to keep the pad dimensions of the sketch "Sketch_Tube" dynamic, I'm using the Expression Editor to ensure the pad length changes with the model width changes and the Datum Plane Attachment Offset changes, thus I type in:
Sketch.Constraints.Box_Length + (<<DatumPlane_Box_Side>>.Att

....and nothing shows. The Expression Editor does not suggest the Attachmentxxxx properties. So if I know the property I need and type that in like this:
Sketch.Constraints.Box_Length + (<<DatumPlane_Box_Side>>.AttachmentOffset.Base.z) *2

....it shows the correct dimension being 42mm.

From what I'm told, the Expression Editor does not automatically suggest the following properties of another object:
`AttachmentEngine
AttachmentType
AttachmentOffset
AttachmentSupport
ExpressionEngine
ExternalGeo
MapMode
MapPathParameter
MapReversed`

I would like to suggest we add these properties in to the Auto-Suggest feature please.

### Full version info

```shell
OS: Windows 11 build 22631
Architecture: x86_64
Version: 1.1.0dev.39560 (Git) Conda
Build type: Release
Branch: main
Hash: 1c35d08d87bb6dec7cb5adf51b2ed460ce43096d
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/Australia (en_AU)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Qt default
Installed mods:
* A2plus 0.4.68
* Assembly4 0.50.15
* Curves 0.6.52
* fasteners 0.5.33
* Gridfinity 0.6.0
```

### Subproject(s) affected?

None

### Anything else?

My references and test model:
[https://forum.freecad.org/viewtopic.php?p=799058&sid=616dec9cf3730032f64b7b31edbb10f2#p799058](url)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

pitch-circle, 2024-12-23:
Please could you say how, in principle, this is different to issue #14947 which you have already raised.

The problem in both of these is that autocomplete of Expressions does not currently work for everything.

What you say above is a further example of the general problem.

If you take the time to search through all open Expressions-related issues, you will likely find other past examples of this being raised.

It might be useful to add your comments to your own past issue or to one of the others rather than start a whole new one. That way, anyone who takes on the task of fixing the problem will see everything relevant in one place.

JDMils, 2024-12-27:
For issue https://github.com/FreeCAD/FreeCAD/issues/14947, my goal here is for FreeCAD to allow the user to type an expression and for FC to provide relevant properties for the user to choose. At the moment, when the user selects to create an expression, FC shows the current dimension of the object being constrained. The user types "Sketch" and selects the current sketch from the list of available matching items and FC then shows the following "Sketch." with the "." provided, however no available properties for the Sketch object.

In nearly all programming environments, when the user for example types the object name "Sketch" they signal to the editor to provide all relevant properties by typing the "." which then immediately shows all relevant properties.

To do the same in FC, the user has to press SPACEBAR then BACKSPACE and the editor shows all relevant properties. Thus, in 14947, I'm asking that the expression editor does NOT add the "." after the user selects an object then when the user types "." show all relevant properties.

Better still, when the user select an object such as "Sketch", let the expression editor add the "." automatically like it currently does now, however provide all relevant properties for the selected object automatically for the user to choose from.

------------------------------

In Issue 18687, the Expression editor does NOT show all relevant properties no matter what, the user must already know the property name in order to select it to be part of the expression and if the user does not know the property name then the user cannot pick the property they need in order to complete their expression. As shown above, Expression editor does NOT show the property "AttachmentOffset" for the selected object and it was with the help of a forum user that I was able to find and use the property name to complete what I was trying to achieve.

What I'm asking in this issue is for the Expression editor to should show ALL relevant properties of the selected object.

The 2 issues are 2 completely different issues.

FEA-eng, 2025-09-20:
For me, enter doesn't trigger autocompletion when `Sketch.` is added in the expression editor. I have to actually start typing something. Perhaps it's more of a feature request though.

JDMils, 2025-09-21:
I feel that it is important to reveal those specific properties which are CURRENTLY not viewable in the Expression Editor so that they can be used in custom expressions. I've tested Windows FC 43071 and the same issue exists.

FEA-eng, 2026-10-25:
> From what I'm told, the Expression Editor does not automatically suggest the following properties of another object:
> `AttachmentEngine AttachmentType AttachmentOffset AttachmentSupport ExpressionEngine ExternalGeo MapMode MapPathParameter MapReversed`

Seems still relevant

```
OS: Windows 11 build 26300
Architecture: x86_64
Version: 27.1.0dev.20261020 (Git shallow)
Build date: 2026/10/20 22:15:52
Build type: Release
Branch: (HEAD detached at 9946038aa)
Hash: 9946038aace97af288eb28f2727a169f9d1158c7
Python 3.13.15, Qt 6.11.2, Coin 4.0.10 (bundled), Pivy 0.6.11 (bundled), Vtk 9.7.0, boost 1_90, Eigen3 5.0.1, PySide 6.11.2
shiboken 6.11.2, xerces-c 3.3.0, Clipper2 2.0.1, IfcOpenShell 0.8.5, OCC 8.0.1
Locale: Polish/Poland (pl_PL)
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
QPA/File dialog/Color dialog: windows/via Win32/via Qt
Navigation Style/Orbit Style/Rotation Mode: Touchpad/Rounded Arcball/Window center
Logical DPI/Physical DPI/Pixel Ratio: 96/113.268/1.25
OpenGL version/vendor/renderer: 4.6.0 - Build 27.20.100.9664/Intel/Intel(R) UHD Graphics 620
```

maxwxyz, 2026-11-07:
There seems to be a fix in AstoCAD: https://github.com/AstoCAD/FreeCAD/commit/6cdd711b8bd0fe0dbdd7ce9c19128698ed2c70d5

The expression completer includes properties from extensions, such as AttachmentOffset and MapMode. The separate Enter-key completion behavior discussed above is not changed.