FreeCAD/FreeCAD #18690: CAM: Toolholder on machine simulation

AdrianCM95, 2024-12-23:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

CAM workbench to include toolholder on the simulation and tool library. See also this issue I created on BTL that describes the implementation of the library in detail: 
https://github.com/knipknap/better-tool-library/issues/44#issuecomment-2494512000

Target is to check collisions and have a more complete simulation tool.

On the implementation side, other softwares like Bobcad Cam, use a STL model created by a form where basic shapes (cylinders and cones) can be stacked one on top of each other to form the holder geometry. Then, this file is positioned relative to the tool tip by an offset (tool stick-out).

### Full version info

```shell
OS: TUXEDO OS 3 (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda AppImage
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: Spanish/Spain (es_ES)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Qt default
```

### Subproject(s) affected?

CAM

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Connor9220, 2024-12-25:
While the current tool library doesn't directly support this, as an experiment, I took my ISO30 model and added it to one of the tool bit shapes.  I added a StickOut Attribute, and this is the result.

[image]

Would be nice if the StickOut attribute was on the main page, instead of under the attributes, since it would be a very import one to include with ALL endmills.

f you give a fairly accurate stick out length, you can see if your going to have any issues with deep pocket milling.  Because, Overall length, doesn't give you the full picture.

luzpaz, 2024-12-25:
cc @shaise

Connor9220, 2024-12-25:
[image]

https://github.com/user-attachments/assets/4f3c4ab0-a61b-439e-b068-fe9036eb1970

Connor9220, 2024-12-25:
One Additional Note:  The OLD Cam simulator really struggles with this.  I'm guessing it's the more complicated shape, or addition of another shape.  My CPU throttles really hard with this.

AdrianCM95, 2024-12-26:
In order to extend the concept, there geometry used in the simulation could be considered "useful" or "cosmetic" for collision detection and "visible" or "hidden" just to display it.

This can be also applied to the machine geometry and fixtures.

That way resources can be saved.

shaise, 2024-12-26:
Basically, from the simulator perspective it does not care if the tool is bare or comes along with the holder. It is just a shape, so i thgink the CAM WB should supply the simulator with both shapes, and a switch on the cam can select either one. 
Bare in mind, that because of performance issues, the simulator might show artifacts when the tool shape have some concave features in it. but if it is just for collision detection, I would say its good enough.

Connor9220, 2024-12-26:
> Basically, from the simulator perspective it does not care if the tool is bare or comes along with the holder. It is just a shape, so i think the CAM WB should supply the simulator with both shapes, and a switch on the cam can select either one. Bare in mind, that because of performance issues, the simulator might show artifacts when the tool shape have some concave features in it. but if it is just for collision detection, I would say its good enough.

If we could have more contrast or pause the sim if it detects that more material was removed than what was expected, that would would work.  Here is a sim of a deep pocket milling with collection.  You can see at the end the collision.

https://github.com/user-attachments/assets/221347f9-57db-4d07-a803-8adebebd39fa

shaise, 2024-12-26:
It is not easy to programmatically compare between the end object and the cut one, however if I get the tool and the holder as separate objects, I can shade the cuts made by the tool holder as bright red, and this will immediately pop up to the eye.

Connor9220, 2024-12-26:
> It is not easy to programmatically compare between the end object and the cut one, however if I get the tool and the holder as separate objects, I can shade the cuts made by the tool holder as bright red, and this will immediately pop up to the eye.

I figured as much.  Based on my experiments, it will require a code change in other parts.  I tried separating the holder and tool into two separate bodies.  FreeCAD only displayed the first body (which was the tool). In practice, it would be nice to be able to list the tool holders in the tool bit editor, and select the tool holder for each tool. Have a directory for the tool holders, similar to what we have for the tool bit Shapes.