FreeCAD/FreeCAD #18691: 3DConnexion SpaceMouse jumps from a 2D viewpoint into 3D

karamellpelle, 2024-12-23:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

A 3DConnexion SpaceMouse makes the interaction with a 3D CAD better and more efficient, it's easy to move around in the world. But a SpaceMouse is also very efficient for moving and zooming in a 2D world, and together with your regular mouse in your other hand, this makes a good workflow when editing 2D.

But if I edit a sketch in FreeCAD, using the SpaceMouse "jumps out of 2D" and now the sketch is in 3D world. The ability to edit 2D sketches in 3D is important, but I would like to lock my SpaceMouse to 2D when I _start_ editing sketches in 2D view. And when I edit the sketch in 3D, the SpaceMouse works as usual. 

## Suggestion
* When jumping into "2D edit mode" (for example by _Edit sketch_ or _View sketch_), lock SpaceMouse to 2D (move and zoom, not rotate).
* Let SpaceMouse work as usual when editing the sketch in 3D.
* Make it possible to jump out of 2D and into 3D (i.e. into non-orthographic) while editing. In Blender you can achieve this by middle clicking your regular mouse.
* Create a Preference setting: "Lock SpaceMouse in 2D ☑️"

Since the SpaceMouse has buttons, you could even configure one of them to toggle 2D/3D while editing sketches.

### Full version info

```shell
OS: Windows 11 build 22631
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/Fusion
Installed mods: 
  * fasteners 0.5.33
  * freecad.gears 1.3.0
  * IconThemes
  * OpenTheme 2024.9.1
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-12-23:
Sometimes you'd need to orbit to select external geometry. Is the space mouse navigation in 2D possible in TechDraw or Spreadsheet? 
@FreeCAD/design-working-group  FYI

karamellpelle, 2024-12-25:
Hi! I did a test, and SpaceMouse navigation worked in _TechDraw_ (move + zoom) but not in _Spreadsheet_ . 

Yes, it's necessary to orbit around in 3D while doing 2D-editing of sketches, so the SpaceMouse should be able to do this. But it's very handy how the SpaceMouse works in Blender when the view is aligned to one of the X, Y or Z planes. There you can use it to efficiently move around in your sketch in 2D. And you can jump out and use your SpaceMouse in 3D too.

----

I have just started using FreeCAD and am very excited about the project!