FreeCAD/FreeCAD #18696: [Issue] sketches print as white lines

evan-king, 2024-12-23:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When printing/print previewing a view of a sketch, all non-background colors are preserved, including the default white unconstrained geometry...for ink.  Resulting print is, predictably, unusable.

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * CurvedShapes 1.0.5
  * DynamicData 2.53.0
  * lattice2 1.0.0
```

### Subproject(s) affected?

Sketcher

### Anything else?

While not quite a bug per se, I do think this is a legitimate use-case - it's pretty handy for quickly mocking up 2d shapes/assemblies and then printing what's essentially a layout guide.  I feel it would be appropriate for sketcher specifically to specify separate print colors, or at least automatically invert colors fairly near to white.

Workarounds:
 - switch the sketch lines from default white to black (a detriment to clarity/contrast for in-program view, so I do it every time)
 - fully constrain the sketch to get light green lines (also not an ideal color, nor necessarily prudent/worth the effort)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-12-23:
Could you add screenshots and a demo file. How do you print or do you export an print?

evan-king, 2024-12-23:
While viewing the sketch (i.e. in sketch edit mode) open print preview (File->Print preview...) or print directly (CTRL+P or File->Print...).  Results are as shown, and preview matches printed output (aside from aliasing in the preview).

[image]

[print-demo.zip](https://github.com/user-attachments/files/18234035/print-demo.zip)

luzpaz, 2024-12-25:
Confirmed  
[image]
```
OS: Manjaro Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.39741 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: 0d128517dedd154bbecfbabe67094ad56c8d46f5
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
```

maxwxyz, 2024-12-25:
Was this different than before? I'd not consider it a blocker if we have released 1.0 as is and might consider it a feature request.

mwganson, 2024-12-27:
Here is a macro to toggle the edge color to black and when you run it again it toggles back to the previous color.  Note that the current color is saved only for the current session, so if you save the changes and restart FreeCAD the old color is now lost.

```
pg = FreeCAD.ParamGet("User parameter:BaseApp/Preferences/View")
white = 0xffffffff
black = 0x000000ff

current = pg.GetUnsigned("EditedEdgeColor", white)

if not hasattr(FreeCAD, "oldEdgeColor"):
    if current != black:
        FreeCAD.oldEdgeColor = current

if current != black:
    pg.SetUnsigned("EditedEdgeColor", black)
else:
    pg.SetUnsigned("EditedEdgeColor", FreeCAD.oldEdgeColor)
    del FreeCAD.oldEdgeColor
```
Usage: create a new macro in the Macro -> Macros dialog, name it something like SketchColorToggler.  The macro should be run while the Sketch editor is open.  Run it again to reset back to the original edge color after printing / print previewing.

maxwxyz, 2024-12-27:
@kadet1090 I guess this now an issue as previously the sketch colors did not change automatically on dark themes.

xtemp09, 2025-08-31:
@luzpaz, what is the solution for this problem? Are you suggesting that the white color must be change from white to black before printing and switched back after?

luzpaz, 2025-08-31:
@xtemp09 if we do that then letting the user know somehow that this is not the current color representation ?

FEA-eng, 2026-09-29:
Still relevant

[image]

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 26.3.0dev.20260922 (Git shallow)
Build date: 2026/09/22 20:26:30
Build type: Release
Branch: (HEAD detached at 019f5c50a)
Hash: 019f5c50a945743b89cabf422ac5f5f1e297d890
Python 3.13.15, Qt 6.11.2, Coin 4.0.10 (bundled), Pivy 0.6.11 (bundled), Vtk 9.7.0, boost 1_90, Eigen3 5.0.1, PySide 6.11.2
shiboken 6.11.2, xerces-c 3.3.0, Clipper2 , OCC 8.0.1
Locale: Polish/Poland (pl_PL)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/
QPA/File dialog/Color dialog: windows/via Win32/via Qt
Navigation Style/Orbit Style/Rotation Mode: Blender/Rounded Arcball/Window center
Logical DPI/Physical DPI/Pixel Ratio: 96/90.0633/1
OpenGL version/vendor/renderer: 4.6.0 NVIDIA 474.82/NVIDIA Corporation/Quadro K4000/PCIe/SSE2
```

luzpaz, 2026-10-04:
Further discussion needed here.

evan-king, 2026-10-11:
I would love to see print styles excluded from theming/treated as a separate theme.  Dark themes are never going to have screen colors appropriate for printing in subtractive color space.  So conceptually, there could either be:
  - UI/config/rendering paths just for managing print colors specially
  - additional print-only details that the theming system would have to adopt and all themes opt into overriding to the extent consistency would be appropriate
  - or simply a different theme applied to the printing output rendered, that by default is always light.

As a user I'd find any of these options acceptable, with perhaps option two being most appealing in theory.  As a user likely to tweak things just to my exact liking, option 1 sounds more reasonable (and the least effort to maintain) and option 3 still quite a bit more manageable (especially if any color configuration that spans themes is split into light and dark mode variants).

As a developer, I'd probably pick option three as the most value for development effort.  That is, however, without any insight into the complexities specific to this project/codebase.

luzpaz, 2026-10-13:
Thanks for the feedback @evan-king. 
@maxwxyz this needs further discussion by the UI/UX group