FreeCAD/FreeCAD #18698: 'ascii' codec can't encode character u'\...'

Doriangaensslen, 2024-12-23:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I've Stumbled upon a new bug on iOS today

Reproduce: 

1. Create a CAM job
2. Create a Tool that has special Characters (Example: see screenshot with a 
[Testfile.zip](https://github.com/user-attachments/files/18232053/Testfile.zip)
"ä" in its name) 
3. choose GRBL as processor
4. chose an action, where the tool with the special character is used (in the Screenshot) Pocket_Shape).
5. Click "Post Process"

[image]

I guess the following error with "xe4" is ISO-8859-1 (european languages), in which case it's "ä"

```18:48:10  PostProcessor:  script_module postprocessing...
18:48:12  done postprocessing.
18:48:18  Running the Python command 'CAM_Post' failed:
Traceback (most recent call last):
  File "/Applications/FreeCAD.app/Contents/Resources/Mod/CAM/Path/Post/Command.py", line 207, in Activated
    self._write_file(fname, gcode, policy)
  File "/Applications/FreeCAD.app/Contents/Resources/Mod/CAM/Path/Post/Command.py", line 135, in _write_file
    f.write(gcode)

'ascii' codec can't encode character '\xe4' in position 275: ordinal not in range(128)```

### Full version info

```shell
OS: macOS 15.0.1
Architecture: arm64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * btl 0.9.9
```

### Subproject(s) affected?

CAM

### Anything else?

I found a very old Problem, which led me to the solution:
https://forum.freecad.org/viewtopic.php?style=3&p=799146#p799146

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Doriangaensslen, 2024-12-23:
I tested with another File and another Tool which had also "ä" in its name on Freecad 1.0 on my PopOs, but this seems to work. So i think its just MacOS and Windows related.

luzpaz, 2024-12-23:
> I've Stumbled upon a new bug on iOS today  

I believe you meant macOS here

luzpaz, 2024-12-23:
Any chance you can test on Weekly ? (https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds)

Doriangaensslen, 2024-12-24:
> > I've Stumbled upon a new bug on iOS today
> 
> I believe you meant macOS here

Yes absolutetly true.

> Any chance you can test on Weekly ? (https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds)

I tried that build two times, and could not install (it told me that the application is damaged). However i had no problem with the 1.0 Release

Doriangaensslen, 2024-12-24:
Somehow, the Fix posted here: https://forum.freecad.org/viewtopic.php?p=799303#p799303 did not work for me.

luzpaz, 2024-12-24:
> "FreeCAD.app" is damaged and can't be opened.

What version did you try to install? Did you just run in to the code-signing issue where you need to employ the right click context menu to open it ? Or is the weekly macOS build broken ?

JohnOCFII, 2024-12-24:
> > "FreeCAD.app" is damaged and can't be opened.
> 
> What version did you try to install? Did you just run in to the code-signing issue where you need to employ the right click context menu to open it ? Or is the weekly macOS build broken ?

@Doriangaensslen 

On macOS arm machines, there is a separate installation issue.  Until the new installation setup is resolved, to test the weekly builds, the user needs to open a Terminal window and execute the following command:

`sudo xattr -r -d com.apple.quarantine /Applications/FreeCAD.app`

Doriangaensslen, 2024-12-29:
> `sudo xattr -r -d com.apple.quarantine /Applications/FreeCAD.app`

Thanks, i was able to open the weekly build [FreeCAD_weekly-builds-39747-conda-macOS-arm64-py311.dmg](https://github.com/FreeCAD/FreeCAD-Bundle/releases/download/weekly-builds/FreeCAD_weekly-builds-39747-conda-macOS-arm64-py311.dmg)

But sadly, the issue still persists if il add a tool with special characters.

Syres916, 2025-01-01:
@Doriangaensslen could you please modify your local copy of `src/Mod/CAM/Path/Post/Command.py` as per https://github.com/FreeCAD/FreeCAD/compare/main...Syres916:FreeCAD:CAM_Fix_Post_Write_File_Using_Codecs
If it works as I believe it should, I can see if an additional test can be made for non-ascii characters, I've done the same for Techdraw in the last few days.

Doriangaensslen, 2025-01-05:
I can confirm, that this fixes this specific issue! Good work!

LarryWoestman, 2025-05-31:
Yes, this should be fixed by #20833.

closed 2025-06-01 by luzpaz