FreeCAD/FreeCAD #18701: Can`t setup a line size after applying the "Constrain symmetric" for line and axis

uaboss, 2024-12-23:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Ubuntu 24.04.1 LTS
I check this for freecad v 1.0 - either Snap, and AppImage packages.
To reproduce:
- Create a new Parametric part.
- Create a body
- Create a sketch
- Draw vertical line trough X axis
- Select two end-line points and X axis
- Click on "Constrain symmetric" button
- Select line
- Click on "Dimension" button
- Set lemgth to 25 mm (or any other size) and press "Ok"

You will get error:
<class 'ValueError'>: Datum 25 mm for the constraint with index 9 is invalid

(please check Traceback log in "Full version info field")

### Full version info

```shell
21:33:04  pyException: Traceback (most recent call last):
  File "<string>", line 1, in <module>
<class 'ValueError'>: Datum 29.99 mm for the constraint with index 9 is invalid
21:33:04  Unnamed#Sketch: App.getDocument('Unnamed').getObject('Sketch').setDatum(9,App.Units.Quantity('29.990000 mm'))
21:33:40  pyException: Traceback (most recent call last):
  File "<string>", line 1, in <module>
<class 'ValueError'>: Datum 18.98 mm for the constraint with index 9 is invalid
21:33:40  Unnamed#Sketch: App.getDocument('Unnamed').getObject('Sketch').setDatum(9,App.Units.Quantity('18.980000 mm'))
21:34:35  pyException: Traceback (most recent call last):
  File "<string>", line 1, in <module>
<class 'ValueError'>: Datum 68.26 mm for the constraint with index 9 is invalid
21:34:35  Unnamed#Sketch: App.getDocument('Unnamed').getObject('Sketch').setDatum(9,App.Units.Quantity('68.260000 mm'))
21:35:11  pyException: Traceback (most recent call last):
  File "<string>", line 1, in <module>
<class 'ValueError'>: Datum 68.26 mm for the constraint with index 9 is invalid
21:35:11  Unnamed#Sketch: App.getDocument('Unnamed').getObject('Sketch').setDatum(9,App.Units.Quantity('68.260000 mm'))
21:35:15  Sketch: Sketch with redundant constraints
Please remove the following redundant constraint:
6

21:40:55  pyException: Traceback (most recent call last):
  File "<string>", line 1, in <module>
<class 'ValueError'>: Datum 29.70 mm for the constraint with index 2 is invalid
21:40:55  Unnamed#Sketch: App.getDocument('Unnamed').getObject('Sketch').setDatum(2,App.Units.Quantity('29.700000 mm'))
```

### Subproject(s) affected?

Other (specify in description)

### Anything else?

[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

tarman3, 2024-12-23:
Before set dimension constraint you need to remove redundant constraints.
In this case need to remove vertical constraint.

uaboss, 2024-12-23:
Thank you Tarman3 it is works now.

uaboss, 2024-12-23:
Just need to remove redundant constrains.
Not the issue.
My bad.

closed 2024-12-23 by luzpaz