FreeCAD/FreeCAD #18702: FreeCAD menus do not appear until after clicking into the FreeCAD application window

JohnOCFII, 2024-12-23:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When launching FreeCAD on macOS, the application menus (File, Edit, etc.) do not show up until after the application window is activated by clicking into it.  Very minor issue, but isn't the way a typical macOS app performs.

Problem is also present in 1.0.0 released version.

### Full version info

```shell
OS: macOS 15.2
Architecture: arm64
Version: 1.1.0dev.39708 (Git) Conda
Build type: Release
Branch: main
Hash: 173ff4da1ef4f075d56c0dc617d1bf00d5f49d4b
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/Fusion
Installed mods: 
  * FeedsAndSpeeds 0.6.0
  * OpenTheme 2024.9.1
  * fasteners 0.5.33
  * Curves 0.6.52
```
```

### Subproject(s) affected?

None

### Anything else?

<img width="1728" alt="FreeCAD - no menus" src="https://github.com/user-attachments/assets/12b609d5-247a-44c8-9dd3-29f23d52af3b" />

<img width="1728" alt="FreeCAD menus shown properly" src="https://github.com/user-attachments/assets/fad3593e-cd5d-4a9f-8a35-92fc29e13004" />

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-12-29:
I cannot confirm this issue, it is there without the need to interact with the UI or click somewhere, weekly and stable
```
OS: macOS 13.7.2
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * freecad.gears 1.2.0
  * OpenTheme 2024.8.17
  * CfdOF 1.27.5
  * Defeaturing 1.2.2
  * Manipulator 1.5.7
  * fasteners 0.5.27
  * CurvedShapes 1.0.9
  * sheetmetal 0.4.24
  * Curves 0.6.45

```

JohnOCFII, 2024-12-29:
Sadly, I've found it to be quite variable.  A few days ago I completed removed my user.cfg, system.cfg and all the other related support files for FreeCAD (preference packs, macros, etc.).  On the first few launches of FreeCAD, I did not see the issue.  Later, as I added changes in `Preferences` it started to show up again.  I thought "AH HA!" it is related to OpenThemes... but later, after once again removing all FreeCAD support files and NOT loading OpenThemes, the issue showed up again -- randomly it seems.

Currently, I'm not experiencing the issue.  Maybe leave it open for a bit to see if anyone else comes along with ideas? Clearly it isn't a critical issue.

```
OS: macOS Sequoia (15.2)
Architecture: arm64
Version: 1.1.0dev.39747 (Git) Conda
Build type: Release
Branch: main
Hash: 7083362acb8529017b58ea03bf7416da0192b25b
Python 3.12.0, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/
Installed mods: 
  * OpenTheme 2024.9.1
```

hyarion, 2024-12-30:
I saw this issue when start page was hogging all cpu in RCs. This makes me believe that it is a raise condition somewhere.

@JohnOCFII Can you try doing something cpu intensive and see if it happens more often than if the system is idling?

dhoepfl, 2025-03-20:
I saw the same problem when I did some tests for #20303. I can reproduce it on the very first start of FreeCAD (1.0) as follows:

* Create a new macOS user (skipping everything you can, everything on default)
* Start FreeCAD
* Use default units
* Create a new document
* Switch to workbench “Part”.
  * Menu disappeared 
* Unrelated steps:
  * Create sphere
  * Switch to Model Tree
  * Select sphere
* Switch workbench to “<none>”
  * Toolbar disappeared 

Here’s a recording of what I did:

https://github.com/user-attachments/assets/87bdd9a6-a97c-4718-9038-acb045ebab52

cloudlesssmarthome, 2025-04-07:
also getting random no menus on macOS 15.4 and FreeCAD 1.0

JohnOCFII, 2025-04-09:
This is so random!  But it still happens occasionally.  While investigating the macOS 15.4 and `numpy` issue, I even saw the problem occur in FreeCAD 0.21.2!  Some more screen shots and data from the current weekly is below.  

```
OS: macOS Sequoia (15.4)
Architecture: arm64
Version: 1.1.0dev.41138 (Git) Conda
Build date: 2025/04/04 17:31:11
Build type: Release
Branch: main
Hash: cab9053be825b8f09d686d5894f333915e4ee378
Python 3.11.11, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, IfcOpenShell 0.8.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/
Logical DPI/Physical DPI/Pixel Ratio: 72/127.5/2
Installed mods: 
  * OpenTheme 2024.9.1
  * Telemetry 1.0.0
```

Just after launch without touching any keys:

[image]

and

[image]

After a keypress with FreeCAD as the active App:

[image]

JohnOCFII, 2025-04-09:
> Then I discovered if I just ignore it for a second, switch to finder, and switch back to FreeCad, the menus appear.

Yes, that certainly works.  It seems that any input to the FreeCAD App (switching into it from another App or the Finder, or pressing a key in the FreeCAD window) is enough to re-draw the menus.

hyarion, 2025-04-09:
I've seen this happen, but it's easy to workaround and difficult to reproduce.

It feels like a race condition - maybe the program hasn't loaded "enough" before we try to setup the menu, causing it to load a default empty menu bar.

JohnOCFII, 2025-04-17:
I haven't seen this for a while but it did just pop up again.  Super low priority, but just an odd curiosity.  

[image]

hsvfan-jan, 2025-05-06:
Still happening with
```
OS: macOS 15.4.1
Architecture: arm64
Version: 1.1.0dev.41585 (Git)
Build date: 2025/05/06 03:23:37
Build type: Release
Branch: main
Hash: 261ce705a93e98826389f20c4a99238757dfaab3
Python 3.11.12, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 
shiboken , xerces-c 3.3.0, IfcOpenShell 0.8.0, OCC 7.8.1
Locale: English/Germany (en_DE)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 72/127.5/2
```

luzpaz, 2025-05-14:
> It feels like a race condition - maybe the program hasn't loaded "enough" before we try to setup the menu, causing it to load a default empty menu bar.

Is this a Qt issue or something we can hack a `ifdef` fix for macOS ?