FreeCAD/FreeCAD #18703: View pane is not getting updated after moving a point programaticall in part design

ekkazan, 2024-12-23:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

View pane is not getting updated after moving a point programaticall in part design. A video is attached. Video is a bit long but you can see that when I put a single dot and change its visibility, it appears on the moved position. I think that its properties are gettin updated but not applying.

https://github.com/user-attachments/assets/914ff33d-5c0c-48b5-a976-e8a77e50f6a5

### Full version info

```shell
OS: Windows 11 build 22631
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: Turkish/Turkey (tr_TR)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Syres916, 2024-12-23:
After any geometry/constraint changes I would expect to execute:

`App.activeDocument().recompute(None,True,True)`

which will also update the screen as well as the solver.

ekkazan, 2024-12-24:
> After any geometry/constraint changes I would expect to execute:
> 
> `App.activeDocument().recompute(None,True,True)`
> 
> which will also update the screen as well as the solver.

It works but still it takes a time, before 1.0 this was not necessary and it was very fast. I am not sure if this is really a bug but it was a thing that I am using much.

FEA-eng, 2026-02-13:
Still relevant

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.2.0dev.20260210 (Git shallow)
Build date: 2026/02/10 18:10:36
Build type: Release
Branch: (HEAD detached at ecbe21ca0)
Hash: ecbe21ca03a59723c90ce5a677ae7ccea06ebc12
Python 3.11.14, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Navigation Style/Orbit Style/Rotation Mode: CAD/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/90.0633/1
```

FEA-eng, 2026-08-16:
Still relevant

```
OS: Windows 11 build 26200
Architecture: x86_64
Version: 26.3.0dev.20260812 (Git shallow)
Build date: 2026/08/12 05:41:29
Build type: Release
Branch: (HEAD detached at dc1ef6e88)
Hash: dc1ef6e880d6f5cf5d5a6eb3749373141eb0deb8
Python 3.11.14, Qt 6.8.3, Coin 4.0.10 (bundled), Pivy 0.6.11 (bundled), Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, Clipper2 , IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL) [ OS: Polish/Poland (pl_PL) ]
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
QPA/File dialog/Color dialog: windows/via Win32/via Qt
Navigation Style/Orbit Style/Rotation Mode: Blender/Trackball/Window center
Logical DPI/Physical DPI/Pixel Ratio: 96/81.5973/1
OpenGL version/vendor/renderer: 4.6.0 NVIDIA 610.88/NVIDIA Corporation/NVIDIA GeForce RTX 5080/PCIe/SSE2
```

maxwxyz, 2026-11-07:
There seems to be a fix in AstoCAD: https://github.com/AstoCAD/FreeCAD/commit/26fc278d563f4e10d24822cf1238bb92a3ec01f0

A successful moveGeometry call notifies the sketch view to redraw. Could be picked up for a PR.