FreeCAD/FreeCAD #18705: Assembly Crashes on Creation of Joint

johnamon, 2024-12-24:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When applying multiple basic joints in the Assembly workbench, FreeCAD crashes.  This behaviour is repeated across Mac OS and Linux

### Full version info

```shell
Here is the crash report

-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               freecad [74284]
Path:                  /Applications/FreeCAD.app/Contents/Resources/bin/freecad
Identifier:            freecad
Version:               ???
Code Type:             X86-64 (Native)
Parent Process:        launchd [1]
User ID:               501

Date/Time:             2024-12-24 07:57:31.0478 +0000
OS Version:            macOS 15.1.1 (24B91)
Report Version:        12
Bridge OS Version:     9.1 (22P1072)
Anonymous UUID:        37BA82BD-2B08-81D3-F9B8-DB511FCC738B

Sleep/Wake UUID:       37B07D07-DDA2-42C3-8AAC-B634462B4898

Time Awake Since Boot: 1500000 seconds
Time Since Wake:       811053 seconds

System Integrity Protection: enabled

Crashed Thread:        0

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000058
Exception Codes:       0x0000000000000001, 0x0000000000000058

Termination Reason:    Namespace SIGNAL, Code 11 Segmentation fault: 11
Terminating Process:   exc handler [74284]

VM Region Info: 0x58 is not in any region.  Bytes before following region: 4350615464
      REGION TYPE                    START - END         [ VSIZE] PRT/MAX SHRMOD  REGION DETAIL
      UNUSED SPACE AT START
--->  
      __TEXT                      103512000-103532000    [  128K] r-x/r-x SM=COW  /Applications/FreeCAD.app/Contents/Resources/bin/freecad

Kernel Triage:
VM - (arg = 0x3) mach_vm_allocate_kernel failed within call to vm_map_enter

Thread 0 Crashed:
0   libOndselSolver.1.0.1.dylib   	       0x1627e3146 MbD::Part::qX(std::__1::shared_ptr<MbD::FullColumn<double>>) + 6
1   libOndselSolver.1.0.1.dylib   	       0x16271ed25 MbD::ASMTSpatialContainer::updateMbDFromPosition3D(std::__1::shared_ptr<MbD::FullColumn<double>>) + 325
2   AssemblyApp.so                	       0x11fadb568 Assembly::AssemblyObject::doDragStep() + 360
3   AssemblyGui.so                	       0x1623b316f AssemblyGui::ViewProviderAssembly::tryMouseMove(SbVec2s const&, Gui::View3DInventorViewer*) + 2447
4   AssemblyGui.so                	       0x1623b274e AssemblyGui::ViewProviderAssembly::mouseMove(SbVec2s const&, Gui::View3DInventorViewer*) + 30
5   libFreeCADGui.dylib           	       0x105e44e9a Gui::ViewProvider::eventCallback(void*, SoEventCallback*) + 938
6   libCoin.80.0.3.dylib          	       0x108bc6ce5 SoEventCallback::handleEvent(SoHandleEventAction*) + 197
7   libCoin.80.0.3.dylib          	       0x1089b2489 SoAction::traverse(SoNode*) + 105
8   libCoin.80.0.3.dylib          	       0x108b7d293 SoChildList::traverse(SoAction*, int, int) + 131
9   libCoin.80.0.3.dylib          	       0x108bd0b75 SoGroup::doAction(SoAction*) + 101
10  libCoin.80.0.3.dylib          	       0x108bec658 SoSeparator::handleEvent(SoHandleEventAction*) + 40
11  libFreeCADGui.dylib           	       0x105d267f5 Gui::SoFCUnifiedSelection::handleEvent(SoHandleEventAction*) + 645
12  libCoin.80.0.3.dylib          	       0x1089b2489 SoAction::traverse(SoNode*) + 105
[... 332 lines cut ...]
__DATA_CONST                      61.1M      630 
__DATA_DIRTY                      1482K      209 
__FONT_DATA                        2352        1 
__GLSLBUILTINS                    5174K        1 
__LINKEDIT                       238.4M      166 
__OBJC_RO                         76.3M        1 
__OBJC_RW                         2355K        3 
__TEXT                           912.3M      805 
__TPRO_CONST                       272K        2 
dyld private memory                640K        2 
mapped file                      259.1M       40 
owned unmapped memory              404K        1 
shared memory                      796K       20 
===========                     =======  ======= 
TOTAL                              2.9G     3036 
TOTAL, minus reserved VM space     2.4G     3036
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

tarman3, 2024-12-24:
Please attach file with simple example and write steps to reproduce

luzpaz, 2025-01-14:
More info needed:  
- [ ] full About info for v1.0 and v1.1
- [ ] example file
- [ ] steps to reproduce

brad2014, 2025-02-26:
Reproducible crash after creating Joint
Test file and crash report: [github-FreeCAD-FreeCAD-issues-18705.zip (47k)](https://wbr.tech/share/github-FreeCAD-FreeCAD-issues-18705.zip)

FreeCad Version: 
```
OS: macOS 14.5
Architecture: arm64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
```

To reproduce:

1. Open test1.FCstd (in above zip file)
2. Make "Assembly" active object
3. Create Distance Joint between LegLeft.Edge2 and Brace.Edge1, distance 0.
4. Zoom out slightly
5. Click on test1.Assembly.Body001.Pad001.Face6 (Brace.Face6)
6. Zoom out slightly
7. Click on test1.Assembly.Body001.Pad001.Face6 (Brace.Face6)
8. Crash (test1.crash.txt in above zip file)

brad2014, 2025-02-26:
The following steps from a fresh FreeCAD instance reproduce the crash:

1. Start FreeCAD, Open new file.
2. Create a Body, sketch in XY a 50mm x 50mm rectangle anchored at origin, pad to 20mm.
3. Create a Body001, sketch in XY a 30mm x 30xx rectangle, pad to 20mm.
4. In Assembly workbench, create an assembly, move Body and Body001 to Assembly, position Body to left of Body001.
5. Ground Body (select Body, click on Toggle Grounded.)
6. Create a distance joint of 0 distance between Body.Pad.Edge5 and Body001.Pad.Edge1.
7. Click-drag (to move) Body001.
8. Crash

```
OS: macOS 14.5
Architecture: arm64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
```

linguini1, 2025-04-09:
Could not quite elicit a crash, but I was able to severely stall the application while playing with the built-in assembly example. 

[image]

Steps:
- Launch FreeCAD
- Open AssemblyExample
- Make Assembly object the active object
- Click and drag on the assembly to make it move, observe major stall and prompt to terminate program.

My system is Arch Linux (Linux 6.13.8-arch1-1 #1 SMP PREEMPT_DYNAMIC Sun, 23 Mar 2025 17:17:30 +0000 x86_64 GNU/Linux). CPU Intel i7-12700K and GPU is AMD ATI Radeon RX 6600, 48GB RAM. I am using Hyprland with the Wayland compositor. I am able to provide any other system information that might help diagnose the issue.

luzpaz, 2025-04-09:
Full [About](https://wiki.freecad.org/About) info helps

linguini1, 2025-04-09:
> Full [About](https://wiki.freecad.org/About) info helps

Thank you! My about info is as follows:

```
OS: Arch Linux (Hyprland/wayland)
Architecture: x86_64
Version: 1.0.0.39109 (Git)
Build type: Release
Branch: makepkg
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.13.2, Qt 6.9.0, Coin 4.0.3, Vtk 9.4.2, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/
Installed mods: 
  * Gridfinity 0.11.4
  * OpenTheme 2024.9.1
  * fasteners 0.5.38
  * Render 2024.12.15
```

I also found this forum post in which the presented workaround of launching FreeCAD using `QT_QPA_PLATFORM=xcb freecad` works for me, the performance *drastically* improves, feels like a 10x speedup. Something must be up with the FreeCAD interacts with Wayland? https://forum.freecad.org/viewtopic.php?style=10&t=89408

closed 2025-11-23 by luzpaz