FreeCAD/FreeCAD #18707: Compilation issue on armel, armhf with debundled OipenGL headers, (glEnable(GL_MULTISAMPLE) )

coldtobi, 2024-12-24:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

On Debian machines freecad fails to build with (Debian provided) OpenGL headers, as on these plattforms GL_MULTISAMPLE is not available. (Debian buildd log for armel is [here](https://buildd.debian.org/status/fetch.php?pkg=freecad&arch=armel&ver=1.0.0%2Bdfsg-1%7Eexp1&stamp=1734983848&raw=0))

The only location in the code using the missing define is [src/Mod/CAM/PathSimulator/AppGL/DlgCAMSimulator.cpp:234](https://github.com/FreeCAD/FreeCAD/blob/abb50a4daa43abee9e6b263dd82e5e876c978c6a/src/Mod/CAM/PathSimulator/AppGL/DlgCAMSimulator.cpp#L234)

It seems that on other occasions FreeCAD just defined  the missing value (see PR 18706) but that seemed historical as the code is not using this define anymore.

Looking at documentation, https://registry.khronos.org/OpenGL-Refpages/gl4/html/glEnable.xhtml, I think the call `glEnable(GL_MULTISAMPLE);` isn't needed as it GL_MULTISAMPLE defaults already to "true".):

`
glEnable and glDisable enable and disable various capabilities. Use glIsEnabled or glGet to determine the current setting of any capability. The initial value for each capability with the exception of GL_DITHER and GL_MULTISAMPLE is GL_FALSE. T**he initial value for GL_DITHER and GL_MULTISAMPLE is GL_TRUE.** `

Said that, I'd like to hear feedback before crafting a PR, because I'm not versed in OpenGL.

### Full version info

```shell
freecad 1.0
```

### Subproject(s) affected?

CAM

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-12-25:
@adrianinsaval FYI

adrianinsaval, 2024-12-25:
@shaise might know more about this topic

wwmayer, 2024-12-25:
Does it work if you include glext.h there `#include <GL/glext.h>`?

shaise, 2024-12-25:
> @shaise might know more about this topic

I will investigate. I think we can remove this line as it seems to have no effect.

closed 2025-01-27 by yorikvanhavre