FreeCAD/FreeCAD #18711: PartDesign: Planes (XY XZ YZ) are very small

Pendelf, 2024-12-24:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

It seems to me that the planes (XY XZ YZ) are very small. Maybe it's worth making them bigger or adding this option to the settings. I have a Full HD monitor

[image]

### Full version info

```shell
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.39708 (Git) Conda
Build type: Release
Branch: main
Hash: 173ff4da1ef4f075d56c0dc617d1bf00d5f49d4b
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: Russian/Russia (ru_RU)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * Curves 0.6.52
  * fasteners 0.5.33
  * PieMenu 1.10.2
  * sheetmetal 0.6.2
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

tarman3, 2024-12-24:
Try last weekly build

kaiwas, 2024-12-24:
> Try last weekly build

Confirming the issue. Weekly Build 39741-conda-Linux-x86_64-py311.AppImage (today's)

It may depend on the pixel density or monitor resolution. In my case, it's 1920x1080

[image]

ed3davis, 2024-12-24:
Same issue here. Since new monitor.  2560x1440.

[image]

OS: openSUSE Tumbleweed (KDE/plasmawayland/xcb)
Architecture: x86_64
Version: 1.1.0dev.39741 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: 0d128517dedd154bbecfbabe67094ad56c8d46f5
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * Curves 0.6.51
  * fasteners 0.5.29
  * Manipulator 1.5.7

maxwxyz, 2024-12-25:
Duplicate of https://www.underconsideration.com/brandnew/

closed 2024-12-25 by maxwxyz