FreeCAD/FreeCAD #18712: UI: Tool tips don't show up past first separator in menus

psommerfeld, 2024-12-24:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When hovering my mouse over menu items I see tool tips appear within the first section (before the separator bar), not not in any following sections. This is on Mac.
This happens on tool tips in the context menu I right click on something in the main area, and also the main drop-down menu.

### Full version info

```shell
OS: macOS 14.0
Architecture: arm64
Version: 1.1.0dev.39708 (Git) Conda
Build type: Release
Branch: main
Hash: 173ff4da1ef4f075d56c0dc617d1bf00d5f49d4b
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * fasteners.backup1686222008.1392 0.3.50 (Disabled)
  * ProDarkThemePreferencePack 1.0.0
  * BIM021 2021.12.0
```

### Subproject(s) affected?

None

### Anything else?

https://github.com/user-attachments/assets/2d77f697-45da-4062-a785-ab64622b0293

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

tarman3, 2024-12-24:
Maybe some of items just dont have tool tip

https://github.com/user-attachments/assets/d7a2bcb6-fe31-4007-bff2-33710840f4d5

<details>
  <summary>FreeCAD 1.0.0.39109 (Git) makepkg, Arch Linux (KDE/plasma/wayland)</summary>

```
OS: Arch Linux (KDE/plasma/wayland)
Architecture: x86_64
Version: 1.0.0.39109 (Git)
Build type: Release
Branch: makepkg
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.12.7, Qt 6.8.0, Coin 4.0.3, Vtk 9.3.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Installed mods: lattice2 1.0.0; CurvedShapes 1.0.13; MeshRemodel 1.10.35; freecad.gears 1.3.0; fasteners 0.5.33; sheetmetal 0.6.12; Silk 0.1.5; Curves 0.6.52
```

</details>

<details>
  <summary>FreeCAD 1.1.0dev.39673 (Git) Conda AppImage, Arch Linux (KDE/plasma/xcb)</summary>

```
OS: Arch Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.39741 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: 0d128517dedd154bbecfbabe67094ad56c8d46f5
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: lattice2 1.0.0; CurvedShapes 1.0.13; MeshRemodel 1.10.35; freecad.gears 1.3.0; fasteners 0.5.33; sheetmetal 0.6.12; Silk 0.1.5; Curves 0.6.52
```

</details>

psommerfeld, 2024-12-25:
Yes, I think you are right.
I found some instances where tool tips appeared after the first section.

closed 2024-12-25 by psommerfeld