FreeCAD/FreeCAD #18713: Draft: makeSketch error

webmite, 2024-12-24:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When attempting to use the Draft2Sketch function from the Draft workbench, if the user viewpoint is from the Bottom, the sketch created is out of alignment.  

### Full version info

```shell
[code]
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.2)
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * A2plus 0.4.60k
  * AnimationFreeCAD 1.0.0
  * Assembly3 0.12.2
  * Assembly4 0.50.6
  * BIM021 2021.12.0
  * Curves 0.6.5
  * fasteners 0.4.56
  * freecad.gears 1.0.0
  * lattice2 1.0.0
  * Manipulator 1.5.0
  * MyWorkbench
  * Nodes 0.1.36
  * PieMenu
  * sheetmetal 0.4.1
[/code]
```

### Subproject(s) affected?

None

### Anything else?

Check the forum discussion at 
[https://forum.freecad.org/viewtopic.php?t=62114&start=10#p799014](url)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

tarman3, 2024-12-24:
Link is incorrect
Please attach file with simple example and write steps to reproduce issue.

webmite, 2024-12-24:
There are many pictures and details in the forum.

Below link is the corrected link.  Did not verify that I allowed for git editor formatting.

[https://forum.freecad.org/viewtopic.php?p=799014#p799014](https://forum.freecad.org/viewtopic.php?p=799014#p799014)

luzpaz, 2024-12-25:
> Version: 0.21.2.33771 (Git)

We're on 1.0.0  
Can you reproduce on 1.0.0 ?

webmite, 2024-12-25:
`OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.39416 (Git) Conda
Build type: Release
Branch: main
Hash: f6268ffd2877dadb4440ade541e12c388a517d6e
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/unset/Qt default
Installed mods: 
  * A2plus 0.4.60k
  * AnimationFreeCAD 1.0.0
  * Assembly3 0.12.2
  * Assembly4 0.50.6
  * BIM021 2021.12.0
  * Curves 0.6.5
  * fasteners 0.4.56
  * freecad.gears 1.0.0
  * lattice2 1.0.0
  * Manipulator 1.5.0
  * MyWorkbench
  * MyWorkbench_001.zip
  * MyWorkbench_002.zip
  * Nodes 0.1.36
  * PieMenu
  * sheetmetal 0.4.1
  * sheetmetal.zip
`
In the above Version without modification, 
creating a sketch viewed from the bottom
and then a sketch viewed from the top
results in 2 sketches, Both with the same placement values 
'except' that the bottom sketch shows a rotation angle of 180 deg on z-axis.

With my proposed patch.
results in 2 sketches, Both with the same placement values 
'except' that the bottom sketch shows a rotation angle of 180 deg on z-axis and
the Rotation 'Axis' is set to (0,0,-1).

I see no behavioral difference when I go to extrude these sketches via the Part Workbench. And both allow Placement to be reset, then reconfigured back to original configuration without 'breaking'.

The z-axis derived from placement of a sketch in my opinion should represent the 
normal direction of the sketch plane.  It should be able to be used by the user to infer the direction of 'positive' extrusion.

PR #18948 (2025-01-09): Draft: fix axis calculation in make_sketch
Fixes #18713.

The code follows the suggestion by imm/webmite:
https://forum.freecad.org/viewtopic.php?p=799014#p799014

@webmite Thank you.
commit ab3d0c31 (2025-01-09): Draft: fix axis calculation in make_sketch

Fixes #18713.

The code follows the suggestion by imm/webmite:
https://forum.freecad.org/viewtopic.php?p=799014#p799014

@webmite Thank you.
closed 2025-01-09 by yorikvanhavre