FreeCAD/FreeCAD #18714: loading freecad addon "fasteners" will fail

Ventusfahrer, 2024-12-24:
Hi there!

I was installing the Freecad AddOn Fasteners from the AddOn store.

When trying to load the Addon via Edit -> Preferences -> Workbenches -> Press Load Button behind Fasteners, the UI fires a popup

[image]

The terminal output contains a little more information:

```
freecad
FreeCAD 1.1.0, Libs: 1.1.0devR39100 (Git)
(C) 2001-2024 FreeCAD contributors
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.

Adding snap-specific PYTHONPATH to sys.path: /home/peter/snap/freecad/common/.local/lib/python3.10/site-packages:/snap/freecad/1180/lib/python3.10/site-packages:/snap/freecad/1180/usr/lib/python3/dist-packages
Qt: Session management error: Could not open network socket
connect failed: No such file or directory
'NoneType' object has no attribute 'setCheckable'
Traceback (most recent call last):
  File "/home/peter/snap/freecad/common/Mod/fasteners/./InitGui.py", line 112, in Activated
    FastenerBase.InitCheckables()
  File "/home/peter/snap/freecad/common/Mod/fasteners/./FastenerBase.py", line 892, in InitCheckables
    matchOuterButton.setCheckable(True)
```

freecad version is:

```
$ snap list freecad
Name     Version        Revision  Tracking     Herausgeber  Hinweise
freecad  1.1-g8865450a  1180      latest/edge  freecad-org  -

```

The corresponding weekly appimage from the download aerea is working fine.

kind regards

Peter

luzpaz, 2024-12-24:
@shaise have you encountered this ? Is this fasteners wb or FreeCAD ?

luzpaz, 2024-12-24:
Looking at the issue, it seems the problem is indeed fasteners. So i'm gonna close this and ask @shaise how to proceed.

closed 2024-12-24 by luzpaz

tarman3, 2024-12-24:
Can reproduce on Archlinux.

The related problem only appears when the Load button is pressed.
When selected through the Toolbar, the Fasteners workbench loads without problems.

<details>
  <summary>FreeCAD 1.0.0.39109 (Git) makepkg, Arch Linux (KDE/plasma/wayland)</summary>

```
OS: Arch Linux (KDE/plasma/wayland)
Architecture: x86_64
Version: 1.0.0.39109 (Git)
Build type: Release
Branch: makepkg
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.12.7, Qt 6.8.0, Coin 4.0.3, Vtk 9.3.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Installed mods: lattice2 1.0.0; CurvedShapes 1.0.13; MeshRemodel 1.10.35; freecad.gears 1.3.0; fasteners 0.5.33; sheetmetal 0.6.12; Silk 0.1.5; Curves 0.6.52
```

</details>

<details>
  <summary>FreeCAD 1.1.0dev.39673 (Git) Conda AppImage, Arch Linux (KDE/plasma/xcb)</summary>

```
OS: Arch Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.39741 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: 0d128517dedd154bbecfbabe67094ad56c8d46f5
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: lattice2 1.0.0; CurvedShapes 1.0.13; MeshRemodel 1.10.35; freecad.gears 1.3.0; fasteners 0.5.33; sheetmetal 0.6.12; Silk 0.1.5; Curves 0.6.52
```

</details>

shaise, 2024-12-25:
> The related problem only appears when the Load button is pressed. When selected through the Toolbar, the Fasteners workbench loads without problems.

Yes this is same for windows. I will try to see what makes the difference. Anyway, why do we need the load button if we can simply select it from the toolbal?

luzpaz, 2024-12-25:
Should I re-open this ticket then ?

shaise, 2024-12-25:
It is now fixed via [Fasteners PR #415](https://github.com/shaise/FreeCAD_FastenersWB/pull/415)