FreeCAD/FreeCAD #18715: Parts requiring QtwebEngine are still in the core

Reqrefusion, 2024-12-24:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

QtwebEngine was removed along with https://github.com/FreeCAD/FreeCAD/pull/18662, but the functions that need this module are still in the core.

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.39673 (Git) Conda
Build type: Release
Branch: main
Hash: 1289997dfd604cb24f060f905f04ec3728d964e4
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: Turkish/Turkey (tr_TR)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
```

### Subproject(s) affected?

Other (specify in description)

### Anything else?

QtwebEngine was removed because it caused various problems, but there are functions dependent on it in many parts of the code. These can be used optionally, but I think they should be included in core because of the features related to Python. It's especially confusing that they created a non-working interface part that everyone can see. Because the general user does not understand why this is happening, and the relevant part is seen by everyone, not specifically those who include QtwebEngine. In this regard, it would be more logical to decide to remove the relevant parts. Because if the functions of a module that is not included in the core can be done, they should not be included in the core either.

In addition, the parts where QtwebEngine continues to exist in some way are help and addon manager. These appeal entirely to the general user, with a low probability of including the module externally. Additionally, these modules maintain their existence with a utilitarian philosophy, which makes it unlikely that they will be used by adding the module externally.

This issue has been opened pursuant to https://github.com/FreeCAD/FreeCAD/pull/18662.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-12-25:
@chennes FYI

chennes, 2024-12-25:
Nothing in the Addon Manager uses QtWebEngine, it was removed some time ago.

Reqrefusion, 2024-12-25:
> Nothing in the Addon Manager uses QtWebEngine, it was removed some time ago.

Sorry, I didn't realize that it was showing files in an old commit while browsing through old commits. I think only the relevant parts are available in the help module.

luzpaz, 2026-10-10:
Did a search on all the instances of `QtwebEngine` in the source: 

https://github.com/FreeCAD/FreeCAD/blob/e9b474306f1da54eae97efe3bc578986b3d02a0c/src/Tools/typing/inputs/overlays/PySide/QtWebEngineWidgets.pyi#L1-L6

https://github.com/FreeCAD/FreeCAD/blob/e9b474306f1da54eae97efe3bc578986b3d02a0c/src/Mod/Help/dlgPreferencesHelp.ui#L199

https://github.com/FreeCAD/FreeCAD/blob/e9b474306f1da54eae97efe3bc578986b3d02a0c/src/Mod/Help/Help.py#L80
https://github.com/FreeCAD/FreeCAD/blob/e9b474306f1da54eae97efe3bc578986b3d02a0c/src/Mod/Help/Help.py#L288-L292
https://github.com/FreeCAD/FreeCAD/blob/e9b474306f1da54eae97efe3bc578986b3d02a0c/src/Mod/Help/Help.py#L446-L450
https://github.com/FreeCAD/FreeCAD/blob/e9b474306f1da54eae97efe3bc578986b3d02a0c/src/Mod/Help/Help.py#L473-L481