FreeCAD/FreeCAD #18717: Older projects might be broken when loading them with v1.1

wwmayer, 2024-12-25:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

* Download the four files from here: https://forum.freecad.org/viewtopic.php?p=799219#p799219
* Start v1.1 and load ZolkoTutorial2.FCStd
* Directly after loading the file the model looks correct (but the placement of some objects are already wrong)
* After recomputing the document the model breaks

The problem is that LCS_Origin uses X_Axis as alignment support and map mode is set to ObjectXY. This changes its placement to have a rotation around (1,1,1) by 120 degree instead of (0,0,1) by 0 degree.
Because ZolkoBase is attached to LCS_Origin it will change its placement after the recompute.

The problem is caused by fc5673064 of PR #18332. Changing the rotations of the coordinate axes directly affect the placement of all objects that use one of the axes as attachment support.

### Full version info

```shell
OS: Ubuntu 22.04.5 LTS (XFCE/xubuntu/xcb)
Architecture: x86_64
Version: 1.1.0dev.39743 (Git)
Build type: debug
Branch: main
Hash: 7fb80448f2748beec21a270cab2af0c967587c00
Python 3.10.12, Qt 5.15.3, Coin 4.0.0, Vtk 7.1.1, OCC 7.5.1
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: unset/Classic theme/Qt default
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Zolko-123, 2024-12-30:
you are mad, clinically mad, I don't see any other explanation for breaking things that have worked for 20 years

wwmayer, 2024-12-30:
Don't blame me! I haven't broken this stuff but reported the regression.

yorikvanhavre, 2025-01-06:
Insulting other users is against the FreeCAD code of conduct. @Zolko-123 I'm sorry but if you do it another time we'll have no choice but ban you from here too. I'll delete your message for now

commit ad314fea (2025-01-20): Core: Fixes #18717

Neither in #18332 nor in any of the referenced issues a reason is given why the placement of the coordinate axes must be changed.
Only doing it to make the client code look more consistent is not a valid reason.

This change breaks forward and backward compatibility and with #18717 we already have a hard to solve issue. Furthermore, at this
point of time it's not even clear what other collateral damage the change may cause.

So, this commit hides all the implementation details of the DatumElement and restores the original placement for the axes.
This commit makes the line m [... cut]