FreeCAD/FreeCAD #18718: Volumenkörper verliert beim Neuberechnen die Farbe

notenwendig, 2024-12-25:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Volumenkörper verliert beim Neuberechnen die Farbe nach Import von Datei aus Version: 0.21.1.33668 +26 (Git)

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: unset/unset/Qt default
```

### Subproject(s) affected?

Part

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

tarman3, 2024-12-25:
Please attach file with exampe

Syres916, 2024-12-25:
This is another duplicate of https://github.com/FreeCAD/FreeCAD/issues/18584

closed 2024-12-26 by maxwxyz