FreeCAD/FreeCAD #18720: PolarPattern only valid for specific number of occurences

maxwxyz, 2024-12-25:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Forum report: https://forum.freecad.org/viewtopic.php?t=93123
---

I try to create a gear. I know there is a special tool for making gears, but that is not the case of the topic.
If I set 'Occurrences' to 36, then it is OK. But if I set 'Occurrences' to 35, then the output is weird (see screen).
[image]

File:

[Q100_motor_gear_36.zip](https://github.com/user-attachments/files/18246411/Q100_motor_gear_36.zip)

### Full version info

```shell
OS: Ubuntu 24.04.1 LTS (ubuntu:GNOME/ubuntu/xcb)
Architecture: arm64
Version: 1.0.0.39109 (Git) Conda AppImage
Build type: Release
Branch: (HEAD detached at 1.0.0)
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-12-25:
Can confirm and any other number other than 36 has issues.
```
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.39708 (Git) Conda
Build type: Release
Branch: main
Hash: 173ff4da1ef4f075d56c0dc617d1bf00d5f49d4b
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Installed mods: 
  * CfdOF 1.29.4
  * CurvedShapes 1.0.13
  * Curves 0.6.52
  * dodo 1.0.1
  * fasteners 0.5.33
  * freecad.gears 1.3.0
  * FreeCAD_SketchArch
  * Manipulator 1.5.7
  * OpenTheme 2024.9.1
  * OpticsWorkbench 1.0.26
  * Rocket 4.0.1
  * sheetmetal 0.6.2
  * Silk 0.1.5

```

tarman3, 2024-12-25:
Set Refine = False for AdditiveHelix feature

[image]

Groberen, 2025-04-23:
This bug still present in most recent weekly build:

```
OS: Fedora Linux 41 (KDE Plasma) (KDE/plasma/wayland)
Architecture: x86_64
Version: 1.1.0dev.41354 (Git) AppImage
Build date: 2025/04/21 18:01:03
Build type: Release
Branch: main
Hash: fd390acd53dae91898145a22d74050d0d648a45b
Python 3.11.12, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 
shiboken , xerces-c 3.2.5, IfcOpenShell 0.8.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/141.951/1
```

I created a forum thread about this before I found this post (simple model attached):
https://forum.freecad.org/viewtopic.php?p=823283#p823283