FreeCAD/FreeCAD #18723: Exception in Geometry.cpp when moving cursor around in Sketcher

DavidGriffith, 2024-12-25:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Moving the cursor around while drawing something in Sketcher on a face frequently results in:

```
10:56:44  <Exception> Geometry.cpp(993): bool Part::GeomCurve::closestParameter(const Base::Vector3d&, double&) const -- N4Base14CADKernelErrorE
```

I've reduced the problem to a simple project file.  To trigger the bug, go into Sketcher and select the "Constrain horizontal/vertical" tool.  Then wave the cursor around the sketch, in particular around the horizontal line to the left of the semicircle.

[sketcher-param-bug.FCStd.gz](https://github.com/user-attachments/files/18246969/sketcher-param-bug.FCStd.gz)

### Full version info

```shell
OS: Devuan GNU/Linux 5 (daedalus) (MATE/lightdm-xsession/xcb)
Architecture: x86_64
Version: 1.0.0.39150 (Git)
Build type: Debug
Branch: releases/FreeCAD-1-0
Hash: d9e8fe0d7d99fd48fdeb5a0f6da5790ae3f54b69
Python 3.11.2, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.8.2.dev
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * Assembly3 0.12.2
  * sheetmetal 0.6.11
  * freecad.gears 1.3.0
  * Assembly4 0.50.15
  * A2plus 0.4.68
  * InventorLoader 1.5.1
  * fasteners 0.5.33
  * Manipulator 1.5.7
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

tarman3, 2024-12-25:
Can not reproduce

https://github.com/user-attachments/assets/0a23066a-2443-42ba-aeac-e41b3940764b

<details>
  <summary>FreeCAD 1.0.0.39109 (Git) makepkg, Arch Linux (KDE/plasma/wayland)</summary>

```
OS: Arch Linux (KDE/plasma/wayland)
Architecture: x86_64
Version: 1.0.0.39109 (Git)
Build type: Release
Branch: makepkg
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.12.7, Qt 6.8.0, Coin 4.0.3, Vtk 9.3.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Installed mods: lattice2 1.0.0; CurvedShapes 1.0.13; MeshRemodel 1.10.35; freecad.gears 1.3.0; fasteners 0.5.33; sheetmetal 0.6.12; Silk 0.1.5; Curves 0.6.52
```

</details>

<details>
  <summary>FreeCAD 1.1.0dev.39673 (Git) Conda AppImage, Arch Linux (KDE/plasma/xcb)</summary>

```
OS: Arch Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.39741 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: 0d128517dedd154bbecfbabe67094ad56c8d46f5
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: lattice2 1.0.0; CurvedShapes 1.0.13; MeshRemodel 1.10.35; freecad.gears 1.3.0; fasteners 0.5.33; sheetmetal 0.6.12; Silk 0.1.5; Curves 0.6.52
```

</details>

Syres916, 2024-12-25:
Confirmed using the two _OCCT 7.8.2dev_ builds but not using the _OCCT 7.6.3_ build:

```
OS: Linux Mint 22 (X-Cinnamon/cinnamon/xcb)
Architecture: x86_64
Version: 1.0.0.39150 (Git)
Build type: Release
Branch: releases/FreeCAD-1-0
Hash: d9e8fe0d7d99fd48fdeb5a0f6da5790ae3f54b69
Python 3.12.3, Qt 5.15.13, Coin 4.0.3, Vtk 9.1.0, OCC 7.8.2.dev
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: Behave-dark.qss/Behave-dark/Qt default
Installed mods: 
  * FreeCAD-themes 2024.9.14
  * Defeaturing 1.2.2
  * sheetmetal 0.6.2
  * Behave-Dark-Colors 0.1.1
  * FreeCAD_SketchArch
  * 3dfindit-freecad-integration 1.2.0
  * Curves 0.6.52
  * Assembly4 0.50.15
```

```
OS: Linux Mint 22 (X-Cinnamon/cinnamon/xcb)
Architecture: x86_64
Version: 1.1.0dev.39709 (Git)
Build type: Release
Branch: main
Hash: cad2f33b4e02b92659de3fb27f686f91c94b447f
Python 3.12.3, Qt 5.15.13, Coin 4.0.3, Vtk 9.1.0, OCC 7.8.2.dev
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: Behave-dark.qss/Behave-dark/Qt default
Installed mods: 
  * FreeCAD-themes 2024.9.14
  * Defeaturing 1.2.2
  * sheetmetal 0.6.2
  * Behave-Dark-Colors 0.1.1
  * FreeCAD_SketchArch
  * 3dfindit-freecad-integration 1.2.0
  * Curves 0.6.52
  * Assembly4 0.50.15
```

```
OS: Linux Mint 22 (X-Cinnamon/cinnamon/xcb)
Architecture: x86_64
Version: 1.1.0dev.39709 (Git)
Build type: Release
Branch: main
Hash: cad2f33b4e02b92659de3fb27f686f91c94b447f
Python 3.12.3, Qt 5.15.13, Coin 4.0.2, Vtk 9.1.0, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: Behave-dark.qss/Behave-dark/Qt default
Installed mods: 
  * FreeCAD-themes 2024.9.14
  * Defeaturing 1.2.2
  * sheetmetal 0.6.2
  * Behave-Dark-Colors 0.1.1
  * FreeCAD_SketchArch
  * 3dfindit-freecad-integration 1.2.0
  * Curves 0.6.52
  * Assembly4 0.50.15

```

DavidGriffith, 2024-12-26:
Well, this is disappointing.  I upgraded OCC from 7.6.3 to 8.7.2 to cure a problem with spirals not rendering correctly.  See https://forum.freecad.org/viewtopic.php?p=795523

DavidGriffith, 2024-12-27:
I found that using OCC version 8.7.1 avoids both this problem and the faulty spiral problem.

luzpaz, 2024-12-28:
Upstream ticket should be made since it seems that there is a regression between 8.7.1 and 8.7.2dev

FEA-eng, 2025-10-30:
I can't reproduce this, but it might be OS-specific.

```
OS: Windows 11 build 26200
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/10/21 12:07:13
Build type: Release
Branch: main
Hash: 2096d0bd8603b429fdc0c746c34b3d73c3289fe1
Python 3.11.14, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Navigation Style/Orbit Style/Rotation Mode: Touchpad/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/113.268/1.25
```