FreeCAD/FreeCAD #18725: Crash in Preferences --> "Manage Preference Packs"

JohnOCFII, 2024-12-26:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Clicking on the (eye?) icon causes an immediate crash.  In particular, was attempting to hide the `OpenPreferences` Preferences Pack. 

```
-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               freecad [27667]
Path:                  /Applications/Other Apps/FreeCAD-39741.app/Contents/Resources/bin/freecad
Identifier:            freecad
Version:               ???
Code Type:             ARM-64 (Native)
Parent Process:        launchd [1]
User ID:               501

Date/Time:             2024-12-25 19:47:26.3024 -0600
OS Version:            macOS 15.2 (24C101)
Report Version:        12
Anonymous UUID:        E1A9B3E8-17C0-3635-633F-31D076EC9805

Sleep/Wake UUID:       6BF57B7C-103E-47B1-972B-AF1EBA47AE5D

Time Awake Since Boot: 270000 seconds
Time Since Wake:       40205 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000012
Exception Codes:       0x0000000000000001, 0x0000000000000012

Termination Reason:    Namespace SIGNAL, Code 11 Segmentation fault: 11
Terminating Process:   exc handler [27667]

VM Region Info: 0x12 is not in any region.  Bytes before following region: 4342382574
      REGION TYPE                    START - END         [ VSIZE] PRT/MAX SHRMOD  REGION DETAIL
      UNUSED SPACE AT START
--->  
      __TEXT                      102d38000-102d58000    [  128K] r-x/r-x SM=COW  /Applications/Other Apps/FreeCAD-39741.app/Contents/Resources/bin/freecad

Thread 0 Crashed::  Dispatch queue: com.apple.main-thread
0   libQt5Widgets.5.15.15.dylib   	       0x1046c40ac QDialog::showEvent(QShowEvent*) + 8
1   libFreeCADGui.dylib           	       0x10549c3a0 Gui::Dialog::DlgPreferencePackManagementImp::showEvent(QShowEvent*) + 2148
2   libFreeCADGui.dylib           	       0x10549e628 QtPrivate::QFunctorSlotObject<Gui::Dialog::DlgPreferencePackManagementImp::addTreeNode(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>> const&, std::__1::vector<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>>, std::__1::allocator<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>>>> const&, Gui::Dialog::DlgPreferencePackManagementImp::TreeWidgetType)::$_2, 0, QtPrivate::List<>, void>::impl(int, QtPrivate::QSlotObjectBase*, QObject*, void**, bool*) + 224
3   libQt5Core.5.15.15.dylib      	       0x104ea3fec 0x104d5c000 + 1343468
4   libQt5Core.5.15.15.dylib      	       0x104e8712c QCoreApplication::notifyInternal2(QObject*, QEvent*) + 176
5   libQt5Core.5.15.15.dylib      	       0x104e8712c QCoreApplication::notifyInternal2(QObject*, QEvent*) + 176
6   CoreFoundation                	       0x19ba50828 __CFRunLoopDoSource0 + 176
7   CoreFoundation                	       0x19ba5058c __CFRunLoopDoSources0 + 244
8   CoreFoundation                	       0x19ba4f128 __CFRunLoopRun + 840
9   CoreFoundation                	       0x19ba4e724 CFRunLoopRunSpecific + 588
10  HIToolbox                     	       0x1a6fa6530 RunCurrentEventLoopInMode + 292
11  HIToolbox                     	       0x1a6fac17c ReceiveNextEventCommon + 216
12  HIToolbox                     	       0x1a6fac508 _BlockUntilNextEventMatchingListInModeWithFilter + 76
13  AppKit                        	       0x19f5b9034 _DPSNextEvent + 660
14  AppKit                        	       0x19ff1d2d4 -[NSApplication(NSEventRouting) _nextEventMatchingEventMask:untilDate:inMode:dequeue:] + 688
15  AppKit                        	       0x19f7e6c08 -[NSApplication _doModalLoop:peek:] + 216
16  AppKit                        	       0x19fb2426c __33-[NSApplication runModalSession:]_block_invoke_2 + 84
17  AppKit                        	       0x19fb241fc __33-[NSApplication runModalSession:]_block_invoke + 112
18  AppKit                        	       0x19f7e5110 _NSTryRunModal + 100
19  AppKit                        	       0x19fb240cc -[NSApplication runModalSession:] + 148
20  libqcocoa.dylib               	       0x108dab284 0x108d7c000 + 193156
21  libFreeCADGui.dylib           	       0x1053f2f60 Gui::Command::_invoke(int, bool) + 516
22  libFreeCADGui.dylib           	       0x1053f2d20 Gui::Command::invoke(int, Gui::Command::TriggerSource) + 368
23  libQt5Core.5.15.15.dylib      	       0x104ea3fec 0x104d5c000 + 1343468
24  libQt5Core.5.15.15.dylib      	       0x104e8712c QCoreApplication::notifyInternal2(QObject*, QEvent*) + 176
25  CoreFoundation                	       0x19ba50828 __CFRunLoopDoSource0 + 176
26  CoreFoundation                	       0x19ba5058c __CFRunLoopDoSources0 + 244
27  CoreFoundation                	       0x19ba4f128 __CFRunLoopRun + 840
28  CoreFoundation                	       0x19ba4e724 CFRunLoopRunSpecific + 588
29  HIToolbox                     	       0x1a6fa6530 RunCurrentEventLoopInMode + 292
30  HIToolbox                     	       0x1a6fac17c ReceiveNextEventCommon + 216
31  HIToolbox                     	       0x1a6fac508 _BlockUntilNextEventMatchingListInModeWithFilter + 76
32  AppKit                        	       0x19f5b9034 _DPSNextEvent + 660
33  AppKit                        	       0x19ff1d2d4 -[NSApplication(NSEventRouting) _nextEventMatchingEventMask:untilDate:inMode:dequeue:] + 688
34  AppKit                        	       0x19f5ac060 -[NSApplication run] + 480
35  libqcocoa.dylib               	       0x108dab3c0 0x108d7c000 + 193472
36  freecad                       	       0x102d3ee6c main + 3228
37  dyld                          	       0x19b5e8274 start + 2840
```

### Full version info

```shell
OS: macOS 15.2
Architecture: arm64
Version: 1.1.0dev.39741 (Git) Conda
Build type: Release
Branch: main
Hash: 0d128517dedd154bbecfbabe67094ad56c8d46f5
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/Fusion
Installed mods: 
  * OpenTheme 2024.9.1
```
```

### Subproject(s) affected?

None

### Anything else?

<img width="504" alt="Screenshot 2024-12-25 at 7 39 41 PM" src="https://github.com/user-attachments/assets/07f6abad-07f0-4187-ba67-2e23a88a7073" />

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

JohnOCFII, 2024-12-26:
[image]

maxwxyz, 2024-12-26:
Does this also happen in the stable version or is this not possible there?

xtemp09, 2024-12-26:
@maxwxyz, the crash was introduced by the PR #18157, so it is absent in the stable version.

```ruby
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: Russian/Russia (ru_RU)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
```

wwmayer, 2024-12-27:
When refactoring the code I was already wondering about this block:
```
    if (event)
        QDialog::showEvent(event);
```
Because Qt never sends null events to the event handlers I considered this extra check as superfluous and removed it. Now when looking at the code again I realized that the explicit calls of `showEvent(nullptr);` are on purpose.

But IMO this is a weird code construct that should be avoided. To me it looks like one further step of refactoring is needed here where the code from `DlgPreferencePackManagementImp::showEvent` (except of the `QDialog::showEvent(event);`) should be moved to a new function.

Then `DlgPreferencePackManagementImp::showEvent` and all places where `showEvent(nullptr);` is called should call this new function.

PR #18726 (2025-01-20): Fix the crash in the interface for managing preference packs
Closes #18725. The cause of the crash is a direct invocation of `QDialog::showEvent(QShowEvent *event)` with `nullptr`, leading to the segmentation fault. [^1] Simple minuscule refactoring fixes it.

[^1]: https://github.com/qt/qtbase/blob/d89cef439f5c1a58aeff879a12d9a33292764b7f/src/widgets/dialogs/qdialog.cpp#L851-L860
commit 2fcbe6ee (2025-01-20): Fix the crash in the interface for managing preference packs (#18726)

* Fix the crash in the interface for managing preference packs

Closes #18725

* Modernize the code by removal of redundant void
closed 2025-01-20 by chennes