FreeCAD/FreeCAD #18727: BIM: Closing sketch toggles all hidden BIM obejects in visible again

maxwxyz, 2024-12-26:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I have a BIM project with Site/Building/Level structure. When editing a Sketch and then closing the active sketch, all BIM elements inside the Building hierarchy which are hidden, are toggled to be visible again.
Let's say you work on a wall sketch in level 1 and close the sketch, all levels above will be visible again, hiding the wall you have just edited. You manually need to toggle each object you do not want to see.

### Full version info

```shell
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.39708 (Git) Conda
Build type: Release
Branch: main
Hash: 173ff4da1ef4f075d56c0dc617d1bf00d5f49d4b
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * CfdOF 1.29.4
  * CurvedShapes 1.0.13
  * Curves 0.6.52
  * dodo 1.0.1
  * fasteners 0.5.33
  * freecad.gears 1.3.0
  * FreeCAD_SketchArch (Disabled)
  * Manipulator 1.5.7
  * OpenTheme 2024.9.1
  * OpticsWorkbench 1.0.26
  * Rocket 4.0.1
  * sheetmetal 0.6.11
  * Silk 0.1.5
```

### Subproject(s) affected?

BIM

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-12-27:
Was also reported here https://forum.freecad.org/viewtopic.php?t=93300 for CAM objects

maxwxyz, 2024-12-27:
Maybe related: https://github.com/FreeCAD/FreeCAD/issues/18728

semhustej, 2025-07-21:
I think this is the similar issue, but it shows in a different way - it hides the parent object when the sketch is exited (level in this case). See behavior in the video:

https://github.com/user-attachments/assets/4d19a200-ab09-48f7-89dc-2987bca7a35e

Roy-043, 2025-10-23:
This is related to the "Hide all objects that depend on the sketch" preference.
https://wiki.freecad.org/Sketcher_Preferences#Display

If you uncheck that option the issue does not occur.

Roy-043, 2025-10-23:
I have a hard time following that video.

Have you restarted FreeCAD after changing the preference I have mentioned?

The issue is related to that preference, but also to how Core handles the visibility of group-like objects. When that preference is checked the objects the sketch depends on are hidden. These include top-level objects. So for a BIM wall there may be 3 objects: wall -> level -> building. All 3 are hidden by Sketcher (= good). And when the sketch is closed the 3 object are made visible again (= good). But this is where Core decides to do its own 'thing': when a top-level group or group-like object is made visible, all its descendants are made visible as well. It is that behavior we see here.

semhustej, 2025-10-23:
> 
> The issue is related to that preference, but also to how Core handles the visibility of group-like objects. When that preference is checked the objects the sketch depends on are hidden. These include top-level BIM objects. So for a wall sketch there may be 3 objects: wall -> level -> building. All 3 are hidden by Sketcher (= good). And when the sketch is closed the 3 object are made visible again (= good). But this is where Core decides to do its own 'thing': when a top-level group or group-like object is made visible, all its descendants are made visible as well. It is that behavior we see here.
> 

Yes, this is the issue. It leads non-consistent behavior when entering and exiting  sketch edit mode. I did not change the default sketcher settings.

Steps to reproduce what happens in video:

- open sample file (remove the zip extension)

[sketch-edit-different-behavior.FCStd.zip](https://github.com/user-attachments/files/23099614/sketch-edit-different-behavior.FCStd.zip)

- enter edit mode of `Sketch001`
- exit edit mode of `Sketch001`
- all object have their visibility set as before entering edit mode
- now enter edit mode of `Sketch`
- exit edit mode of `Sketch`
- the parent objects of the `Sketch` object are now hidden (which was not the case before entering sketch edit mode). This is confusing for user (even the `Slab`, which the `Sketch` object is base to). At first sight it seems as if some error happened and the slab got deleted.

The only difference between `Sketch` and `Sketch001` objects at the start of the exercise is the visibility state of their `Building` parent object.

From a user perspective I think no object visibility should change during `enter sketch edit mode` ->  `exit sketch edit mode` process.

sbillin, 2025-12-14:
The OP's issue is frustrating when using BIM with sketches. Thankfully, unchecking the "Hide all objects that depend on the sketch" preference fixes the issue for me.

Possibly what's happening is a buggy attempt to restore the user's visibility settings. When "Hide all objects that depend on the sketch" is checked, the system automatically hides all the dependent objects when entering the sketch and then automatically unhides them when leaving the sketch, presumably to undo the first action of hiding them.  However, this implementation may fail to take into account that if an object was already hidden prior to editing the sketch, then the first automatic hide operation is essentially a no-op, and the system should not unhide any objects that the user already had hidden in the first place.

I have not confirmed any of this in the code.  This is just an impression of what may be going wrong based on observation of system behavior through the UI.