FreeCAD/FreeCAD #18728: BIM: Activating Working Plane unhides all hidden mesh objects

maxwxyz, 2024-12-26:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When working in BIM I often need to change the working plane.
If I do so using the BIM views panel or by double clicking a working plane proxy, all my meshes which are inside a Group folder at root document level are visible again. 
I need to manually hide all objects again, after setting the working plane.

### Full version info

```shell
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.39708 (Git) Conda
Build type: Release
Branch: main
Hash: 173ff4da1ef4f075d56c0dc617d1bf00d5f49d4b
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * CfdOF 1.29.4
  * CurvedShapes 1.0.13
  * Curves 0.6.52
  * dodo 1.0.1
  * fasteners 0.5.33
  * freecad.gears 1.3.0
  * FreeCAD_SketchArch (Disabled)
  * Manipulator 1.5.7
  * OpenTheme 2024.9.1
  * OpticsWorkbench 1.0.26
  * Rocket 4.0.1
  * sheetmetal 0.6.11
  * Silk 0.1.5
```

### Subproject(s) affected?

BIM

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-12-26:
It also unhides all hidden BIM objects in the BIM project hierarchy (Site/Building/Levels)

maxwxyz, 2025-01-07:
Also confirmed on 
```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.39786 (Git) Conda
Build type: Release
Branch: main
Hash: 013eb6e6353ab6e5077ab4696653c9b493442638
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * Curves 0.6.51
  * fasteners 0.5.32
  * freecad.gears 1.2.0
  * Manipulator 1.5.7

```

@Roy-043 @yorikvanhavre FYI

maybe related: https://github.com/FreeCAD/FreeCAD/issues/18727

yorikvanhavre, 2025-01-07:
It shouldn't, of course :)
I'll have a look!

Roy-043, 2025-01-14:
This is intended behavior.

https://wiki.freecad.org/Draft_WorkingPlaneProxy#Context_menu
https://wiki.freecad.org/Draft_WorkingPlaneProxy#Properties ("Restore State" and "Visibility Map")

closed 2025-01-14 by Roy-043