FreeCAD/FreeCAD #18729: BIM: Move underlying wires/base feature with BIM object

maxwxyz, 2024-12-26:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Currently, when moving or editing (offset/mirror/scale) a BIM object, e.g. a wall, the underlying base feature (e.g. a line) does not change. 
It would be great if there is an option to move the base feature of BIM objects when moving the actual object (similar to "Move with host").

### Full version info

```shell
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.39708 (Git) Conda
Build type: Release
Branch: main
Hash: 173ff4da1ef4f075d56c0dc617d1bf00d5f49d4b
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * CfdOF 1.29.4
  * CurvedShapes 1.0.13
  * Curves 0.6.52
  * dodo 1.0.1
  * fasteners 0.5.33
  * freecad.gears 1.3.0
  * FreeCAD_SketchArch (Disabled)
  * Manipulator 1.5.7
  * OpenTheme 2024.9.1
  * OpticsWorkbench 1.0.26
  * Rocket 4.0.1
  * sheetmetal 0.6.11
  * Silk 0.1.5
```

### Subproject(s) affected?

BIM

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-12-26:
I guess not relevant if there is no underlying base feature in the future

Roy-043, 2024-12-27:
All BIM components (walls, windows, etc.) have a "Move Base" property. Please experiment with that property.

maxwxyz, 2024-12-27:
@Roy-043 thank you!

closed 2024-12-27 by maxwxyz

Roy-043, 2024-12-27:
I don't know.

yorikvanhavre, 2025-01-07:
More than often base geometry is used also for other things, so this can have unwanted consequences. IMHO it's at the moment safer to let it off by default. Also, I plan to make walls base-less by default, so this will become less of an issue.