FreeCAD/FreeCAD #18730: BIM: Trimex for straight wall segments start/end face

maxwxyz, 2024-12-26:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When using trimex in BIM, a new copy of the object/face is created.
It would be great when using trimex on the start or end face of a straight wall, to just increase or decrease the wall segment length, similar if I would use the trimex command directly on the underlying wire to shorten/lengthen the wall segment. 

Currently I cannot shorten the wall itself, as a new object is created, so I have to trimex the underlying wire which could be located anywhere in the document, without the possibility to reference/snap to local features.

### Full version info

```shell
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.39708 (Git) Conda
Build type: Release
Branch: main
Hash: 173ff4da1ef4f075d56c0dc617d1bf00d5f49d4b
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * CfdOF 1.29.4
  * CurvedShapes 1.0.13
  * Curves 0.6.52
  * dodo 1.0.1
  * fasteners 0.5.33
  * freecad.gears 1.3.0
  * FreeCAD_SketchArch (Disabled)
  * Manipulator 1.5.7
  * OpenTheme 2024.9.1
  * OpticsWorkbench 1.0.26
  * Rocket 4.0.1
  * sheetmetal 0.6.11
  * Silk 0.1.5
```

### Subproject(s) affected?

BIM

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

ishananand06, 2024-12-29:
Hey @maxwxyz, I'm new to open source and would love to work on this feature.

maxwxyz, 2024-12-29:
@ishananand06 go ahead and create a PR!

paullee0, 2024-12-31:
Good to have more contributor :)

@maxwxyz, do you expect trimex to work on Wall with and without Base object?

maxwxyz, 2024-12-31:
yes

yorikvanhavre, 2025-01-06:
Good idea. It works already like that for Draft stretch.

maxwxyz, 2025-01-22:
related: https://github.com/FreeCAD/FreeCAD/issues/5608

closed 2026-09-12 by Roy-043