FreeCAD/FreeCAD #18731: BIM: Autoconstraint base feature sketch on creation

maxwxyz, 2024-12-26:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When creating BIM objects like walls the default setting is now to create a WallTrace sketch instead of a Draft wire.
When editing the sketch, there are no (auto)constraints in the base feature sketch.
Depending on the way how the wall is drawn, it could set auto-constraints to the WallTrace sketch:
- horizontal/vertical constraints
- length of the segment
- angles for polylines
- diameter/radius
- depending on snapping settings, e.g. parallel constraints
- ...

### Full version info

```shell
https://github.com/FreeCAD/FreeCAD/issues/5608
```

### Subproject(s) affected?

BIM

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2025-01-03:
Just a small note: Currently unconstrained single line sketches can be edit with [Draft_Edit](https://wiki.freecad.org/Draft_Edit). This feature would probably break that.

yorikvanhavre, 2025-01-06:
Do we really want that? It would for ex. make it impossible to move an enpoint of a wall out of its alignment. This might be a good thing, I'm not sure...