FreeCAD/FreeCAD #18732: BIM: Ability to save current setting as preset

maxwxyz, 2024-12-26:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When creating a BIM wall or window/door, there is the ability to create the new object based on a preset.
The dropdown has only options for choosing existing presets.
There is no ability to save the current object or state in the task dialog (when creating a new wall) as a new preset.
I also did not find any possibility to create presets for different walls/windows in my current BIM project. 

It would be great if you could
- save an existing object as preset
- save the state of the task dialog on creation as preset
- create and manage presets for the entire BIM project (e.g. define wall presets and doors/windows prior to drawing in the document)

### Full version info

```shell
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.39708 (Git) Conda
Build type: Release
Branch: main
Hash: 173ff4da1ef4f075d56c0dc617d1bf00d5f49d4b
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * CfdOF 1.29.4
  * CurvedShapes 1.0.13
  * Curves 0.6.52
  * dodo 1.0.1
  * fasteners 0.5.33
  * freecad.gears 1.3.0
  * FreeCAD_SketchArch (Disabled)
  * Manipulator 1.5.7
  * OpenTheme 2024.9.1
  * OpticsWorkbench 1.0.26
  * Rocket 4.0.1
  * sheetmetal 0.6.11
  * Silk 0.1.5
```

### Subproject(s) affected?

BIM

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-12-28:
Additionally it would be great to set a path to a preset directory directly in the BIM preferences, instead of adding a custom string parameter using the parts_library plugin.

paullee0, 2024-12-31:
There seems some similar requests.

I am not sure the exact workflow, but users may just make Clone or Link with the created window/doors, put the latter in a dedicated folder.  Or there is other expectation?

yorikvanhavre, 2025-01-07:
Note that if you have the FreeCAD library installed, you are able to use the "Doors" and "Windows" objects there as a preset. This could be extended with other windows already present in the document, but honestly it does not work very well...

The main problem here is that the window presets are created by code, and this is to ensure all the needed parameters are correctly done so the window works as expected (openings, etc..).

But maybe the creation dialog should indeed offer an easy way to duplicate an existing window.

Another thing is that with NativeIFC we are now introducing the concept of types (or families). So you could define a type then create different windows based on that type. We could look into how that could be used internally, for ex. To have several windows use a same sketch.