FreeCAD/FreeCAD #18733: BIM: Default wall height should use level height

maxwxyz, 2024-12-26:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

As there is no way to define a preset for new walls, it would be great if walls would have the wall height of the parent level (building part) height as their default height.
My new walls are always created at 20 cm width and 300 cm height but I need 7,6 cm wide walls and only 230 cm height.

### Full version info

```shell
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.39708 (Git) Conda
Build type: Release
Branch: main
Hash: 173ff4da1ef4f075d56c0dc617d1bf00d5f49d4b
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * CfdOF 1.29.4
  * CurvedShapes 1.0.13
  * Curves 0.6.52
  * dodo 1.0.1
  * fasteners 0.5.33
  * freecad.gears 1.3.0
  * FreeCAD_SketchArch (Disabled)
  * Manipulator 1.5.7
  * OpenTheme 2024.9.1
  * OpticsWorkbench 1.0.26
  * Rocket 4.0.1
  * sheetmetal 0.6.11
  * Silk 0.1.5
```

### Subproject(s) affected?

BIM

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Syres916, 2024-12-26:
The condition here should be only _if_ the parent is a BIM object as the Wiki states `a Wall can be created as a standalone object` and is used by non-Arch users for a variety of uses that may not be apparent to some, for example, https://forum.freecad.org/viewtopic.php?p=792347#p792347

paullee0, 2024-12-31:
There happens to be a discussion (recent update) about setting defaults for Wall - https://forum.freecad.org/viewtopic.php?p=800209#p800209.

And user can set the height to 0 ... see wiki:

" Height: The height of the wall. Ignored if the wall is based on a solid. If set to zero, and the wall is inside a [floor] (https://wiki.freecad.org/Arch_Floor) object with its height defined, the wall will automatically take the value of the floor height."