FreeCAD/FreeCAD #18734: BIM: Cloning stairs will not clone railing

maxwxyz, 2024-12-26:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When cloning a stair where the railing was automatically generated, the railing is not cloned when the stairs object is cloned/copied.
Just the normal stairs are cloned and there is no easy way to setup the new railing for the clone.

### Full version info

```shell
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.39708 (Git) Conda
Build type: Release
Branch: main
Hash: 173ff4da1ef4f075d56c0dc617d1bf00d5f49d4b
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * CfdOF 1.29.4
  * CurvedShapes 1.0.13
  * Curves 0.6.52
  * dodo 1.0.1
  * fasteners 0.5.33
  * freecad.gears 1.3.0
  * FreeCAD_SketchArch (Disabled)
  * Manipulator 1.5.7
  * OpenTheme 2024.9.1
  * OpticsWorkbench 1.0.26
  * Rocket 4.0.1
  * sheetmetal 0.6.11
  * Silk 0.1.5
```

### Subproject(s) affected?

BIM

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

paullee0, 2024-12-31:
Interesting, railing was one very basic feature I added when I started learning Python some time ago.  Indeed, there are many problems with it.

It is not part of the Stairs itself, but in fact ArchPipe which somehow follow the outline of Stairs.  So you need to clone the Railings on both sides if needed.

Just curious what is your workflow?

maxwxyz, 2024-12-31:
@paullee0 
1. Create a line (side view), 
2. create a stair object with the line as basis. The rails are getting generated automatically. 
3. When I clone the stairs object (just stairs selected), only the stair object, without the rails. 

I now have deleted all rails, because rails also did not get hidden, when the stairs object or the parent BIM level gets hidden, so they were always visible in the document. BTW is there an easy way to create new rails to an existing stairs?

paullee0, 2024-12-31:
Your workflow is standard as described in the wiki.  But as the Railings are objects separate from the Stairs itself (currently), cloning Stairs won't clone Railings.  Users need to clone Railings separately if desirable.

Maybe we needs better ideas how Railings are implemented - like 1 Window objects get all the component (not separate objects) e.g. frame, glass pane, louvre?
 
> I now have deleted all rails, because rails also did not get hidden, when the stairs object or the parent BIM level gets hidden, so they were always visible in the document. BTW is there an easy way to create new rails to an existing stairs?

Indeed, there is no tool to re-create the railing, it would be done though, should not be too difficult - just it is part of the original Stairs code which creates everything.   But the easier way, before a new tool, is to select the base Wire/Sketch object and click Stairs again :)  Then, delete the old one.

maxwxyz, 2024-12-31:
@paullee0 also the visibility issue is annoying :)

Roy-043, 2025-06-26:
> BTW is there an easy way to create new rails to an existing stairs?

You can create a separate pipe and then update the "Railing Left/Right" property to point to that object. Recompute and a RailingWire is created and the pipe updated.

semhustej, 2025-07-08:
>What about adding the railings as additions to the stairs? Then they will get cloned together. There could be a preference option for that...

When I tried to Clone the stair with railings as additions to the stairs (I `Arch_Add` them manually), only the stair was cloned. Not sure if this is what you meant though.

semhustej, 2025-07-08:
> Your workflow is standard as described in the wiki. But as the Railings are objects separate from the Stairs itself (currently), cloning Stairs won't clone Railings. Users need to clone Railings separately if desirable.

Like this I guess:

https://github.com/user-attachments/assets/70d06f16-d797-40dd-b8c7-5e50af75b195

> 
> Maybe we needs better ideas how Railings are implemented - like 1 Window objects get all the component (not separate objects) e.g. frame, glass pane, louvre?
> 
This is an interesting idea. It would be definitely handy in some cases. I am not sure though, if the work is worth the reward.
My experience from Revit is that parametric railings can get very complicated with lots of corner cases.

> Indeed, there is no tool to re-create the railing, it would be done though, should not be too difficult - just it is part of the original Stairs code which creates everything.

This sounds interesting too. It might be a tool located in the RMB conrtext menu (`Recreate railings`). It would feel more professional than advising users to re-create the stair and delete it :).

>But the easier way, before a new tool, is to select the base Wire/Sketch object and click Stairs again :) Then, delete the old one.

Some stairs can be created without a base object and in that case it's this wouldn't work, would it?
Also, if you have a base object but you changed the stair height, width, the newly created stair with railing would not be created correctly, would it?

Roy-043, 2025-07-08:
The railings are already children of the stairs. Adding them as additions is not possible.

There used to be a separate building part tool. I have always understood that as the BIM equivalent of an assembly, in other words a way of grouping objects that belong together. A building part can be cloned. There is an issue with the colors. And a bigger issue with the nesting (the railings have to be added to the BP group manually). No doubt there are also IFC related issues (typing, import/export).

[stairs-buildingpart-clone.FCStd.zip](https://github.com/user-attachments/files/21119395/stairs-buildingpart-clone.FCStd.zip)
(Remove .zip)

Roy-043, 2026-04-08:
Before v0.20 railings were additions. But this caused recompute issues. See #8444.