FreeCAD/FreeCAD #18735: [CORE] STEP File Import Takes Excessive Time

luisDias89, 2024-12-26:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The issue lies in the performance of importing STEP files.
I tested it in SolidWorks, and the attached file is imported in 1 minute, with a dialog window accompanying the process.

In FreeCAD, it takes 1 hour!! , and the software seems to freeze during the operation.

Additionally, after importing the file, if I save it in FreeCAD format, it takes a very long time to open again when I try to load it.

This is important to resolve as it is a key factor preventing a transition to FreeCAD.

Exemple: Reading Step File from Solidowrks
[image]

Import STEP FILE to test:
[EixoY.zip](https://github.com/user-attachments/files/18251682/EixoY.zip)

Model Imported: 
[image]

### Full version info

```shell
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.39708 (Git) Conda
Build type: Release
Branch: main
Hash: 173ff4da1ef4f075d56c0dc617d1bf00d5f49d4b
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: Portuguese/Portugal (pt_PT)
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/Fusion
Installed mods: 
  * OpenTheme 2024.9.1
  * sheetmetal 0.5.6
```

### Subproject(s) affected?

File formats

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luisDias89, 2024-12-26:
I have already tried troubleshooting the issue and reached a conclusion.
What FreeCAD struggles with is importing a helix: this is what causes it to take so long.
[image]

Syres916, 2024-12-26:
@luisDias89 as usual with these things there's more than one issue at play here. Helix in themselves are not a big blocker for OpenCascade to import from Step files. I can import the attached thread Step file made using 0.21.x back into a new file in seconds with my cores compiling another version of FreeCAD at the same time on a 12 year old machine.

But, you are using a dev version of FreeCAD that has a known issue with adding volume/mass to every object in the document and that includes imports. This new code has been found to not have any facility to turn off the excessive CPU load created by it so the developer is working either to fix or revert it.

I personally never raise an issue with a Step file unless an independent 3rd party product like VariCAD has been tested as well especially if the output from VariCAD is significantly different from the originating product. Until I can leave my PC alone to allow VariCAD to complete the import of your test file as it stalls at 15% I won't be able to verify whether the output is significantly different. Hopefully I'll get an answer either way later tonight.

[M10By300Thread-M10x300-ScrewTap.step.zip](https://github.com/user-attachments/files/18253933/M10By300Thread-M10x300-ScrewTap.step.zip)

luisDias89, 2024-12-26:
@Syres916 1st: The file you sent imports in seconds in both 1.0 and 1.1Dev.
2nd: I’m using version 1.1 and I always test the issue in version 1.0 as well (to verify it’s not caused by the development version).

OK, that seems logical. Calculating the mass and volume of a cube is easy.
Calculating for an object with irregular geometry, unless the algorithm is extremely well-optimized, seems like it would significantly increase computational power requirements.
Perhaps a temporary solution to make it work would be to add a processing time limiter or a checkbox to disable volume and mass processing during import.

Syres916, 2024-12-26:
VariCAD took 19 minutes 35 seconds to load stating 2 solids weren't valid.

Kuzma30, 2024-12-26:
Test on [EixoY.zip](https://github.com/user-attachments/files/18251682/EixoY.zip) 
import this file less than minute
```

OS: Windows 11 build 22621
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: Ukrainian/Ukraine (uk_UA)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
```

oursland, 2024-12-29:
This could be related to https://github.com/FreeCAD/FreeCAD/issues/16849.

kpemartin, 2024-12-29:
@oursland that is a possibility. It all depends on how many `DocumentObject`s are created importing the STEP file. I have not had a chance to see how the upcoming fixes for #16849 affect this.

DemoFreak, 2025-02-11:
@Syres916 
> I personally never raise an issue with a Step file unless an independent 3rd party product like VariCAD has been tested as well especially if the output from VariCAD is significantly different from the originating product. Until I can leave my PC alone 

Because you mentioned VariCAD: I have several models here that load in VariCAD in under a minute and can then be used normally, in particular you can move the models around in the room without any problems. The models consist of several thousand components on a PCB, plus mounting elements and cooling profiles.

FreeCAD 1.0.0 takes a good 10 minutes to load the model and then every click in the GUI leads to a further 1-2 minutes waiting time, during which one core is fully utilized. You don't even get to move the model.

I can make one of the models available on request, but cannot upload it publicly here.

kpemartin, 2025-02-12:
@DemoFreak I might find such a sample useful for further speed-up investigations after #16849 is resolved.

kpemartin, 2025-02-12:
I just tried importing your Eixoy.STEP file with the PR included and it took under a minute, perhaps 30 seconds. I was just timing it in my head because I don't need a stopwatch to see if something (still) takes 10 minutes.

I used the default STEP import options (Enable STEP compound merge, nothing else checked, Mode is single document).

The other example you sent me took about 4 minutes to load. I don't observe any unreasonable delays in selecting objects, or zooming, panning, or rotating the display.

Even though my PR doesn't directly affect such screen manipulations, after an import FC will save a backup of the model. The interval for this is based on a timer, and after a lengthy import it can start immediately, and the app freezes until this is complete. This might explain the sluggish behaviour you observed. This first saving of the backup model is something my PR helps with which may be why it seems "fixed" to me.

Actually I just noticed that I have this AutoRecovery Save turned off, probably because it causes me grief when analyzing these performance problems.

Until this PR is merged and available maybe you can turn off the AutoRecovery Save (in Preferences->General->Document), at the expense of, well, having no autorecovery save... This might eliminate the sluggishness you observed.

oursland, 2025-02-12:
@DemoFreak @kpemartin 

To checkout and test a PR there are a few approaches:

1. Use the [`gh` CLI tool](https://cli.github.com/): `gh pr checkout 18676`

2. Check out the user's fork and branch with `git`    

    git add remote -f kpemartin https://github.com/kpemartin/FreeCAD.git
    git checkout kpemartin/Issue16849B

4. Check out the PR directly:

    git fetch -f origin pull/18676/head:pull18676
    git checkout pull18676

The `gh` CLI is likely the most straightforward.

kpemartin, 2025-10-31:
Thanks for the correction and the additional file. Rather than having the information scattered in several posts, I've updated my previous post to include the additional file. The new one takes a really long time for not that many objects...

kpemartin, 2025-11-05:
Oddly, the code seems to spend an inordinate fraction of its time in `ImportOCAF::loadColors`

kpemartin, 2025-11-05:
The code in `ImportOCAF::loadColors` is called on each object in the model.
This code first gets the colour for the overall object, then iterates over all the object's sub-shapes, looking for any per-subshape colour assignments.
For the "confidential" sample there are a total of about 119000 such sub-shapes. The code first looks for a general colour, then a surface colour, and finally a curve colour for each, stopping once it find an assigned colour. In this particular sample most objects, about 101000 of them, have a Surface colour applied, and the remaining 18000 or so have no colour locally applied. This results in about 220000 calls to `XCAFDoc_ColorTool::GetColor(const TopoDS_Shape& S, const XCAFDoc_ColorType type, Quantity_ColorRGBA& color)`, of which 101000 find a colour, and 219000 find no colour. The time spent averages out to 600us per call, but that all adds up to over half the CPU time spent importing the file.

This could be sped up a bit, at least for this particular file, by first looking for a surface colour, then a general colour, but...
1. That would generate different results for an object that has both general and surface colours applied (though I don't know how other code would treat this situation; the comments near the definition of the `XCAFDoc_ColorType` enum imply that surface and curve colours override general ones so the current code could be considered wrong if both are specified)
2. Though it would speed up this file import, it would slow down others that use general colours rather than surface colours.

It does strike me, however, that `XCAFDoc_ColorTool::GetColor(const TopoDS_Shape& S, const XCAFDoc_ColorType type, Quantity_ColorRGBA& color)` seems to be taking a very long time to do what should be a simple property access,

Because I'm on WIndows I'm using the RelWithDebInfo build and taking the XCAF stuff from the libpack with no debug info, so I can't profile that code. If I want to investigate deeper I'll have to build my own XCAF library/ies with debug info. Or perhaps these exist in the libpack and I can force them to load instead of the non-debug ones.

I also have not looked into what the code actually does with the colour info returned by `ImportOCAF::loadColors`; after all, the Topo_Shapes already have colours assigned (that's what the code is querying), so what are the colours being used for?

kpemartin, 2025-11-05:
...and the answer seems to be that *nothing* uses the colours fetched by `ImportOCAF::loadColors`!
At the end of the function it calls virtual `applyColors` with the colour vector. This method defaults to a no-op for the `ImportOCAF` class. There is a derived` ImportOCAFCmd` class that saves the colour vector in a map indexed by the `Part::Feature *`, but that map is private and the only public access method for it is never called, and in fact the entire `ImportOCAFCmd` class is unreferenced, as far as I can tell.

So it would appear that `ImportOCAF::loadColors` should itself be virtual, do nothing in the base `ImportOCAF` class, and the current slow code should be in an (unused) override in `ImportOCAFCmd`. The `applyColors` method should go away.

Or perhaps the entire `ImportOCAFCmd` class should be removed because it is unused. The only functionality it provides is to produce this map of `Part::Feature *` to vector of colours.

Though I still feel that `XCAFDoc_ColorTool::GetColor(const TopoDS_Shape& S, const XCAFDoc_ColorType type, Quantity_ColorRGBA& color)` is inordinately slow...

stefano2734, 2026-09-12:
Occt 8.0.1 with significant changes is available, so a new test with actual dev and libpack 3.5.5  from old 7.8.1 to this is perhaps here a good idea.