FreeCAD/FreeCAD #18736: UI: Docked panel in overlay mode too tall

thomasgi1, 2024-12-26:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I'm running FC on a 14'' MacBook AIr. So, I'm using the overlay mode for the task and property panel to get most of the 3D modelling space. However, in overlay mode the panels are too tall, so that they obscure the status bar. Even worse is that the panel content at the bottom edge is not visible. For instance, it is not possible the close the 'Check Geometry' task dialog since the close button is not visible and scrolling is not possible.

[image]

### Full version info

```shell
OS: macOS 12.7.6
Architecture: x86_64
Version: 1.1.0dev.39708 (Git) Conda
Build type: Release
Branch: main
Hash: 173ff4da1ef4f075d56c0dc617d1bf00d5f49d4b
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * MeshRemodel 1.10.35
  * DynamicData 2.70.0
  * OpenTheme 2024.9.1
  * Defeaturing 1.2.2
  * fasteners 0.5.33
  * CurvedShapes 1.0.13
  * Curves 0.6.52
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-12-29:
I cannot reproduce this, where did you snap your panels to? Does it make a difference when you position / snap them somewhere else?

thomasgi1, 2024-12-29:
> I cannot reproduce this, where did you snap your panels to? Does it make a difference when you position / snap them somewhere else?

Yes, the position makes a difference:

Snap to the right side, navigation cube on the right -> panel too tall (panel obscures the navigation cube):

[image]

Snap to the right side, navigation cube on the left -> panel too tall (panel obscures the status bar):

[image]

Snap to the left side, navigation cube on the right -> panel size ok:

[image]

Snap to the left side, navigation cube on the left -> panel size ok:

[image]

So, only the snapping the task/property panel to the right side causes the problem.

maxwxyz, 2024-12-29:
On my right side you can snap it between the navicube, similar to your last image.

thomasgi1, 2024-12-29:
I can't snap the panel on the right side between the menubar and the navicube. The navicube is completely ignored. See the difference in the snapping behaviour in the attached video:

https://github.com/user-attachments/assets/03f9e2f4-5d9f-4579-b0c0-3ce630778c12

thomasgi1, 2024-12-30:
After inspecting the source code I found that the overlay sizes can be influenced by some user parameters - especially the "Offset2" parameter. In my user.cfg file I found following:

```
<?xml version="1.0" encoding="UTF-8" standalone="no" ?>
<FCParameters>
  <FCParamGroup Name="Root">
    <FCParamGroup Name="BaseApp">
      ...
      <FCParamGroup Name="MainWindow">
        ...
        <FCParamGroup Name="DockWindows">
          ...
          <FCParamGroup Name="OverlayRight">
            ...
            <FCInt Name="Offset2" Value="-140"/>
            ...
          </FCParamGroup>
          ...
        </FCParamGroup>
        ...
      </FCParamGroup>
    </FCParamGroup>
    ...
  </FCParamGroup>
</FCParameters>
```

I have no idea, how the "Offset2" got the -140 value. However, after changing it to 0 the right overlay panel behaves as expected.

closed 2024-12-30 by thomasgi1