FreeCAD/FreeCAD #18740: Missing "Create a Datum" on both toolbar and menu

carribeiro, 2024-12-26:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

It's actually pretty simple. The menu optionand toolbr for "Create a Datum" aren't showing up. If they're moved elsewhere please let me know. I couldn't find any mention of it.

[image]

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.39708 (Git) Conda
Build type: Release
Branch: main
Hash: 173ff4da1ef4f075d56c0dc617d1bf00d5f49d4b
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: Portuguese/Brazil (pt_BR)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * LCInterlocking
  * sheetmetal 0.2.61
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

tarman3, 2024-12-26:
Also there are no **Create Part** (Std_part)

maxwxyz, 2024-12-26:
@FreeCAD/design-working-group / @PaddleStroke what about moving the group command in the "Tools" menu?

carribeiro, 2024-12-26:
I could not find any info about it. It's mentioned in the Release Notes for 1.1.0 at https://wiki.freecad.org/Release_notes_1.1. It should at least be listed there (preferably with the discussion about the reasoning for the move).

carribeiro, 2024-12-26:
Also - I tried to customize the toolbar to put the create datum button there, but the toolbar editor was "empty" when I selected the Part Design toolbar, so I avoided doing it. It's a separate issue. I'll check it later.

carribeiro, 2024-12-26:
Finally - I don't understand the reasoning, I'm assuming I must be doing something wrong, because I tend to create planes while I'm working. I started doing it because it made my designs a bit more robust against some TNP issues. Perhaps this is not needed anymore, but on the other hand, I don't understand why the "Create Datum" operation was moved to the "Structure" toolbar. Is there any doc explaining the reasoning?

tarman3, 2024-12-26:
Because now it is not part of PartDesign Workbench.
You can create Datum plane  in any other workbench, e.g. in Part.

carribeiro, 2024-12-26:
Now I understand, but I think the release note as it is is pretty sparse and does not convey the breadth of the changes. I'm assuming that the idea is to have a more robust set of operations to deal with coordinate systems and planes in a single place (hence the new "structure" tab), but its effects overall are far from clear for anyone not closely following the discussion.

I guess this may be closed but adding a note re:the new datum plane location on the Wiki would make sense.

closed 2025-07-19 by maxwxyz