FreeCAD/FreeCAD #18742: [CORE] Issue with Helix Parameter Visualization

luisDias89, 2024-12-27:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I was trying to create a Subtractive Helix and noticed that some parameters result in a flawed visualization.
Here's an example: with 4.9, it looks fine, but if I set it to 5, the visualization doesn't render correctly.

Subtract Helixe :
[image]

With 4.9, 5.1, or 5.001, it's okay.
[image]

Problem, if i put 5 on pitch:
[image]
Image:
[image]

Tested on 1.0 anda 1.1dev

File to test:
[SCR1605-Y.zip](https://github.com/user-attachments/files/18256838/SCR1605-Y.zip)

### Full version info

```shell
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.39708 (Git) Conda
Build type: Release
Branch: main
Hash: 173ff4da1ef4f075d56c0dc617d1bf00d5f49d4b
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: Portuguese/Portugal (pt_PT)
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/Fusion
Installed mods: 
  * OpenTheme 2024.9.1
  * sheetmetal 0.5.6
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

tarman3, 2024-12-27:
Duplicate #18743

luisDias89, 2024-12-28:
@tarman3 I opened this one just for the visualization issue, the other was for the performance problem and software freezing.
Perhaps close the other issue, since we now understand that the performance problem is due to the Deviation being set at 0.5%
And leave this one open to address the issue of how it handles circles

luisDias89, 2024-12-28:
@chennes

maxwxyz, 2024-12-28:
@kadet1090 might be solved with your new transparent preview?

FEA-eng, 2026-05-10:
Still relevant

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.2.0dev.20260429 (Git shallow)
Build date: 2026/04/29 07:07:16
Build type: Release
Branch: grafted,grafted
Hash: a785d40ac4c37f559b0b82fbaf9bf7dc68a468e4
Python 3.11.15, Qt 6.8.3, Coin 4.0.8, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL) [ OS: Polish/Poland (pl_PL) ]
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Navigation Style/Orbit Style/Rotation Mode: Blender/Rounded Arcball/Window center
Logical DPI/Physical DPI/Pixel Ratio: 96/113.268/1.25
```