FreeCAD/FreeCAD #18743: [CORE] Helix crach freecad

luisDias89, 2024-12-27:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The problem is when creating a helix with a length of 743.
If I use lengths up to 100 or 200, the software handles it well. Beyond that, it sometimes freezes, and I have to restart the software.
I have an i7 8700K with 32 GB of memory, so I don't think it's because of the PC.

It's the kind of problem that's too frequent; it's impossible to work with this feature.
It just keeps freezing, and the correct visualization doesn't appear.
if i put 743 on with of helix and press ok. The software freeze.
It's necessary to run several tests. It doesn't always freeze.

[image]
[image]
[image]

 I teste on 1.0 anda 1.1Dev
 File: 
[SCR1605-Y.zip](https://github.com/user-attachments/files/18256952/SCR1605-Y.zip)

### Full version info

```shell
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: Portuguese/Portugal (pt_PT)
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/Fusion
Installed mods: 
  * OpenTheme 2024.9.1
  * sheetmetal 0.5.6
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

tarman3, 2024-12-27:
Move point of circle outside

[image]

luisDias89, 2024-12-27:
@tarman3 Alright, I just don’t know how to move the dot to the right. It drags the whole circle

Actually, the sketch is like this. The same sketch creates two helices.
SolidWorks handles this very well; the same sketch does everything at once. Here, I have to split it into two operations using the same sketch. I'd like to know how to remove the central point from here
[image]

tarman3, 2024-12-27:
https://github.com/user-attachments/assets/7a168358-d5c8-492e-99ca-c4c064204655

luisDias89, 2024-12-27:
@tarman3 I find all the bugs in this. This tool is still a draft...

https://github.com/user-attachments/assets/d7ec4603-ffc6-41ef-90d9-d9d1a4424d4d

tarman3, 2024-12-27:
Put only one wire in sketch for helix

luisDias89, 2024-12-27:
@tarman3 i use 
[image]
with 20 sides.
How do I place only one wire? The way you showed me?
The problem is that this way it doesn't get as heavy in terms of processing. If I add a circle, the model becomes very heavy computationally. Do you agree with the idea?

GrabCAD: https://grabcad.com/library/scr1605-800mm-1

tarman3, 2024-12-27:
> with 20 sides. How do I place only one wire? The way you showed me?
I wrote workaround only for circle. 

> If I add a circle, the model becomes very heavy computationally. Do you agree with the idea?
At this time model to complicated for FreeCAD.

luisDias89, 2024-12-27:
@tarman3 This image is just SolidWorks. Notice what the software does to the circle. Even though it's a circle, it is being computed as a polygon to reduce computation time. Is it possible to do the same in FreeCAD?
[image]

tarman3, 2024-12-27:
In FreeCAD is the same

luisDias89, 2024-12-27:
@tarman3 Ok, so here's the problem. To be somewhat similar to SolidWorks, I have to set a 1.5 deviation. And believe me, the difference in processing is immediately noticeable.
However, on the other hand, small circles become deformed.
In SolidWorks, the algorithm progressively handles the geometries, limiting the minimum number of angles to be processed depending on the size.
Look at this SolidWorks image:
[image]
[image]

Now look at the FreeCAD images, I managed to improve the helix, but the rest has low definition:
[image]
[image]

luisDias89, 2024-12-27:
[image]

tarman3, 2024-12-27:
Solidworks using geometrical kernel Parasolid behind which is the Siemens corporation.
I believe FreeCAD will get better, but I wouldn't compare FreeCAD and Solidworks.

tarman3, 2024-12-27:
> In SolidWorks, the algorithm progressively handles the geometries, limiting the minimum number of angles to be processed depending on the size.

This idea is interesting.
I suggest you create an Issue

luisDias89, 2024-12-28:
@chennes This issue can be closed, as we have determined that the performance problem is related to the default Deviation value being set to 0.5%, and a new one has been opened for the geometric processing issue caused by the Kernel, 'we think

tarman3, 2024-12-28:
You created this issue, so you can close by yourself

luisDias89, 2024-12-28:
ty, I hadn't noticed that. closed

closed 2024-12-28 by luzpaz