FreeCAD/FreeCAD #18744: Draft Move / BIM Copy: Constraining only to one axis zeroes both other axes

maxwxyz, 2024-12-27:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

With Draft_Move or BIM_Copy you can move objects either in relative or global mode.
When pressing the `X` key, it only moves in the X-axis and zeroes out Y and Z axis. This only works (and makes sense) in `Relative` (delta) mode.

Without relative checked, in global mode, e.g. to move it to a defined position in X, when pressing `X` both other axes will be set to their global 0, so the part moves in all directions and centers to Y and Z = 0. When switching out of relative mode, it should freeze the global values of the other two axes which are not chosen, so the part only moves in the direction the user intends.

### Full version info

```shell
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.39708 (Git) Conda
Build type: Release
Branch: main
Hash: 173ff4da1ef4f075d56c0dc617d1bf00d5f49d4b
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * CfdOF 1.29.4
  * CurvedShapes 1.0.13
  * Curves 0.6.52
  * dodo 1.0.1
  * fasteners 0.5.33
  * freecad.gears 1.3.0
  * FreeCAD_SketchArch (Disabled)
  * Manipulator 1.5.7
  * OpenTheme 2024.9.1
  * OpticsWorkbench 1.0.26
  * Rocket 4.0.1
  * sheetmetal 0.6.11
  * Silk 0.1.5
```

### Subproject(s) affected?

Draft

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-12-27:
@Roy-043 FYI

maxwxyz, 2024-12-27:
Same applies for Arch_Wall, BIM_Door, Arch_Window, Arch_Curtain_Wall, Arch_Panel,... @yorikvanhavre FYI

Roy-043, 2024-12-27:
There is perhaps a misunderstanding.  Global and relative are not mutually exclusive. Global means the global CS is used, as opposed to the working plane CS. Relative means coordinates are measured from the previous point, as opposed to from the origin of the CS. Please provide clearer steps.

maxwxyz, 2024-12-27:
@Roy-043 ok then in Global absolute mode (Relative not checked) it zeroes all other axes than the one I want to adjust.
Let's say I have a window in any position and want to center it to global X = 0 and do not want to change any other axis (Y and Z should stay the same).
As soon as I choose to move it in global mode (set the choosen point to global X 0) and press X, to only change the X value (because as soon as I move the mouse cursor, all other values change and the original position is lost in the task dialog) the values of Y and Z are set to zero, but I just wanted to move it in X.
This is also annoying when trying to move it in the 3D view or use snapping: as soon as relative is not checked and global is chosen, pressing the desired axis which should be the only one to change, all other 2 change to 0.

Roy-043, 2024-12-27:
OK, the zero values in the GUI are indeed confusing, but the result of the move should be correct.

maxwxyz, 2024-12-27:
They are not only confusing, but the selected object also moves in these two axes to global 0 ...

Roy-043, 2024-12-27:
Are you perhaps holding down the X key?

maxwxyz, 2024-12-27:
No. Just try it: Select object, hit X, enter a value in X and click OK. It moves to Y and Z at 0
Demo:

https://github.com/user-attachments/assets/18b89fff-ca42-4d85-aeb2-7fa373c6dab7

Roy-043, 2024-12-27:
Confirmed.

maxwxyz, 2024-12-29:
this also happens in local mode, so basically always when relative is unchecked

PR #19196 (2025-01-23): Draft: DraftGui.py fix displayPoint
Fixes #18744.
commit 07689920 (2025-01-23): Draft: DraftGui.py fix displayPoint

Fixes #18744.
closed 2025-01-23 by yorikvanhavre