FreeCAD/FreeCAD #18745: Crash while trim circle

tarman3, 2024-12-27:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

FreeCAD crash while trim circle

https://github.com/user-attachments/assets/e6705815-9acc-4c74-8793-ee8c5fb85afe

### Full version info

```shell
<details>
  <summary>FreeCAD 1.1.0dev.39673 (Git) Conda AppImage, Arch Linux (KDE/plasma/xcb)</summary>

OS: Arch Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.39741 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: 0d128517dedd154bbecfbabe67094ad56c8d46f5
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: lattice2 1.0.0; CurvedShapes 1.0.13; MeshRemodel 1.10.35; freecad.gears 1.3.0; fasteners 0.5.33; sheetmetal 0.6.12; Silk 0.1.5; Curves 0.6.52
```

</details>
```

### Subproject(s) affected?

Sketcher

### Anything else?

[circle_crash.zip](https://github.com/user-attachments/files/18259126/circle_crash.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-12-27:
Confimed

<details><summary><b>Stacktrace for v1.1.0dev.39741 (Git) Conda AppImage</b></summary>

```
FreeCAD 1.1.0, Libs: 1.1.0devR39741 (Git)
(C) 2001-2024 FreeCAD contributors
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.

Sketcher constraint number 3 is malformed!
Program received signal SIGSEGV, Segmentation fault.
#0  /usr/lib/libc.so.6(+0x3d1d0) [0x7f77ac04c1d0]
#1  0x7f7756db185a in GCS::ConstraintP2LDistance::error() from /tmp/.mount_FreeCAgjLInH/usr/lib/Sketcher.so+0x5a
#2  0x7f7756da506c in void GCS::System::identifyConflictingRedundantConstraints<Eigen::SparseQR<Eigen::SparseMatrix<double, 0, int>, Eigen::COLAMDOrdering<int> > >(GCS::Algorithm, Eigen::SparseQR<Eigen::SparseMatrix<double, 0, int>, Eigen::COLAMDOrdering<int> > const&, std::map<int, int, std::less<int>, std::allocator<std::pair<int const, int> > > const&, std::map<int, int, std::less<int>, std::allocator<std::pair<int const, int> > > const&, std::vector<double*, std::allocator<double*> >&, Eigen::Matrix<double, -1, -1, 0, -1, -1>&, int, int, int&) from /tmp/.mount_FreeCAgjLInH/usr/lib/Sketcher.so+0xc8c
#3  0x7f7756da71ae in GCS::System::diagnose(GCS::Algorithm) from /tmp/.mount_FreeCAgjLInH/usr/lib/Sketcher.so+0x90e
#4  0x7f7756da7c1d in GCS::System::initSolution(GCS::Algorithm) from /tmp/.mount_FreeCAgjLInH/usr/lib/Sketcher.so+0x97d
#5  0x7f7756e3a757 in Sketcher::Sketch::setUpSketch(std::vector<Part::Geometry*, std::allocator<Part::Geometry*> > const&, std::vector<Sketcher::Constraint*, std::allocator<Sketcher::Constraint*> > const&, int) from /tmp/.mount_FreeCAgjLInH/usr/lib/Sketcher.so+0x387
#6  0x7f7756d13fde in Sketcher::SketchObject::solve(bool) from /tmp/.mount_FreeCAgjLInH/usr/lib/Sketcher.so+0x9e
#7  0x7f7756d196b6 in Sketcher::SketchObject::delConstraints(std::vector<int, std::allocator<int> >, bool) from /tmp/.mount_FreeCAgjLInH/usr/lib/Sketcher.so+0x276
#8  0x7f7756d25bec in Sketcher::SketchObject::trim(int, Base::Vector3<double> const&) from /tmp/.mount_FreeCAgjLInH/usr/lib/Sketcher.so+0xb7c
#9  0x7f7756de4a83 in Sketcher::SketchObjectPy::trim(_object*) from /tmp/.mount_FreeCAgjLInH/usr/lib/Sketcher.so+0x93
#10  0x7f7756de4ea4 in Sketcher::SketchObjectPy::staticCallback_trim(_object*, _object*) from /tmp/.mount_FreeCAgjLInH/usr/lib/Sketcher.so+0x24
#11  /tmp/.mount_FreeCAgjLInH/usr/bin/../lib/libpython3.11.so.1.0(+0x206068) [0x7f77ae806068]
#12  /tmp/.mount_FreeCAgjLInH/usr/bin/../lib/libpython3.11.so.1.0(_PyObject_MakeTpCall+0x26c) [0x7f77ae7e5b5c]
#13  /tmp/.mount_FreeCAgjLInH/usr/bin/../lib/libpython3.11.so.1.0(_PyEval_EvalFrameDefault+0x6f2) [0x7f77ae7f1532]
#14  /tmp/.mount_FreeCAgjLInH/usr/bin/../lib/libpython3.11.so.1.0(+0x2abd1d) [0x7f77ae8abd1d]
#15  /tmp/.mount_FreeCAgjLInH/usr/bin/../lib/libpython3.11.so.1.0(PyEval_EvalCode+0xa4) [0x7f77ae8ab794]
#16  /tmp/.mount_FreeCAgjLInH/usr/bin/../lib/libpython3.11.so.1.0(+0x2cabf3) [0x7f77ae8cabf3]
#17  /tmp/.mount_FreeCAgjLInH/usr/bin/../lib/libpython3.11.so.1.0(+0x2c669a) [0x7f77ae8c669a]
#18  /tmp/.mount_FreeCAgjLInH/usr/bin/../lib/libpython3.11.so.1.0(PyRun_StringFlags+0x69) [0x7f77ae8ba9c9]
#19  0x7f77adb4653c in Base::InterpreterSingleton::runString[abi:cxx11](char const*) from /tmp/.mount_FreeCAgjLInH/usr/bin/../lib/libFreeCADBase.so+0x6c
#20  0x7f77af1beb34 in Gui::Command::_runCommand(char const*, int, Gui::Command::DoCmd_Type, char const*) from /tmp/.mount_FreeCAgjLInH/usr/bin/../lib/libFreeCADGui.so+0x94
#21  0x7f77af1becc3 in Gui::Command::_doCommand(char const*, int, Gui::Command::DoCmd_Type, char const*, ...) from /tmp/.mount_FreeCAgjLInH/usr/bin/../lib/libFreeCADGui.so+0xe3
#22  0x7f77095f5662 in void Gui::cmdAppObjectArgs<int&, double&, double&>(App::DocumentObject const*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, int&, double&, double&) from /tmp/.mount_FreeCAgjLInH/usr/lib/SketcherGui.so+0x1a2
#23  /tmp/.mount_FreeCAgjLInH/usr/lib/SketcherGui.so(+0x207204) [0x7f7709607204]
#24  /tmp/.mount_FreeCAgjLInH/usr/lib/SketcherGui.so(+0x20733d) [0x7f770960733d]
#25  0x7f7709772bd6 in SketcherGui::ViewProviderSketch::mouseButtonPressed(int, bool, SbVec2s const&, Gui::View3DInventorViewer const*) from /tmp/.mount_FreeCAgjLInH/usr/lib/SketcherGui.so+0x626
#26  0x7f77af5cd860 in Gui::ViewProvider::eventCallback(void*, SoEventCallback*) from /tmp/.mount_FreeCAgjLInH/usr/bin/../lib/libFreeCADGui.so+0x160
#27  0x7f77abad346e in SoEventCallback::handleEvent(SoHandleEventAction*) from /tmp/.mount_FreeCAgjLInH/usr/bin/../lib/./libCoin.so.80+0xbe
#28  0x7f77ab8d54e8 in SoAction::traverse(SoNode*) from /tmp/.mount_FreeCAgjLInH/usr/bin/../lib/./libCoin.so.80+0x68
#29  0x7f77aba8cb9a in SoChildList::traverse(SoAction*, int, int) from /tmp/.mount_FreeCAgjLInH/usr/bin/../lib/./libCoin.so.80+0xea
#30  0x7f77abadc348 in SoGroup::doAction(SoAction*) from /tmp/.mount_FreeCAgjLInH/usr/bin/../lib/./libCoin.so.80+0x58
#31  0x7f77abaf713a in SoSeparator::doAction(SoAction*) from /tmp/.mount_FreeCAgjLInH/usr/bin/../lib/./libCoin.so.80+0x2a
#32  0x7f77ab8d54e8 in SoAction::traverse(SoNode*) from /tmp/.mount_FreeCAgjLInH/usr/bin/../lib/./libCoin.so.80+0x68
#33  0x7f77aba8cb9a in SoChildList::traverse(SoAction*, int, int) from /tmp/.mount_FreeCAgjLInH/usr/bin/../lib/./libCoin.so.80+0xea
#34  0x7f77abadc348 in SoGroup::doAction(SoAction*) from /tmp/.mount_FreeCAgjLInH/usr/bin/../lib/./libCoin.so.80+0x58
[... cut ...]
#69  /usr/lib/libc.so.6(+0x25e08) [0x7f77ac034e08]
#70  /usr/lib/libc.so.6(__libc_start_main+0x8c) [0x7f77ac034ecc]
#71  /tmp/.mount_FreeCAgjLInH/usr/bin/freecad(+0x7c2a) [0x56300be69c2a]
```

</details>

OS: Manjaro Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.39741 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: 0d128517dedd154bbecfbabe67094ad56c8d46f5
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion

wwmayer, 2024-12-27:
With v1.0 it works correctly. So, it must be a recent regression in v1.1

mwganson, 2024-12-27:
No crash in Windows, but I get this report view error:

16:20:21  pyException: Traceback (most recent call last):
  File "<string>", line 1, in <module>
<class 'OSError'>: Access violation
16:20:21  circle_crash#Sketch002: App.getDocument('circle_crash').getObject('Sketch002').trim(0,App.Vector(749.315211,9.774529,0))
16:20:21  circle_crash#Sketch002: Failed to trim edge

OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.39708 (Git) Conda
Build type: Release
Branch: main
Hash: 173ff4da1ef4f075d56c0dc617d1bf00d5f49d4b
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Light/Fusion

depthoffocus, 2024-12-28:
This does not crash in 39747 on the Mac. There are still other issues relating to trimming (#18497 ) that I can't help think might be related)

OS: macOS 14.7.2
Architecture: x86_64
Version: 1.1.0dev.39747 (Git) Conda
Build type: Release
Branch: main
Hash: 7083362acb8529017b58ea03bf7416da0192b25b
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * ThreadProfile 1.92.0
  * freecad.gears 1.3.0
  * OpenTheme 2024.9.1
  * fasteners 0.5.34
  * lattice2 1.0.0
  * CurvedShapes 1.0.13
  * Curves 0.6.52

wwmayer, 2024-12-28:
> <class 'OSError'>: Access violation

This means it actually crashes, too. Thanks to a special compiler option the segfault is translated into an exception.

AjinkyaDahale, 2025-01-07:
Crash not observed in the following, but there is a spurious report of some constraint being malformed (probably in an intermediate step).

```
OS: Ubuntu Core 22 (GNOME/gnome/xcb)
Architecture: x86_64
Version: 1.1.0dev.39788 (Git) Snap 1253
Build type: Release
Branch: main
Hash: f34a8b5bfdeb489301c847ac4108ac31b0205907
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
```

Also, seems to have been fixed in #18665.

AjinkyaDahale, 2025-01-07:
> There are still other issues relating to trimming (#18497) that I can't help but think might be related, since trimming-related crashes seem to have turned up at about the same time.

Wouldn't be surprised if it's caused by #15811. However, that was necessary since trim before was pretty much an 800 line behemoth and is being effectively rewritten.

AjinkyaDahale, 2025-03-13:
Cannot reproduce in
```
OS: Ubuntu Core 22 (GNOME/gnome/xcb)
Architecture: x86_64
Version: 1.1.0dev.40711 (Git) Snap 1337
Build date: 2025/03/12 16:27:41
Build type: Release
Branch: main
Hash: 2ad23cec23de917dab14f08fa6232410169a4ed2
Python 3.10.12, Qt 5.15.15, Coin 4.0.0, Vtk 7.1.1, IfcOpenShell 0.8.0, OCC 7.7.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/Fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/92.6073/1
Installed mods: 
  * OpenTheme 2024.9.1
```
The "malformed constraints" issue I mentioned is also no longer observed (I don't recall what was happening but I don't see anything now).

johnsonm, 2025-04-22:
I have reproduced what I understand to be this issue reliably recently in multiple builds, including most recently one of current main `3b89cd23`

[segfault.zip](https://github.com/user-attachments/files/19842341/segfault.zip)

Editing the sketch and trimming the portion of the circle between the two node highlights reliably crashes with a segfault for me.

[image]

```
Program received signal SIGSEGV, Segmentation fault.
#0  /lib64/libc.so.6(+0x1a050) [0x7f4817917050]
#1  0x7f472c0fb917 in bool Base::BaseClass::is<Part::GeomCircle>() const from /var/home/johnsonm/d/FreeCAD/build/Mod/Sketcher/Sketcher.so+0x1b
#2  0x7f472c0a81b6 in Sketcher::SketchObject::isClosedCurve(Part::Geometry const*) from /var/home/johnsonm/d/FreeCAD/build/Mod/Sketcher/Sketcher.so+0x18
#3  0x7f472c0aee62 in Sketcher::SketchObject::trim(int, Base::Vector3<double> const&) from /var/home/johnsonm/d/FreeCAD/build/Mod/Sketcher/Sketcher.so+0x47e
#4  0x7f472c314886 in Sketcher::SketchObjectPy::trim(_object*) from /var/home/johnsonm/d/FreeCAD/build/Mod/Sketcher/Sketcher.so+0xa4
#5  0x7f472c305021 in Sketcher::SketchObjectPy::staticCallback_trim(_object*, _object*) from /var/home/johnsonm/d/FreeCAD/build/Mod/Sketcher/Sketcher.so+0xf3
#6  /lib64/libpython3.13.so.1.0(+0x174cee) [0x7f481a620cee]
#7  /lib64/libpython3.13.so.1.0(_PyObject_MakeTpCall+0x2ab) [0x7f481a5e70cb]
#8  /lib64/libpython3.13.so.1.0(_PyEval_EvalFrameDefault+0x4ab5) [0x7f481a6020b5]
#9  /lib64/libpython3.13.so.1.0(PyEval_EvalCode+0xab) [0x7f481a6d062b]
#10  /lib64/libpython3.13.so.1.0(+0x247740) [0x7f481a6f3740]
#11  /lib64/libpython3.13.so.1.0(+0x241963) [0x7f481a6ed963]
#12  /lib64/libpython3.13.so.1.0(+0x23437e) [0x7f481a6e037e]
#13  0x7f481aeb44e2 in Base::InterpreterSingleton::runString[abi:cxx11](char const*) from /var/home/johnsonm/d/FreeCAD/build/lib/libFreeCADBase.so+0xfc
#14  0x7f481cb524c6 in Gui::Command::_runCommand(char const*, int, Gui::Command::DoCmd_Type, char const*) from /var/home/johnsonm/d/FreeCAD/build/lib/libFreeCADGui.so+0xc6
#15  0x7f481cb5212b in Gui::Command::_doCommand(char const*, int, Gui::Command::DoCmd_Type, char const*, ...) from /var/home/johnsonm/d/FreeCAD/build/lib/libFreeCADGui.so+0x10f
#16  0x7f472a923358 in void Gui::cmdAppObjectArgs<int&, double&, double&>(App::DocumentObject const*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, int&, double&, double&) from /var/home/johnsonm/d/FreeCAD/build/Mod/Sketcher/SketcherGui.so+0x141
#17  0x7f472a8ff68e in SketcherGui::DrawSketchHandlerTrimming::executeCommands(Base::Vector2d) from /var/home/johnsonm/d/FreeCAD/build/Mod/Sketcher/SketcherGui.so+0x136
#18  0x7f472a8ff550 in SketcherGui::DrawSketchHandlerTrimming::releaseButton(Base::Vector2d) from /var/home/johnsonm/d/FreeCAD/build/Mod/Sketcher/SketcherGui.so+0x4a
#19  0x7f472ab2d8b9 in SketcherGui::ViewProviderSketch::mouseButtonPressed(int, bool, SbVec2s const&, Gui::View3DInventorViewer const*) from /var/home/johnsonm/d/FreeCAD/build/Mod/Sketcher/SketcherGui.so+0xe9d
#20  0x7f481d117a2a in Gui::ViewProvider::eventCallback(void*, SoEventCallback*) from /var/home/johnsonm/d/FreeCAD/build/lib/libFreeCADGui.so+0x44c
#21  0x7f481a0ab9e7 in SoEventCallback::handleEvent(SoHandleEventAction*) from /lib64/libCoin.so.80+0xb7
#22  0x7f4819ecd4e8 in SoAction::traverse(SoNode*) from /lib64/libCoin.so.80+0x68
#23  0x7f481a06b808 in SoChildList::traverse(SoAction*, int, int) from /lib64/libCoin.so.80+0xf8
#24  0x7f481a0b5764 in SoGroup::doAction(SoAction*) from /lib64/libCoin.so.80+0x54
#25  0x7f481a0cae2c in SoSeparator::doAction(SoAction*) from /lib64/libCoin.so.80+0x2c
#26  0x7f481cf79825 in Gui::SoFCUnifiedSelection::handleEvent(SoHandleEventAction*) from /var/home/johnsonm/d/FreeCAD/build/lib/libFreeCADGui.so+0x51
#27  0x7f4819ecd4e8 in SoAction::traverse(SoNode*) from /lib64/libCoin.so.80+0x68
#28  0x7f481a06b808 in SoChildList::traverse(SoAction*, int, int) from /lib64/libCoin.so.80+0xf8
#29  0x7f481a0b5764 in SoGroup::doAction(SoAction*) from /lib64/libCoin.so.80+0x54
#30  0x7f481a0cae2c in SoSeparator::doAction(SoAction*) from /lib64/libCoin.so.80+0x2c
#31  0x7f4819ecd4e8 in SoAction::traverse(SoNode*) from /lib64/libCoin.so.80+0x68
#32  0x7f4819ed704b in SoHandleEventAction::beginTraversal(SoNode*) from /lib64/libCoin.so.80+0xbb
#33  0x7f4819ecdb85 in SoAction::apply(SoNode*) from /lib64/libCoin.so.80+0xc5
#34  0x7f481a081738 in SoEventManager::actuallyProcessEvent(SoEvent const*) from /lib64/libCoin.so.80+0x78
#35  0x7f481d0598f8 in SIM::Coin3D::Quarter::QuarterWidget::processSoEvent(SoEvent const*) from /var/home/johnsonm/d/FreeCAD/build/lib/libFreeCADGui.so+0x56
#36  0x7f481d064989 in SIM::Coin3D::Quarter::SoQTQuarterAdaptor::processSoEvent(SoEvent const*) from /var/home/johnsonm/d/FreeCAD/build/lib/libFreeCADGui.so+0x1b5
#37  0x7f481d092599 in Gui::View3DInventorViewer::processSoEventBase(SoEvent const*) from /var/home/johnsonm/d/FreeCAD/build/lib/libFreeCADGui.so+0x23
#38  0x7f481d0f60f3 in Gui::NavigationStyle::processSoEvent(SoEvent const*) from /var/home/johnsonm/d/FreeCAD/build/lib/libFreeCADGui.so+0xab
#39  0x7f481d103e8f in Gui::GestureNavigationStyle::processSoEvent_bypass(SoEvent const*) from /var/home/johnsonm/d/FreeCAD/build/lib/libFreeCADGui.so+0x23
#40  0x7f481d1057ee in Gui::GestureNavigationStyle::AwaitingMoveState::react(Gui::GestureNavigationStyle::Event const&)::{lambda()#1}::operator()() const from /var/home/johnsonm/d/FreeCAD/build/lib/libFreeCADGui.so+0x40
[... cut ...]
#76  /var/home/johnsonm/d/FreeCAD/build/lib/libFreeCADGui.so(+0x1387b8) [0x7f481c9387b8]
#77  0x7f481c938aa8 in Gui::Application::runApplication() from /var/home/johnsonm/d/FreeCAD/build/lib/libFreeCADGui.so+0x1f4
#78  /var/home/johnsonm/d/FreeCAD/build/bin/FreeCAD() [0x402a20]
#79  /lib64/libc.so.6(+0x3248) [0x7f4817900248]
#80  /lib64/libc.so.6(__libc_start_main+0x8b) [0x7f481790030b]
#81  /var/home/johnsonm/d/FreeCAD/build/bin/FreeCAD() [0x401645]
```

If this is actually a different issue, let me know and I'll file a new bug report, of course.

xtemp09, 2025-04-22:
Well, johnsonm's crash seems to have been introduced recently.

Here is the backtrace.

<details>

```cpp
Thread 1 "FreeCAD" received signal SIGSEGV, Segmentation fault.
0x00007fffad381a3a in Base::BaseClass::is<Part::GeomCircle> (this=0x55555a754020)
    at /home/fyr/Downloads/FreeCAD/freecad-source/src/Base/BaseClass.h:188
188         return getTypeId() == T::getClassTypeId();
(gdb) bt
#0  0x00007fffad381a3a in Base::BaseClass::is<Part::GeomCircle> (this=0x55555a754020)
    at /home/fyr/Downloads/FreeCAD/freecad-source/src/Base/BaseClass.h:188
#1  0x00007fffad32714e in Sketcher::SketchObject::isClosedCurve (geo=0x55555a754020)
    at /home/fyr/Downloads/FreeCAD/freecad-source/src/Mod/Sketcher/App/SketchObject.cpp:1736
#2  0x00007fffad32e9b2 in Sketcher::SketchObject::trim (this=0x55555a7496c0, GeoId=0, point=...)
    at /home/fyr/Downloads/FreeCAD/freecad-source/src/Mod/Sketcher/App/SketchObject.cpp:3453
#3  0x00007fffad5d64b0 in Sketcher::SketchObjectPy::trim (this=0x555558ba3730, args=0x7fffc39efd40)
    at /home/fyr/Downloads/FreeCAD/freecad-source/src/Mod/Sketcher/App/SketchObjectPyImp.cpp:1380
#4  0x00007fffad5c5859 in Sketcher::SketchObjectPy::staticCallback_trim (self=0x555558ba3738, 
    args=0x7fffc39efd40) at /home/fyr/Downloads/FreeCAD/build-qt5/src/Mod/Sketcher/App/SketchObjectPy.cpp:3218
#5  0x00007ffff21df468 in ?? () from /lib/x86_64-linux-gnu/libpython3.12.so.1.0
#6  0x00007ffff21827af in _PyObject_MakeTpCall () from /lib/x86_64-linux-gnu/libpython3.12.so.1.0
#7  0x00007ffff211d5ee in _PyEval_EvalFrameDefault () from /lib/x86_64-linux-gnu/libpython3.12.so.1.0
#8  0x00007ffff22a086f in PyEval_EvalCode () from /lib/x86_64-linux-gnu/libpython3.12.so.1.0
#9  0x00007ffff22fb178 in ?? () from /lib/x86_64-linux-gnu/libpython3.12.so.1.0
#10 0x00007ffff22fb29c in ?? () from /lib/x86_64-linux-gnu/libpython3.12.so.1.0
#11 0x00007ffff22fe8b4 in PyRun_StringFlags () from /lib/x86_64-linux-gnu/libpython3.12.so.1.0
#12 0x00007ffff2e2d9ae in Base::InterpreterSingleton::runString[abi:cxx11](char const*) (this=0x555555632480, 
    sCmd=0x555558c82b18 "App.getDocument('segfault').getObject('Sketch').trim(0,App.Vector(-9.530665,3.027611,0))") at /home/fyr/Downloads/FreeCAD/freecad-source/src/Base/Interpreter.cpp:249
#13 0x00007ffff75cf0b9 in Gui::Command::_runCommand (
    file=0x7fffa3907ce0 "/home/fyr/Downloads/FreeCAD/freecad-source/src/Gui/CommandT.h", line=384, 
    eType=Gui::Command::Doc, 
    sCmd=0x555558c82b18 "App.getDocument('segfault').getObject('Sketch').trim(0,App.Vector(-9.530665,3.027611,0))") at /home/fyr/Downloads/FreeCAD/freecad-source/src/Gui/Command.cpp:721
#14 0x00007ffff75cec75 in Gui::Command::_doCommand (
    file=0x7fffa3907ce0 "/home/fyr/Downloads/FreeCAD/freecad-source/src/Gui/CommandT.h", line=384, 
    eType=Gui::Command::Doc, sCmd=0x7fffa3907cb0 "App.getDocument('%s').getObject('%s').%s")

```

```ruby
OS: Ubuntu 24.04.1 LTS (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.41308 (Git)
Build date: 2025/04/18 04:41:18
Build type: Unknown
Branch: main
Hash: 6be09c29f6217c6338cf09dcc9a658454287bd83
Python 3.12.3, Qt 5.15.13, Coin 4.0.2, Vtk 9.1.0, boost 1_83, Eigen3 3.4.0, PySide 5.15.13
shiboken 5.15.13, SMESH 7.7.1.0, xerces-c 3.2.4, OCC 7.6.3
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/unset/Qt default
Logical DPI/Physical DPI/Pixel Ratio: 96.0584/96.0584/1
```

<details>

johnsonm, 2025-04-22:
Oops, I missed the hidden stack trace in @luzpaz 's post earlier. Not seeing it, I didn't realize it was at least different-looking. Should I open a separate issue?

johnsonm, 2025-04-28:
With #20951 in place, no need for additional noise from me.

depthoffocus, 2025-05-10:
In 41628 I can't reproduce the OP's segfault, and @johnsonm 's is gone too of course.

AjinkyaDahale, 2025-07-20:
@maxwxyz I think this can be closed.

closed 2025-07-20 by luzpaz