FreeCAD/FreeCAD #18747: Crash on MacOS when opening the Preferences

titanic-jb, 2024-12-27:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Working on MacOS 15.1.1 with the latest FreeCAD for ARM, it crashes, whenever I try to open the Preferences, either by using the keyboard shortcut CMD-, or by clicking on the menues.

### Full version info

```shell
OS: macOS 15.1.1
Architecture: arm64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: unset/Classic/Qt default
Installed mods: 
  * DynamicData 2.62.0
  * BillOfMaterials 0.0.8.2
  * freecad.gears 1.2.0
  * fasteners 0.5.21
  * sheetmetal 0.4.14
  * ExtremeProLight 1.5.0
  * Assembly4 0.50.12
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

titanic-jb, 2024-12-27:
The attached ZIP-file contains the content of the automatic generated Bug Report otherwise sent to Apple.

[bugreport.zip](https://github.com/user-attachments/files/18259497/bugreport.zip)

wwmayer, 2024-12-27:
I cannot confirm the crash but according to the callstack it crashes inside `DlgPreferencesImp::createPreferencePage` that invokes the lambda function:
```
    auto resetMargins = [](QWidget* widget) {
        widget->setContentsMargins(0, 0, 0, 0);
        widget->layout()->setContentsMargins(0, 0, 0, 0);
    };
```
This is indeed unsafe code because there is no guarantee that a widget has set a layout.

menkej, 2024-12-27:
I can not confirm the crash on m3 pro. Wild guess: maybe something is messed up with your app or dialog settings, eventually by using different FreeCad Versions. AFAIR they all use the same settings folder. Try backing up/removing your /Users/YourUser/Library/Preferences/FreeCAD folder and see if the problem persists.

titanic-jb, 2024-12-28:
> Try backing up/removing your /Users/YourUser/Library/Preferences/FreeCAD folder and see if the problem persists.

 Unfortunately, it does. :-(

[bugreport2.zip](https://github.com/user-attachments/files/18266122/bugreport2.zip)

hyarion, 2025-01-08:
A fix for 1.1.0dev is merged and this should be fixed in latest dev build.
The PR is marked for backport so it will most likely be fixed in a stable 1.0.1 too if such a release is made

hyarion, 2025-01-18:
@titanic-jb can you see if this was still occurs?

titanic-jb, 2025-01-22:
> [@titanic-jb](https://github.com/titanic-jb) can you see if this was still occurs?

Sorry, if this question sounds stupid to you.

Actually, the download page offers me the same version, I have already installed, therefore I doubt, that you ment to download that same version again, did you?

Hence, I I guess, I have to download (and install :-)) the developer version 1.1.0dev, right?

titanic-jb, 2025-01-22:
> > [@titanic-jb](https://github.com/titanic-jb) can you see if this was still occurs?
> 
> Hence, I I guess, I have to download (and install :-)) the developer version 1.1.0dev, right?

I installed the latest macOS build of yesterday, but I wasn't able to start it at all.  Finder checked the app and found an error and suggested to move the app into trash.  :-(

hyarion, 2025-01-22:
> I installed the latest macOS build of yesterday, but I wasn't able to start it at all. Finder checked the app and found an error and suggested to move the app into trash. :-(

> [!TIP]
>  On macOS arm machines, there is an installation issue. Until the new installation setup is resolved, to test the weekly builds, the user needs to open a Terminal window and execute the following command:
> `sudo xattr -r -d com.apple.quarantine /Applications/FreeCAD.app`
>
> from: https://github.com/FreeCAD/FreeCAD-Bundle/issues/360#issuecomment-2561365902

titanic-jb, 2025-01-23:
> On macOS arm machines, there is an installation issue. Until the new installation setup is resolved, to test the weekly builds, the user needs to open a Terminal window and execute the following command: `sudo xattr -r -d com.apple.quarantine /Applications/FreeCAD.app`

This helped.  I was able to start FreeCAD and afterwards was able to open the Preferences without crash.  Thank you. ❤

hyarion, 2025-01-23:
@maxwxyz you can close this issue as OP has verified that the issue has been fixed

closed 2025-01-23 by maxwxyz