FreeCAD/FreeCAD #18748: BIM: No possibility to create curved stairs

maxwxyz, 2024-12-27:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

In BIM it is not possible to create curved stairs. It is only possible to create stairs from straight line segments.
It would be great if arcs / circles or maybe even B-splines could be supported as base feature for stairs.

These are some examples (spiral staircase or a slightly curved stair):
[image]

[image]

@yorikvanhavre FYI

### Full version info

```shell
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.39708 (Git) Conda
Build type: Release
Branch: main
Hash: 173ff4da1ef4f075d56c0dc617d1bf00d5f49d4b
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * CfdOF 1.29.4
  * CurvedShapes 1.0.13
  * Curves 0.6.52
  * dodo 1.0.1
  * fasteners 0.5.33
  * freecad.gears 1.3.0
  * FreeCAD_SketchArch (Disabled)
  * Manipulator 1.5.7
  * OpenTheme 2024.9.1
  * OpticsWorkbench 1.0.26
  * Rocket 4.0.1
  * sheetmetal 0.6.11
  * Silk 0.1.5
```

### Subproject(s) affected?

BIM

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

paullee0, 2024-12-31:
There were a number of discussion indeed on Stairs, but not sure if there is any on curve flight - there were ideas on how curve outline on both sides of a flight, but the flight is still straight.

I would really like to implement curved stairs that currently trying to refactor the Stairs tool to make it work on a complex sketch.  Hope there are peoples interested.

BTW, I am trying to find way to let Sketch to show additional sign (e.g. arrow to represent direction of edge) which would help with feature like this one.

maxwxyz, 2024-12-31:
@FlachyJoe might help. Otherwise the arrow could be made of lines with construction geometry

paullee0, 2024-12-31:
Thanks @maxwxyz.  Hope @FlachyJoe may provide some pointers, thanks in advance :)

FlachyJoe, 2025-01-04:
@paullee0 you can give a look to #9257 and #9703 to see how to add a sketch overlay.

paullee0, 2025-01-05:
Thanks @FlachyJoe, I have a look and understand it was implemented in C++, right?

Any possible way to do this with Python?

FlachyJoe, 2025-01-05:
> Any possible way to do this with Python?

I don't think so. The whole Sketcher stuff is written in C++.

paullee0, 2025-01-05:
Thanks, hope something can be done in Sketcher::**SketchObjectPython**

I use this rather than Sketcher::SketchObject

reset12, 2025-01-06:
Try to use a Macro from [Macro ](https://wiki.freecad.org/Macros_recipes)  Object Creation
if it fit your needs.

paullee0, 2025-01-30:
@maxwxyz I have been refactoring ArchStairs.py since earlier this year but not submitted here yet.  I learnt a few git command like git fetch, merge, commit, push, but not much about github.

Not sure if any others are working on the same concurrently and all important issues about Stairs, any comments as to how development could be done collaboratively before one may push a commit and discussed important issues?  

Thanks :)

yorikvanhavre, 2025-02-04:
There is no hard rule there... The key is, as much as possible, to do your changes as incrementally as possible... Instead of one big PR that changes the whole file, several smaller PRs that are easier to handle, and if someone else is working on the same file at the same time, there are more chances that git manages to merge everything nicely. But I don't think anyone else is working on the stairs at the moment.