FreeCAD/FreeCAD #18749: BIM: No possibility to create curved doors or windows

maxwxyz, 2024-12-27:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

It is currently not possible to create curved or non-planar doors or windows (neither hinged, fixed, nor just an opening) on a curved wall segment or any arbitrary shaped face as base feature.
Ideally you could select a face or projected sketch as base feature to create framing and door/window based on the curvature of the host object, a defined set of wires (sketch projected to the wall), or a face on the wall.

Example of curved windows/doors with constant profile in Z-direction:

[image]
[image]
[image]

  
Surface based window example:
[image]

### Full version info

```shell
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.39708 (Git) Conda
Build type: Release
Branch: main
Hash: 173ff4da1ef4f075d56c0dc617d1bf00d5f49d4b
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * CfdOF 1.29.4
  * CurvedShapes 1.0.13
  * Curves 0.6.52
  * dodo 1.0.1
  * fasteners 0.5.33
  * freecad.gears 1.3.0
  * FreeCAD_SketchArch (Disabled)
  * Manipulator 1.5.7
  * OpenTheme 2024.9.1
  * OpticsWorkbench 1.0.26
  * Rocket 4.0.1
  * sheetmetal 0.6.11
  * Silk 0.1.5
```

### Subproject(s) affected?

BIM

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Harshit-kumar1234, 2024-12-27:
for curved doors can't we use the additive pipe feature by just creating a rectangle and sketching a part and using additive pipe we can our desired 3d curved door.

paullee0, 2024-12-31:
See https://wiki.freecad.org/Arch_Window#Defining_window_types

There is all the flexibility in FC to create whatever shape of window/door users want.  Window/Door not created with the Window's task panel type should have no hinge support though.

yorikvanhavre, 2025-01-06:
> Window/Door not created with the Window's task panel type should have no hinge support though

That might be an interesting feature to add, though, if the object has distinct solids, and by allowing users to define an edge to be used as a hinge...

Roy-043, 2026-08-27:
It is also possible to create a hinged custom window. Open the attached file (remove the .zip extension first) and change the Opening property of the window. I am therefore going to close this issue.

[custom_window_opening.FCStd.zip](https://github.com/user-attachments/files/31533016/custom_window_opening.FCStd.zip)

maxwxyz, 2026-08-28:
But you need to manually model the element, not based on a sketch which will be projected onto the host object.
I have rephrased the issue.