FreeCAD/FreeCAD #18750: Feature request: Sketcher: Allow arc constraints to helper lines.

Panda631, 2024-12-27:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Especially, when an arc is small (e.g. fillets) it would be easyer to constrain the green helper lines instead trying to pick the tiny rest of an arc.

Not a bug, but sometimes helpfull.

[image]

### Full version info

```shell
OS: Windows 11 build 22631
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Installed mods: 
  * CurvedShapes 1.0.13
  * Curves 0.6.52
  * EM 2.1.1
  * fasteners 0.5.33
  * Help 1.0.3
  * kicadStepUpMod 10.17.1
  * Pyramids-and-Polyhedrons
  * Render 2024.12.15
  * sheetmetal 0.2.61
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-12-27:
Can you mockup what the feature would look like implemented ?

Panda631, 2024-12-27:
Actually, you need to select the tiny arc to constrain the diameter.

[image]

Allowing to select the green helper circle (hope I got the correct wording) instead of the arc , the selection would be much easyer.

[image]