FreeCAD/FreeCAD #18751: conda scripts do not work with mamba major version 2

j-hap, 2024-12-27:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I use a conda environment to build freecad. I just updated the base environment and got the mamba 2 major release. That version does not support the `--live-stream` option, therefore the scripts 
```
conda/cmake.sh
conda/setup-environment.sh
conda/setup-environment-qt6.sh
```
and probably also `conda/cmake.cmd` (did not test on windows) fail.

Other things I noticed with the scripts:

Both `conda/setup-environment.sh` and `conda/setup-environment-qt6.sh` start with
`mamba env create -p .conda/freecad -f conda/conda-env.yaml`
but it should probably be
`mamba env create -p .conda/freecad -f conda/conda-env-qt6.yaml`
in the second one.

I use the conda build environment in fedora 41 and do a qt 6 build. The 
`conda/environment-qt6.devenv.yml` was missing the packages
```
- libmed
- openmpi-mpicxx
- libboost-python
```
for a successful cmake configure.

### Full version info

```shell
commit 6bb424b, it's a build issue so no specific freecad version
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

oursland, 2024-12-28:
It hasn't been deprecated as documentation wasn't prepared for `pixi`, but I also haven't been testing with the `conda` builds.  For a while they were disabled on CI due to occasional errors.

A PR for documentation on building with `pixi` is up at https://github.com/FreeCAD/DevelopersHandbook/pull/113

JohnOCFII, 2024-12-28:
As a frequent user of your legacy condo build instructions, I was able to install -pixi- and follow your new instructions to successfully build FreeCAD on my Mac.

Only suggestion (outside of your current PR) would be to more completely integrate the -pixi- instructions with the "Setting up for Development" steps, as some of those are required before you actually use the Pixi instructions (e.g., forking and cloning FreeCAD, creating build directory, etc.).

```
OS: macOS Sequoia (15.2)
Architecture: arm64
Version: 1.1.0dev.39747 (Git) Conda
Build type: Release
Branch: main
Hash: 7083362acb8529017b58ea03bf7416da0192b25b
Python 3.12.0, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/
Installed mods: 
  * OpenTheme 2024.9.1
```

oursland, 2024-12-28:
@JohnOCFII I agree with that.  In addition, some have suggested recommending Pixi as the preferred solution as it supports all three major platforms (Windows, macOS, and Linux) using the same commands.  With the inclusion of `pixi.toml`, I have also discovered other devs had already been using Pixi in their development in conjunction with their preferred IDE.

j-hap, 2025-01-03:
I too was able to build successfully with pixi (fedora 41), so if the conda helpers will be deprecated anyway, this issue can be closed

closed 2025-01-03 by maxwxyz