FreeCAD/FreeCAD #18755: BIM: IFC import disappears when moving

maxwxyz, 2024-12-27:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Forum report: https://forum.freecad.org/viewtopic.php?t=93318
---
I made a window and exported as IFC file after I created new file and import this IFC file. It is imported but when I move the window it disappears.Steps to reproduce:
1. Open new BIM file
2. import ifc file
3. Move it with draft_move tool

Demo file:
[custom_window-Window_ifc.zip](https://github.com/user-attachments/files/18261272/custom_window-Window_ifc.zip)

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: Turkish/Turkey (tr_TR)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * dxf-library
  * Manipulator 1.5.7
```

### Subproject(s) affected?

BIM

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

semhustej, 2024-12-27:
It seems to me the IFC project is by default loaded as root only. You have to double-click the project to load the building and than double-click the building to load the building content - the window. Only after that can you move the window.

If you do not do above mentioned steps, you are moving the while IFC project which leads to window disappearing. To make the window visible again, you have to double click the whole project to load the building. This behavior is user-unfriendly, not sure what is possible to do when the objects are loaded only in root mode.

See behavior in video:

https://github.com/user-attachments/assets/2d402236-e56e-497a-bfc3-c91c4a469c99

maxwxyz, 2024-12-28:
Thanks! I've updated the forums link

semhustej, 2024-12-29:
I would not close this issue as I think this should be handled by FreeCAD in a better way. When objects disappear, it is not user friendly and causes confusion. I think that if the situation stays this way, there will be a lot of future bug reports regarding this issue when people start using the IFC import more.

paullee0, 2024-12-31:
It seems to me that one of the confusion comes from the name (should be of the project) it automatically generates - "custom_window".  People would think it is the window object itself.

semhustej, 2025-01-01:
> It seems to me that one of the confusion comes from the name (should be of the project) it automatically generates - "custom_window". People would think it is the window object itself.

Even if the name if the file was different, it would not make sense for the window to disappear when you move the whole project.