FreeCAD/FreeCAD #18756: Sketcher: "Invalid solution from DogLeg solver" error

maxwxyz, 2024-12-27:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Forum report: https://forum.freecad.org/viewtopic.php?t=93061
---
1. Create a sketch
2. Draw a rectangle
3. Use fillet or chamfer tool
4. Click rectangle's any corner point and apply fillet or chamfer
5. Right click and exit from fillet or chamfer command
6. Move the rectangle with mouse 

```
20:04:40  Updating geometry: Error build geometry(4): gp_Circ::SetRadius() - radius should be positive number
20:04:40  Invalid solution from DogLeg solver.
20:04:40  Updating geometry: Error build geometry(4): gp_Circ::SetRadius() - radius should be positive number
20:04:40  Invalid solution from DogLeg solver.
20:04:40  Updating geometry: Error build geometry(4): gp_Circ::SetRadius() - radius should be positive number
20:04:40  Invalid solution from DogLeg solver.
20:04:40  Updating geometry: Error build geometry(4): gp_Circ::SetRadius() - radius should be positive number
20:04:40  Invalid solution from DogLeg solver.

```  

[image]

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.39673 (Git) Conda
Build type: Release
Branch: main
Hash: 1289997dfd604cb24f060f905f04ec3728d964e4
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: Turkish/Turkey (tr_TR)
Stylesheet/Theme/QtStyle: ProDark.qss/ProDark/Qt default
Installed mods: 
  * Dracula 0.0.5
  * FreeCAD-themes 2024.7.24
  * ProDarkThemePreferencePack 1.0.0
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

tarman3, 2024-12-27:
Confirm

https://github.com/user-attachments/assets/6a0e7ae1-40d0-4c71-a044-cf0e0d8beb5b

<details>
  <summary>FreeCAD 1.1.0dev.39673 (Git) Conda AppImage, Arch Linux (KDE/plasma/xcb)</summary>

```
OS: Arch Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.39741 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: 0d128517dedd154bbecfbabe67094ad56c8d46f5
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: lattice2 1.0.0; CurvedShapes 1.0.13; MeshRemodel 1.10.35; freecad.gears 1.3.0; fasteners 0.5.33; sheetmetal 0.6.12; Silk 0.1.5; Curves 0.6.52
```

</details>

itchka, 2026-02-15:
This bug also manifests when editing the control polygons of B-splines and results ultimately in a segfault. 
When freecad is run from a terminal the following output is observed. This is from  a locally compiled version from the git release/1.1 branch commit 8175e0633fd

Updating geometry: Error build geometry(13): gp_Circ::SetRadius() - radius should be positive number
Invalid solution from DogLeg solver.
corrupted size vs. prev_size
Aborted                    (core dumped) freecad

FEA-eng, 2026-08-25:
Still relevant

```
OS: Windows 11 build 26200
Architecture: x86_64
Version: 26.3.0dev.20260720 (Git shallow)
Build date: 2026/07/20 16:51:42
Build type: Release
Branch: grafted,grafted
Hash: 057d51f846f535f4f0b48492e586be0a76b9efe2
Python 3.11.14, Qt 6.8.3, Coin 4.0.10 (bundled), Pivy 0.6.11 (bundled), Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, Clipper2 , IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL) [ OS: Polish/Poland (pl_PL) ]
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
QPA/File dialog/Color dialog: windows/via Win32/via Qt
Navigation Style/Orbit Style/Rotation Mode: Touchpad/Rounded Arcball/Window center
Logical DPI/Physical DPI/Pixel Ratio: 96/113.268/1.25
OpenGL version/vendor/renderer: 4.6.0 Compatibility Profile Context 26.5.2.260413/ATI Technologies Inc./AMD Radeon(TM) Graphics
```