FreeCAD/FreeCAD #18759: FEM: Elmer 3D magnetodynamic analysis failing material assignment check if >2 bodies in analysis

doug-harriman, 2024-12-27:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I am able to successfully run a 3D magnetic analysis if there are only 2 bodies in the analysis.  If a third body is added when using the Elmer Magnetodynamic solver, the GUI reports that "Solid3" does not have a material assigned to it and errors out before writing the `.sif` analysis configuration file.  I have repeatedly verified that all bodies have materials assigned in the GUI.  There seems to be an error in either tracking or reporting of material assignment for this solver.  This problem does not occur in the magnetodynamic2d version of the solver.

A file for error replication is attached.  If "Magnet-2" is removed from the analysis, it will run.
[Magnetics-3D.zip](https://github.com/user-attachments/files/18262166/Magnetics-3D.zip)

GUI error dialog:
[image]

### Full version info

```shell
OS: Windows 11 build 22631
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * FreeCAD-Gridfinity-Workbench 0.6.0
```

### Subproject(s) affected?

FEM

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

FEA-eng, 2024-12-28:
@marioalexis84 FYI

marioalexis84, 2024-12-30:
Here the problem comes from a limitation for selecting references from the 3D view.
First you must be sure to select references of the BooleanFragments object by hiding the Magnet-1, Magnet-2 and Air bodies.

The current selection panel does not allow to distinguish between the BooleanFragments solids connected by the selected Face, so you must indicate the solids manually in the properties panel for both MaterialSolid and MagnetizationMagnet: References -> BooleanFragments [Solid1, Solid3]

doug-harriman, 2024-12-30:
@marioalexis84 - Thank you for the guidance.  I was able to get past the error I reported above by following your instructions.  An example screen shot is shown below.

[image]

I believe that that is showing the proper selections.  Note that it shows "Body" instead of "Solid".  Not sure if that matters, but it is different.

I am still not able to get an analysis to run however.  If you look closely you'll see the reference is: `BooleanFragments [Body., Body001.]`.  Note that both referenced bodies end with a ".".  This seems to be causing an error when I attempt to write the analysis file in the SolverElmer Task dialog.  The error report is:

```
13:40:35  Exception in thread Thread-114 (protector):
13:40:35  Traceback (most recent call last):
13:40:35    File "C:\Program Files\FreeCAD 1.0\bin\Lib\threading.py", line 1045, in _bootstrap_inner
13:40:35      self.run()
13:40:35    File "C:\Program Files\FreeCAD 1.0\bin\Lib\threading.py", line 982, in run
13:40:35      self._target(*self._args, **self._kwargs)
13:40:35    File "C:\Program Files\FreeCAD 1.0\Mod\Fem\femsolver\task.py", line 116, in protector
13:40:35      self.run()
13:40:35    File "C:\Program Files\FreeCAD 1.0\Mod\Fem\femsolver\elmer\tasks.py", line 52, in run
13:40:35      self.check_material_exists()
13:40:35    File "C:\Program Files\FreeCAD 1.0\Mod\Fem\femsolver\run.py", line 420, in check_material_exists
13:40:35      objs = self.get_several_member("App::MaterialObjectPython")
13:40:35             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
13:40:35    File "C:\Program Files\FreeCAD 1.0\Mod\Fem\femsolver\run.py", line 405, in get_several_member
13:40:35      return membertools.get_several_member(self.analysis, t)
13:40:35             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
13:40:35    File "C:\Program Files\FreeCAD 1.0\Mod\Fem\femtools\membertools.py", line 128, in get_several_member
13:40:35      obj_dict["RefShapeType"] = femutils.get_refshape_type(m)
13:40:35                                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
13:40:35    File "C:\Program Files\FreeCAD 1.0\Mod\Fem\femtools\femutils.py", line 338, in get_refshape_type
13:40:35      first_ref_shape = get_element(first_ref_obj[0], first_ref_obj[1][0])
13:40:35                        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
13:40:35    File "C:\Program Files\FreeCAD 1.0\Mod\Fem\femtools\geomtools.py", line 177, in get_element
13:40:35      return part.Shape.getElement(element)  # Face, Edge, Vertex
13:40:35             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
13:40:35  ValueError: {'sclassname': 'class Base::ValueError', 'sErrMsg': 'Invalid shape name Body002.', 'sfile': 'C:\\bld\\freecad_1731959423943\\work\\src\\Mod\\Part\\App\\TopoShapeExpansion.cpp', 'iline': 2019, 'sfunction': 'class Part::TopoShape __cdecl Part::TopoShape::getSubTopoShape(const char *,bool) const', 'swhat': 'Invalid shape name Body002.', 'btranslatable': False, 'breported': False}
```

It looks like that trailing "." is causing issues when attempting to write the file.  

For completeness, here's what the dialog looks like when I select the objects:

[image]

I did retry generating everything.  The images above are for this file: [Magnetics-3D-try-2.zip](https://github.com/user-attachments/files/18278037/Magnetics-3D-try-2.zip)

marioalexis84, 2024-12-31:
> Note that it shows "Body" instead of "Solid". Not sure if that matters

Yes, it matters.
`Body` is an object with a shape. Solid, edge, face, vertex... are subelements of an object.

Since boundary conditions and constraints are applied on solids, faces, edges, etc, the references on the Fem objects expect to receive an object with a subelement, and not just the object.
To run the analysis, do the following:
`MagnetizationMagnet` and `MaterialFluid` seem to have been saved with some error and when I try to assign the BooleanFragments solids it throws an error, so first clear the references by opening the `References` property dialog and pressing `Clear` and `Ok`.
Once you clear the references, open the dialog again, click to the right of `BooleanFragments`, type on the keyboard `Solid1,Solid3`. Then make sure the `BooleanFragments` line is highlighted by pressing on `BooleanFragments` name.
Click `Ok`.

Repeat for `MaterialSolid`.

For `MaterialFluid` you can do the same with the `BooleanFragments` `Solid2`, but in this case it is possible to select the solid from the object's task panel since there is no ambiguity when selecting the solid using one of its faces.

You must be sure that all selections are on `BooleanFragments` subelements since it is the object on which the meshing is performed, regardless of whether other objects appear identical in the 3D view.

[image]

marioalexis84, 2024-12-31:
The Fem task panels for selections are being rewritten, so in a few days it will probably not be necessary to assign solids manually by typing from the keyboard.

doug-harriman, 2024-12-31:
Thank you for the detailed explanation.  I would not have figured that out myself.  For clarity, the names of the solids ("Solid1", "Solid2") are just based on the occurence of the Body in the Boolean fragment, correct?  The first body will be associated with "Solid2", second body is "Solid2", etc.  Correct?   I was not able to figure out any simple Python method to list the solid names.

Looking forward to the updated Fem task selection panels.  Is there an issue I can follow so that I can grab a daily build when the changes are in?

marioalexis84, 2024-12-31:
>the names of the solids ("Solid1", "Solid2") are just based on the occurence of the Body in the Boolean fragment

That guess might be correct (in this case it is), but there is no guarantee that it will always work because it depends on implementation details of the boolean fragments function.
Note also that the initial solids are not the same as the solids in the boolean fragments: the "air" is cut by the two discs.

> Is there an issue I can follow 

When the task panels are updated and merged into the main branch, this issue will be closed and you will receive a notification.

doug-harriman, 2024-12-31:
Last question: If the Solid identifier is not strictly associated with the Boolean Fragment order, how can one find the Solid identifier associated with each body in the Boolean Fragment?  I've been unable to find that information with web searches and unable to figure it out by examining the objects with the Python API.  That's a critical piece of information to getting the analysis correct.

FYI, I'm attempting to document the process at: https://github.com/doug-harriman/magnetic-fem.  Happy to contribute that to other documentation if it would be useful.

marioalexis84, 2025-01-01:
To be on the safe side, it's best to consider boolean fragments as the starting point.
You can track solids by comparing their faces to each face in the boolean fragment using `isPartner`, `isSame`, `isEqual` Part.Shape methods (see the documentation for these methods).
This code snippet might help you highlight the faces of a given solid in the 3D view:
```
def selector(solid, boolean_frag):
  output = []
  for f in solid.Faces:
    for k in enumerate(boolean_frag.Shape.Faces):
      if k[1].isSame(f):
        output.append("Face{}".format(k[0]+1))
  return output

bf = App.ActiveDocument.BooleanFragments
Gui.Selection.addSelection(bf, selector(bf.getSubObject("Solid1"), bf))
```

doug-harriman, 2025-01-04:
For reference, I've created a macro that attempts to map solid names to the bodies in the Boolean Fragment, macro attached:

[mapsolids.FCMacro.txt](https://github.com/user-attachments/files/18308635/mapsolids.FCMacro.txt)

The macro creates a custom property for each Body in the Boolean Fragment and lists the name of the solid that it determined matched in that property.

There are lots of issues with this macro, of particular note: 
* Very little error checking.  
* Automated face checks did not work.  Implemented a rudimentary face check using some geometric properties instead.   Likely there are lots of issues with this approach that could lead to bad matches.

doug-harriman, 2025-01-04:
For reference, I've created a macro that attempts to map solid names to the bodies in the Boolean Fragment, macro attached:

[mapsolids.FCMacro.txt](https://github.com/user-attachments/files/18308635/mapsolids.FCMacro.txt)

The macro creates a custom property for each Body in the Boolean Fragment and lists the name of the solid that it determined matched in that property.

There are lots of issues with this macro, of particular note: 
* Very little error checking.  
* Automated face checks did not work.  Did a rudamentary face check using some geometric properties.   Likely there are lots of issues with htis.

marioalexis84, 2025-03-06:
@maxwxyz this can be marked as solved.

@doug-harriman you will probably want to apply a compound filter to the BooleanFragments to remove the underlying Compound. You can check the shape composition using the `Check Geometry` command from Part module.

closed 2025-03-06 by maxwxyz