FreeCAD/FreeCAD #18763: Sketcher_Scale does not allow scaling< 1.0

ericvsmith, 2024-12-27:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I can't type a value < 1.0 nor enter a value of < 1.0 when entering the second line of Sketcher_Scale. I would expect to be able to enter a value of less than one in order to shrink a sketch. I could not find any documentation suggesting a value < 1.0 is required.

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * EvsWorkbench
  * fasteners 0.5.29
  * MeshRemodel 1.10.34
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

tarman3, 2024-12-27:
https://github.com/user-attachments/assets/f4f4158a-5a5c-4325-be82-aa85f36b281b

ericvsmith, 2024-12-27:
[sheep-yarn-bobbin.zip](https://github.com/user-attachments/files/18263600/sheep-yarn-bobbin.zip)

Interesting. If I create another sketch with just a square on it, it does work. I'm guessing my problem is related to the (many) ellipses on my sketch.

tarman3, 2024-12-27:
Try to use clone feature from Draft

https://github.com/user-attachments/assets/65fb5fd9-4170-4059-8162-68354d56ca79

ericvsmith, 2024-12-31:
Hmm, I thought I responded to this earlier, but now i don't see it. Thank you for the response. Creating a clone worked. I'm not sure if there's an actual bug here (and therefore I should leave this open), or if it's something I did and not a bug. I'll just leave it open and let someone else close it if appropriate.

luzpaz, 2025-05-29:
@theo-vt do you mind weighing-in on this in lieu of  
* https://github.com/FreeCAD/FreeCAD/pull/21084

theo-vt, 2025-05-30:
Hi hello!

The sketch autoscale pr re-uses the code of the sketcher's scaling tool, it works for scaling up and down, but fails on the sketch provided by @ericvsmith 

Thanks,

closed 2025-06-03 by kadet1090