FreeCAD/FreeCAD #18765: Core: There is no consitent mechanism for writing migrations

kadet1090, 2024-12-27:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Currently we don't have any good mechanism to make code that does migrations. i.e. to apply changes to existing document in order to keep them compatible with newer versions. Migrations are usually done as part of restoring process, reacting to reading no longer existing property etc. This is hard to navigate, impossible to make changes in other direction (to export for older versions, for example) and often done at very specific level without good way to apply the migration in context of whole document.

My proposal is to introduce mechanism similar to one found in every web framework for database migrations. Each migration would have associated number that would dictate order of running migrations. Each document would need additional property (or properties if we want migration groups, which would be useful) with number of last applied migration. If there are migrations available with number greater than the document was saved with we need run all migrations and bump the number. New files are created assuming that all migrations are already executed. 

For example, given migrations:
```
0001 - Automatic Sketch Colors
0002 - External Geometry Style
0003 - New LCS Objects
0004 - X Axis Placement
```

and `Document.MigrationLevel` being `2` we would need to execute migrations no. `0003`, `0004` and bump the number to `4`.

Migrations could be implemented as services and registered in service provider, so they can be implemented independently in each module and executed easily from the core. 

Interface for the service could look like this:
```c++
class Migration {
public:
     /// provides forward migration so old documents can be updated to newer standard
     virtual bool migrate(App::Document* document) = 0;
     /// reverts changes made by the migrate method
     virtual bool revert(App::Document* document) = 0;
}
```

All migrations could then be obtained in core using `ServiceProvider::all<Migration>()` method. For convenience we should also provide `MigrationManager` with methods like `void updateToLatest(Document*)`, `bool wasApplied<TMigration>(Document*)` etc. that would hide the service provider related logic from usecases like showing the dialog for document only once. Alternatively we could register in service provider MigrationManagers for each group and rely on them to manage the migrations.

### Full version info

```shell
1.0
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

kadet1090, 2024-12-27:
FYI @hyarion @wwmayer 

This would be a good candidate for RFC process btw.

hyarion, 2024-12-28:
I think this is a great idea!

This sounds very similar to Django's migration system which I use at work: https://docs.djangoproject.com/en/5.1/topics/migrations/ (BSD license btw)

I'm not quite sure how we would apply it to our documents instead of databases but I'm sure we'll figure something out.
One possibility could be to work directly on the xml tree before restore and use functionality similar to Django's RenameModel/RenameField, as well has writing custom functions like Django's data migrations.

Zolko-123, 2024-12-29:
> mechanism similar to one found in every web framework for database migrations

> Django's migration system

please explain why a web database migration process should apply to CAD file formats, it doesn't appear in your explanation. Specifically, please explain how migration would work for large assemblies shared between multiple designers using different FreeCAD versions on different OSs, which is one of the main advantages of FreeCAD over commercial CAD software.

kadet1090, 2024-12-29:
> please explain why a web database migration process should apply to CAD file formats, it doesn't appear in your explanation

Because the goal is quite similar - migration from older data structure to newer one, applying changes in consistent and easy to follow manner. Database migrations are proven way of doing that to databases, we can get inspired from that and adjust the solution for our own needs.

Why is it important? Over time certain details in files can change due to fixing bugs (or "bugs", i.e. defining previously not well defined behaviors), refactoring features or introducing new ones. Examples from just last months include:
 - Changing from transparency to alpha channel
 - Correction of the X axis placement
 - Proposed renaming of Label2 into Description
 - Addition of automatic color change for sketches

FreeCAD already does that, but the code for this can be found in various places like:
 - `DocumentObject#handleChangedPropertyType` method - useful only when type of property gets changed
 - `DocumentObject#onDocumentRestored`, `Persistance::finishRestoring` methods - which this solution would be the most similar
 - https://github.com/FreeCAD/FreeCAD/pull/18761 - introduces yet another way with version comparison
 - probably there are at least dozen more of ways

All methods that I am aware of share problem that the code is often ran in context of object (so for example it is hard to update all depending objects) or property and not whole document. It is also impossible right now to know which migration code was already executed and which was not. Exporting file in compatibility mode so it can be read by older versions of FreeCAD is also not possible as there is no reverse migration code. This system solves all of mentioned issues.

> Specifically, please explain how migration would work for large assemblies shared between multiple designers using different FreeCAD versions on different OSs, which is one of the main advantages of FreeCAD over commercial CAD software.

Given that migrations are already done in FreeCAD simply in not consistent way I don't see any reasons why this should make any troubles. If you see one, can you explain when it would be a problem?

Zolko-123, 2024-12-29:
> Given that migrations are already done in FreeCAD

this is the problem. There should not be **ANY** need for migration at all **IF** the file and data structure were stable **BUT** the current FreeCAD contributors don't give any thoughts to stability and pile up new incompatible shiny stuff. This migration thingy is just a fig-leaf to hide the bad development process that Ondsel has inflicted onto FreeCAD and which lead to the departure of all good developers. FreeCAD went from 0.18 to 0.19 with the introduction of App::Link by @realthunder and there wasn't any need for migration : this is how competent developers work. 

As an example, I can open 10 year old OpenOffice files in the latest LibreOffice version without a hiccup. I can open my assemblies made with 0.21 in 0.19 without problems (I keep old versions around to check for testing). This should be the goal, unfortunately it all collapsed with 0.22 and Ondsel

kadet1090, 2024-12-29:
Please refrain from calling names here, especially when that does not matter at all.

> here should not be **ANY** need for migration at all **IF** the file and data structure were stable **BUT** the current FreeCAD contributors don't give any thoughts to stability and pile up new incompatible shiny stuff.

So how do you implement feature like hhttps://github.com/kadet1090/FreeCAD/blob/c7aa698648ac085e84c9452e8fbf1b9b9710a727/src/Mod/Sketcher/Gui/ViewProviderSketch.cpp#L2944-L2955 ? This sort of thing requires calculating value of newly introduced property based on existing ones. This is example use case for this proposal.

> FreeCAD went from 0.18 to 0.19 with the introduction of App::Link by @realthunder and there wasn't any need for migration : this is how competent developers work.

This is not a good example as this introduced new feature, not changed existing one. Introducing new stuff rarely requires migration strategies.

adrianinsaval, 2024-12-30:
> [...] multiple designers using different FreeCAD versions [...] which is one of the main advantages of FreeCAD over commercial CAD software.

I disagree, I think a great advantage that FreeCAD has over commercial CAD is that anybody has access to any version of FreeCAD they need, they're not locked into whatever version their employer bought. So anybody can decide to move to newer versions of FreeCAD at any point without having fear that other collaborators might not be able to get a compatible version.

Backwards compatibility is important and should be of great concern to all devs, forwards compatibility is nice when easily doable but not of critical importance.

OfficialKris, 2025-01-08:
>
> Specifically, please explain how migration would work for large assemblies shared between multiple designers using different FreeCAD versions on different OSs, which is one of the main advantages of FreeCAD over commercial CAD software. 

I understand this need, especially in regard to the various processes for sharing engineering data. For instance, if you have company A who is using version 1.0 of FC and company B using version 1.1. There is currently no mechanism for sharing version 1.1 data from company A with company B. Now one can argue company A should just upgrade to the latest as there is "no cost" but that may not be true. There may be additional costs than just downloading the software such as a requirement for a certain OS LTS version, requirement for pre-audited software, staff training, internal documentation changes, lost hours with new bugs, etc.

I think the proposed solution is great to prevent more of the "just upgrade to the latest" trend, stops a lot of data corruption bugs, and allows for a "Save as FC v1.0, v1.1, v1.2" option which would satisfy organizational requirements. Thanks @kadet1090, great proposal!