FreeCAD/FreeCAD #18775: [CORE] Kernel geometry with defective proportionality and minimum computation angle limits

luisDias89, 2024-12-28:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

This issue follows the conclusions drawn in issue:
https://github.com/FreeCAD/FreeCAD/issues/18743

After a brainstorming session with @tarman3 regarding the performance issue in creating a Helix with a size 'larger than the test size,' we concluded that what is directly affecting performance is the Deviation value of the geometry.
The value changed from 0.5 to 1.5, resulting in a processing time improvement from 10 minutes to just 3 seconds, both for visualization and model definition! I thought FreeCAD was bad, but it’s all just about poor configurations
[image]

Now we come to the problem I am reporting in this issue: when changing from 0.5 to 1.5, I started noticing that small circles began to lose definition.

Video with the explanation comparing SolidWorks with FreeCAD:
https://github.com/user-attachments/assets/8cfde1a7-1260-4ca7-9d49-bd5ed96804a4

And this directly affects the visualization of the models. They look like squares
[image]
[image]

Conclusion:
The Deviation algorithm needs fixing, and adding a 360º/(Min angle interpolation) option would ensure better accuracy for circle definition. This parameter should be set by default, with the flexibility to adjust it as needed for improved model visualization and performance.

### Full version info

```shell
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.39708 (Git) Conda
Build type: Release
Branch: main
Hash: 173ff4da1ef4f075d56c0dc617d1bf00d5f49d4b
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: Portuguese/Portugal (pt_PT)
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/Fusion
Installed mods: 
  * OpenTheme 2024.9.1
  * sheetmetal 0.5.6
```

### Subproject(s) affected?

OpenSCAD

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

tarman3, 2024-12-28:
Please attach your test file

luisDias89, 2024-12-28:
@tarman3 
[Geometri Test.zip](https://github.com/user-attachments/files/18265470/Geometri.Test.zip)