FreeCAD/FreeCAD #18776: Positive offset on a binder turns holes into squares and breaks all subsequent features

knipknap, 2024-12-28:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

In the attached file, the "Fin pin binder" turns the bound circles into rectangles if the offset is any positive value.
This is from a file that used to work in FC 0.21.

[image]

I also tried changing the offsetting method, to "tangent" or "intersecting", with no change in behavior.

Also note that the subsequent "Fan pin pocket" then breaks the model completely.

### Full version info

```shell
OS: KDE Flatpak runtime (ubuntu:GNOME/ubuntu-xorg/xcb)
Architecture: x86_64
Version: 1.1.0dev.39241 (Git) Flatpak
Build type: Release
Branch: (HEAD detached at cdb9276)
Hash: cdb92768fe4f33d900664415c613f9eced5b8927
Python 3.11.10, Qt 6.7.3, Coin 4.0.3, Vtk 9.2.6, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Installed mods: 
  * btl 0.9.9
  * lattice2 1.0.0
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

wwmayer, 2024-12-28:
When I load it in v0.21 then for _Fin pin binder_ the Support property says it links to the sub-element _Pad014.Sketch026_. But Pad014 doesn't exist.

If I change it with `App.ActiveDocument.Binder002.Support = [(App.ActiveDocument.Body001, ("Sketch026."))]` and recompute the document then the squares turn back into circles.

If I load it with v1.0 or v1.1 there are some cryptic warnings and errors after a recompute.
```
<TopoShape> TopoShapeExpansion.cpp(1887): unnamed lower element Vertex1
<TopoShape> TopoShapeExpansion.cpp(1887): unnamed lower element Vertex2
<ElementMap> <string>(1): duplicate element mapping 'Vertex1 -> Vertex1;D1 Vertex1/Vertex2
<ElementMap> <string>(1): duplicate element mapping 'Edge1 -> Edge1;D1 Edge1/Edge2
<Exception> FeatureDressUp.cpp(209): std::vector<TopoShape> PartDesign::DressUp::getContinuousEdges(const PartDesign::TopoShape &) -- Invalid edge link: ;#93a:5b;:M;CUT;:Hfcd:7,F.Face4
Chamfer: Invalid edge link: ;#93a:5b;:M;CUT;:Hfcd:7,F.Face4
```
But the shape binder still won't change back to a circle.

knipknap, 2024-12-28:
Unfortunately I overwrote the original file from 0.21 by saving it with 1.1 before discovering the problem, so now I can't check whether the problem may be "migration" related or how else it may be introduced.

wwmayer, 2024-12-28:
> So I guess at this point it's almost impossible to reproduce the issue.

For v1.x I can confirm that for a positive offset the shape of the binder are still squares.
It needs to be checked if this is caused by the TNP mitigation code.