FreeCAD/FreeCAD #18777: Mesh Formats only visible in Preferences with Mesh workbench active

pmjdebruijn, 2024-12-28:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Preferences -> Import-Export -> Mesh Formats is invisible if the Mesh workbench is not activated.

I would expect the preferences dialogs to be not dependent on which workbench is selected at all.

In the current state this can be quite confusing for less experienced users.

### Full version info

```shell
1.0.0+41~gd9e8fe0d7d
```

### Subproject(s) affected?

Mesh

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

depthoffocus, 2024-12-28:
Those preferences (and the hooks that they have into the default File menu mesh exports, which they definitely affect) are seemingly provided by the workbench code, which needs to load first.

The workaround is to tick auto-load for that workbench in Preferences. Then they will be available whenever you export, regardless of whether you specifically open the workbench.

I do think it would be useful to have these mesh tessellation preferences that are distinct from the view preferences available at all times.

JohnOCFII, 2024-12-28:
It seems especially relevant as you can export in "Mesh Formats" without ever using the Mesh workbench.

depthoffocus, 2024-12-28:
> It seems especially relevant as you can export in "Mesh Formats" without ever using the Mesh workbench.

My understanding based on my testing is that if you don't load the Mesh workbench, the File > Export mesh deviation/tessellation settings are those of the viewport, which leads a lot of people to set much higher definitions on the viewport than they need for modelling, so that they get a good STL.

Once the Mesh workbench is loaded, the File > Export route appears to use the preference settings, which means you can have different settings for faster modelling and better exports.

It does seem logical that these settings belong in core.

mwganson, 2024-12-28:
There could be a notification of some sort in preferences to inform the user something along the lines of:  "Some settings that are workbench-specific will not be available here until the workbench has been opened in this session."  A popup message would be annoying, but there could be a "Don't show this again" checkbox.

depthoffocus, 2024-12-29:
> There could be a notification of some sort in preferences to inform the user something along the lines of: "Some settings that are workbench-specific will not be available here until the workbench [...]

Or there could just be a placeholder panel for the Mesh formats until the real one is loaded, with a note explaining that Mesh formats export preferences are provided by the Mesh workbench. You could then maybe have a "load" button in *that* panel that loaded the Mesh workbench, and a tickbox for "always load on startup" that affected the toggle for that workbench.

JohnOCFII, 2024-12-29:
That doesn't make sense these preferences are of value even if a user NEVER loads the Mesh workbench!  A standard export (say, after using PartDesign) allows exporting in 3MF and STL, both of which benefit from those preferences.

maxwxyz, 2024-12-29:
You should never just export the object as mesh as it will result in the crazy mesh structure used to display the model inside FreeCAD. Always manually convert it to an STL with the Mesh WB.
So the actual issue is that there is no warning?

pmjdebruijn, 2024-12-29:
@depthoffocus ah, that explains some inconsistent behavior I've been seeing as well.

But wouldn't the short term fix be enabling autoloading the mesh workbench by default? At least then the behavior will be consistent and the 'Mesh Formats" preferences will be generally available.

At least until refactoring can happen.

pmjdebruijn, 2024-12-29:
I'm not too familiar with the FreeCAD codebase, but this seems to work:

```
diff -Nurpd a/src/Mod/Start/StartMigrator.py b/src/Mod/Start/StartMigrator.py
--- a/src/Mod/Start/StartMigrator.py	2024-12-27 14:21:06.000000000 +0100
+++ b/src/Mod/Start/StartMigrator.py	2024-12-29 13:34:10.204344595 +0100
@@ -66,6 +66,7 @@ class StartMigrator2024:
         if autoload_module == "StartWorkbench":
             self.start_prefs.SetBool("ShowOnStartup", True)
             self.general_prefs.SetString("AutoloadModule", "PartDesignWorkbench")
+            self.general_prefs.SetString("BackgroundAutoloadModules", "MeshWorkbench,PartDesignWorkbench")
         else:
             self.start_prefs.SetBool("ShowOnStartup", False)

```

depthoffocus, 2024-12-29:
> You should never just export the object as mesh as it will result in the crazy mesh structure used to display the model inside FreeCAD. Always manually convert it to an STL with the Mesh WB. So the actual issue is that there is no warning?

If you start up the Mesh workbench and then use File > Export, the STL file generated seems to be somewhat under the influence of the Mesh workbench, though? Because the preferences the Mesh workbench adds to export formats take precedence over the (display-oriented) defaults. 

So my feeling is the actual issue here is that File > Export apparently behaves differently depending on whether the Mesh workbench has been loaded. 

The documentation for this makes this counterintuitive state of affairs more befuddling to a new user:

https://wiki.freecad.org/Import_Export_Preferences#Mesh_Formats says:

_This page is only shown if the  [Mesh Workbench](https://wiki.freecad.org/Mesh_Workbench) has been loaded in the current FreeCAD session._

Then that page (https://wiki.freecad.org/Mesh_Workbench) says:

_There are some [export preferences related to Mesh Formats](https://wiki.freecad.org/Import_Export_Preferences#Mesh_Formats) but these are not used by commands belonging to this workbench. They are used by the [Std Export](https://wiki.freecad.org/Std_Export) command._

If, as the Mesh Workbench documentation says, these panels and their effects are only used by Std_Export, then the intuitive behaviour would be that they should be in core.

If the correct answer is that one should *never* as you say use File > Export for STL, then... why is it possible at all? :-)

Do the additions the Mesh workbench panels provide not resolve the issues you're talking about?